FormerLurker/Octolapse #430: External Camera Setup - Full Diagnostic Testing Results in Serial Connection Error

jonhan511, 2020-01-20:
#### If this is a feature request describe it here

N/A

#### Version of Octolapse
 
Octolapse Version: 0.3.4

#### Version of OctoPrint

OctoPrint Version: 1.3.12

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred

1. set up external dslr camera for octolapse
2. set debug to Full Diagnostic - Test Mode
3. started gcode print
4. "Unknown Error" -Disconnects Printer
4. Reconnect, chose second gcode to print, crash again

#### What should have happened?
Test Print Starts and Snapshots from DSLR taken

#### What happened instead?
Serial Connection Disconnects Printer from Octoprint, Disabling Octolapse allows printing to operate normally

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: 0.17.0

#### Printer model & used firmware incl. version

Printer Model: Creality Ender 3 Pro
Printer Firmware Version: Marlin TH3D U1.R2.A1

#### Browser and version of browser, operating system running browser

Browser: Chrome
Browser OS: Windows 10 x64

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: https://github.com/jonhan511/Octolapse-Crash/blob/master/octolapse.gcode

#### Link to settings.json

Link to settings.json with all passwords removed: https://github.com/jonhan511/Octolapse-Crash/blob/master/Settings.json

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://github.com/jonhan511/Octolapse-Crash/blob/master/plugin_octolapse.log

#### Link to octoprint.log

Link to octoprint.log:  https://github.com/jonhan511/Octolapse-Crash/blob/master/octoprint.log

#### Link to contents of Javascript console in the browser

n/a

#### Screenshots and/or videos of the problem:

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2020-01-20:
So I see this error in your octoprint.log:

```
2020-01-20 08:34:04,992 - octoprint.util.comm - ERROR - Something crashed inside the serial connection loop, please report this in OctoPrint's bug tracker:
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 1762, in _monitor
    self._handle_ok()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 2232, in _handle_ok
    self._continue_sending()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 2354, in _continue_sending
    elif job_active and self._send_from_job():
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 2879, in _send_from_job
    self._logger.debug("Command \"{}\" from file not enqueued, doing another iteration".format(line))
UnicodeEncodeError: 'ascii' codec can't encode character u'\ufeff' in position 10: ordinal not in range(128)
```

It looks like the \ufeff character is a unicode zero width non-breaking space.  I couldn't find the character within your gcode file.  It's possible that pasting into gist cleared out the character?  I imported your settings.json and tried a print with your gcode, and it worked as expected.  I suspect one of the following will fix your issue:

1.  Download your gcode from gist and try running that.  If it works, that means there is a unicode character somewhere in your start gcode that should be removed.  Unfortunately \efeff is invisible, so it will be difficult to detect.  It would appear that your cursor gets 'stuck' when trying to move over the character, taking two arrow movements to navigate past an invisible space.
2.  Upgrade to the latest Octoprint release candidate (it could be a unicode conversion problem that has been fixed in the latest release).  This could be why it works for me.
3.  Upgrade to Octolapse 0.4.0rc1.dev3 (devel branch on this git).  This may or may not help.  Does your gcode work when you disable octolapse?

We'll eventually get to the bottom of this.

jonhan511, 2020-01-21:
So I followed your recommendations and updated to the latest Octoprint RC and that seemed to have solved the issue. 

I thought it was noteworthy that serial connection issues only occurred in debug mode. 

Thanks for your help!

closed 2020-01-21 by jonhan511

FormerLurker, 2020-01-21:
I'm very glad to have helped!  The next Octoprint version is pretty excellent, and I'm excited to see it out in the wild!