FormerLurker/Octolapse #432: Octolapse - camera error

saveddrip3d, 2020-01-21:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: Octolapse 0.3.4

#### Version of OctoPrint 

Profile | Name |  
-- | -- | --
Printer | cr-10s pro  |  
Stabilization |   |  
Snapshot |   |  
Rendering |   |  
Cameras | dslr canon m50
Debug |   |  

OctoPrint Version: OctoPrint 1.3.12

Profile | Name |  
-- | -- | --
Printer |   |  
Stabilization |   |  
Snapshot |   |  
Rendering |   |  
Cameras | dslr
Debug |   |  

Profile | Name |  
-- | -- | --
Printer |   |  
Stabilization |   |  
Snapshot |   |  
Rendering |   |  
Cameras | dslr
Debug |   |  

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred

1. time lapse 3d print

#### What should have happened?
time lapse 3d print

#### What happened instead?
Octolapse - camera error

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: 0.15.1

#### Printer model & used firmware incl. version

Printer Model: cr-10s pro / mk3
Printer Firmware Version: PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2020-01-21:
Oh, I see your messages on Youtube.  I was just notified about your responses.  Not sure why I didn't see any notifications before, however.  Darn you youtube :)

Anyway, In addition to the info I asked for above, please also add your take-snapshot.sh script to the gist, so I can double check that.  Also, when you try the script from putty, is it working?  If so, how long does it take before it returns?  You can use the time function to get the execution time, for example:

```
time yourscript.sh
```

Now, look in your camera profile and make sure the **Snapshot Timeout** is long enough (this setting is in milliseconds, where 1000 milliseconds = 1 second).

saveddrip3d, 2020-01-21:
i did get it working. i read your comment on another post about the time the camera was taking to relay the picture. so i increased the pause in the octolapse settings to 6000 from 5000 and its working! when i enter that into putty it returns with real 0m0.005s, user 0m0.001s, sys 0m0.004s

FormerLurker, 2020-01-21:
Hmmm.  0.0005 real seconds sounds impossibly slow, but glad it is working now!!

FormerLurker, 2020-01-21:
Impossibly fast I meant, lol!  It has been a long day...

saveddrip3d, 2020-01-21:
its .005 not .0005 if that makes a difference. sorry if im not answering that right! im not the most tech savvy person lol

FormerLurker, 2020-01-21:
Not really, lol.  It should take at least a second or two.  Not sure what's going on.  Can you run the script without the time command and paste in the output?  Were you feeding in the test parameters (see the test of the script in the guide)?

saveddrip3d, 2020-01-21:
do you mean i should enter yourscript.sh into putty? and the test parameters? /home/pi/scripts/take-snapshot.sh 1 1 "" "/home/pi/scripts" "" "/home/pi/scripts/test.jpg"?

FormerLurker, 2020-01-22:
Exactly, and  the time command from above.  So

time  /home/pi/scripts/take-snapshot.sh 1 1 "" "/home/pi/scripts" "" "/home/pi/scripts/test.jpg"

saveddrip3d, 2020-01-22:
real    0m8.133s
user    0m0.324s
sys     0m0.350s

saveddrip3d, 2020-01-22:
and it seems to be still working! 3 shots in a row no errors.

FormerLurker, 2020-01-22:
Cool.  8 sec is more reasonable.  Glad it is working!!

saveddrip3d, 2020-01-22:
i cant begin to thank you enough for making this possible and for your shockingly quick and patient responses!

FormerLurker, 2020-01-22:
No problem!  Glad to help!

closed 2020-01-22 by saveddrip3d