FormerLurker/Octolapse #433: Filament retracts after snapshot

chaseosborn, 2020-01-23:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version:  Octolapse v0.4.0rc1.dev3

#### Version of OctoPrint

OctoPrint Version: OctoPrint 1.3.12 running on OctoPi 0.16.0

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: NO

#### What were you doing when the problem occurred

1. Start any print and get through layer one
2. Once Octolapse takes a snapshot the filament retracts and will not extrude anymore

#### What should have happened?
Filament should continue to extrude

#### What happened instead?
Prints in mid-air because filament wont extrude
Kinda of looks like the extruder is skipping steps 

#### Operating System running OctoPrint and Octolapse

OS Name: Octopi
Os Version: 0.16.0

#### Printer model & used firmware incl. version

Printer Model: Ender 3 Pro
Printer Firmware Version: Marlin 2.0 bugfix

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: https://gist.github.com/chaseosborn/8fa6cd32894640cbefb6d69c6ebff3ed

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2020-01-23:
Do you know if the ender 3 pro's e axis is affected by G90/G91?  Also, what version of Octolapse are you running?  This could be due to a bug that was fixed in the dev version.

chaseosborn, 2020-01-23:
I am not sure if the G90/G91 command is affecting the e axis or not, in my default start and end code i do have the commands and have never had an issue with the recraction like this, also i am running Octolapse v0.4.0rc1.dev3

FormerLurker, 2020-01-23:
You can run a test to see if g90/g91 influences your extruder through octoprint.  I will write up some instructions and send them tomorrow.  It could be an issue with your printer profile.  In the meantime, please export your settings (gear icon in the Octolapse tab, then export settings), and post them to gist.github.com and paste a link in here.  Thanks!

chaseosborn, 2020-01-23:
That would be great, here is my octolapse settings as well:
https://gist.github.com/chaseosborn/67f8f910f10075c7345448344031f845

FormerLurker, 2020-01-24:
Excellent, thanks!  I will take a look at this tomorrow when I'm in front of my PC and working on Octolapse.  I'm away and on mobile ATM.

chaseosborn, 2020-01-24:
Thank you for the update!

FormerLurker, 2020-01-24:
So, there is something I'd like you to test.  From reading your issue description it looks like you flashed to Marlin 2.0?  Did you customize the firmware at all, or was this from Creality (I'd love a link to exactly where you got the firmware)?  It is possible that the new firmware has changed the printer's behavior slightly.  Please try this simple test (read the comments for details, you don't need to send those) to see if the G90/G91 command affects your extruder:

```
G90 ; set XY axis to absolute
M82 ; Set extruder to absolute mode
G92 E0 ; reset extruder absolute position
G1 E10 F1000; Advance extruder by 10 at a fairly slow speed.  E coordinate should be 10
G91 ; Set XY axis to relative mode (maybe the extruder
G1 E10 F100 ; If G91 influenced your extruder, this command should take six seconds 
; to advance the filament.  If G91 does NOT influence your extruder, it should complete 
; instantly (no motion), so run to your printer within 6 seconds and check for motion
```
Let me know how the above test goes.  If G91 DOES influence your extruder, you should set that in your printer profile.

I also read about a bug recently in the marlin firmware regarding this setting.  There was also a bug in Octolapse until recently that, combined with the marlin issue, would cause big problems.  For that reason I suggest you update to the latest devel branch from the plugin manager with the following url before trying any of this:  https://github.com/FormerLurker/Octolapse/archive/devel.zip

Lastly, there is one other option within the printer profile you could try.  It is experimental, but may solve this problem.  Take a look at the **Snapshot Gcode Axis Compatibility Mode** setting, and disable it.  This will prevent Octolapse from switching the axis mode at all during snapshots.  This is a new feature, and has not been tested as well as the method from V0.3.4 (switch to relative for retraction and lift), but it may eliminate some of your extrusion issues.  However, if the **G90/G91 Influences Extruder** setting in your profile is incorrect, retraction may still be off due to Octolapse misinterpreting the current extruder's position.  Switching to 'Absolute' extrusion mode in your slicer should mitigate that problem somewhat, but it's best to get all of the printer profile settings correct.

chaseosborn, 2020-01-25:
Ok, I have updated to the latest version of octolapse as per your link and ran the test. It looks like G90/91 does influence my extruder so i changed that to true in the settings of octolapse and ran a test print and it is working flawlessly. Thank you so much for your help and this great plugin!

closed 2020-01-25 by chaseosborn

FormerLurker, 2020-01-25:
Can you let me know how your current printer firmware was installed?  I want to make sure the default Ender 3 Pro profile has the correct setting for G90/G91 influences extruder.  It's definitely possible I messed that up when creating the profile because I don't have access to one of those printers.

chaseosborn, 2020-01-26:
My current firmware was installed from an SD card on an skr mini e3. if you need a deeper look into my firmware you can find all the files i compiled for it at https://github.com/chaseosborn/Marlin-bugfix-2.0.x

pdfowler, 2021-05-22:
After checking off every other thing that could have changed since my last good print ~2 days ago, I finally remembered that a) I updated all of my plugins about a week ago, and b) I re-enabled octolapse on Wednesday.

As with @chaseosborn, I'm also on an SKR Mini E3 (the v2 board), and (am pretty sure that I) installed it from the Marlin-bugfix repo. I'm trying to find the configs from my last firmware build and would be happy to help investigating this in any way I can.

FormerLurker, 2021-06-01:
@pdfowler, did you try setting the g90/g91 influences extruder setting to 'True'?  I think that's the issue here.