FormerLurker/Octolapse #434: Extruder disabled mid-print

Cystennin, 2020-01-23:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: Octolapse 0.3.4

#### Version of OctoPrint

OctoPrint Version: Version 1.3.12

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred

1) Disable Octolapse.
2) Slice Cube.STL using Slic3rPE and "Send GCode" to OctoPrint.
Expected Behavior: Cube is printed
Actual Behavior: Cube prints as expected.
3) Enable Octolapse.
4) Reslice Cube.STL using Slic3rPE and "Send GCode" to OctoPrint again.
Expected Behavior: Cube is printed as expected.
Actual Behavior:
Print job completes, however the object is only partly printed - 50-70 of the 125 expected layers in one instance.
Octolapse shows only 50 snapshots taken.
Stepper motor for the extruder is not energized and moves freely.

#### What should have happened?
See above under Expected Behavior

#### What happened instead?
See above under Actual Behavior

#### Operating System running OctoPrint and Octolapse

OS Name: Raspian/OctoPi
Os Version: Version 0.17.0, running on Raspberry Pi 3 Model B Rev 1.2

#### Printer model & used firmware incl. version

Printer Model: Ender-3 PRO with BigTreeTech SKR Mini 1.2
Printer Firmware Version: Not sure - stock version.

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: https://github.com/Cystennin/OctoLapse_BugReport/blob/master/cube.gcode
gist and pastebin were a serious pain with the larger files, so I created a public repo and included direct links

#### Link to settings.json

Link to settings.json with all passwords removed: https://github.com/Cystennin/OctoLapse_BugReport/blob/master/Settings.json

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://github.com/Cystennin/OctoLapse_BugReport/blob/master/plugin_octolapse.log

#### Link to octoprint.log

Link to octoprint.log:  https://github.com/Cystennin/OctoLapse_BugReport/blob/master/octoprint.log

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: https://github.com/Cystennin/OctoLapse_BugReport/blob/master/blue_cube_dd_20200123151853.mp4

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

Cystennin, 2020-01-23:
I noticed that there was a similar issue reported back in december with folks who had updated their controller's to the SilentBoard 1.2 - this issue is here reproduced on an SKR Mini 1.2 which _I think_ has the same drivers on it, the TMC2209's. Other than that they are different firmware I think.

FormerLurker, 2020-01-24:
Thanks for the complete error report!  I will take a look tomorrow when I'm back in front of my PC as I'm away and on mobile right now.

Cystennin, 2020-01-24:
No worries, thanks for looking at it!

FormerLurker, 2020-01-24:
So, towards the very end of the log (Line 276635), Octolapse looks to have found 124 images (one fewer than the number of layers, which isn't surprising).  Where was Octolapse reporting that only 50 images were taken?  Did you have more than one camera going perhaps?

Also, I can verify via the logs that extruder movements are being sent to the printer (Octolapse just logs the sent gcode, it never modifies them in any way), so I think your suspicion about something with the firmware/drivers is causing this.  I have read that retracting too far can cause problems with your current firmware (perhaps a safety feature of some kind), and with that in mind, here is some snapshot gcode generated by Octolapse:

2020-01-23 14:44:20,827     G1 E-2.00000 F3600.000
2020-01-23 14:44:20,830     G91
2020-01-23 14:44:20,835     G1 Z0.600 F7800.000
2020-01-23 14:44:20,838     G90
2020-01-23 14:44:20,839     G1 X-7.600 Y227.600
2020-01-23 14:44:20,839     G1 X121.461 Y98.539
2020-01-23 14:44:20,840     G91
2020-01-23 14:44:20,846     G1 Z-0.600
2020-01-23 14:44:20,849     G1 E2.00000 F3600.000
2020-01-23 14:44:20,852     G90
2020-01-23 14:44:20,854     G1 F1718.000
2020-01-23 14:44:20,854     G1 X121.461 Y121.461 E0.77587

So, it's possible that the final three commands are causing some problems.  I'm not sure if G90/G91 affects your extruder or not (If you know this, let me know), but in that case the your extruder would be left in absolute mode.  Probably not, though, else you'd have some major problems causing slippage and blobing on every layer.  If you do, please let me know.

I have a few ideas you could try to see if the issue goes away:  

1.  Try slicing in absolute extrusion mode and see if that helps or hurts.
2.  You could test your extruder axis and see if it's affected by G90/G91 (I expect it is not).  It's relatively simple to do this.  Here's what I would do for this:  Unload your filament first!  Then, using the Octoprint Terminal, send the following commands to your printer:

'''
G90 ; set XY axis to absolute
M82 ; Set extruder to absolute mode
G92 E0 ; reset extruder absolute position
G1 E10 F1000; Advance extruder by 10 at a fairly slow speed.  E coordinate should be 10
G91 ; Set XY axis to relative mode (maybe the extruder
G1 E10 F100 ; If G91 influenced your extruder, this command should take six seconds 
; to advance the filament.  If G91 does NOT influence your extruder, it should complete 
; instantly (no motion), so run to your printer within 6 seconds and check for motion
'''
Let me know the results of this please.

3.  Go to the Octolapse main settings (Octoprint Settings->Octolapse->Main tab->Edit main settings) and disable the **Show Real Snapshot Time** option.  Perhaps M400s are causing you problems?  Disabling that option will make it so that Octolapse sends only 1 M400 during the snapshot phase.

4.  You could try a test where you disable retraction completely in your test print and inside of Octolapse.  This would be a good way to see if the retraction stuff is all a wild goose chase.

5.  You could try Octolapse V0.4.  Not sure if that would help, but there is a new option that tells Octolapse never to change axis modes at all in the snapshot gcode.  This feature was added so that more streamlined gcode can be generated, and perhaps it could help in your situation?  Let's not try this until you've tried options 1-4.

6.  You could also preemptively create a support ticket for the board manufacturer and see if they know what could cause this.  I see no error reports from you printer, nor do I see any commands that would cause any havoc at all, but if anyone would have any insights into the board and things that can go wrong, it would be the mfg.  Also, is there a homepage/repository for the firmware?  It's much more likely it's a software issue (either firmware, octolapse, or octoprint), or some conflict than a hardware issue (even though your extruder is not energized).

I eagerly await your response!  I'd like to figure out what's up with this, hopefully before the next release drops.

Cystennin, 2020-01-24:
Hmm, can you tell me how to set absolute extrusion mode in Slic3rPE? I can't find it...
I am currently in the middle of a number of print jobs due by Sunday, but will try #2 as soon as I can.
Thanks again for your help on this!

FormerLurker, 2020-01-25:
Yes, go here to specify absolute extrusion:  

[image]

Just uncheck that box and reslice.  However, DEFINITELY try to use the test in step 2 first.  I JUST resolved another similar issue for someone with an Ender 3 pro, and am trying to make sure the G90/G91 influences extruder setting is correct for the default profiles.  Can you also let me know where you got the firmware for your printer, and if it's stock firmware, or specific to your upgrade?

FormerLurker, 2020-02-02:
Any updates?  Had your problem been solved?

FormerLurker, 2020-03-26:
Since this issue has been open for a while, I'm going to go ahead and close it.  Let me know if you want it reopened, and thanks for your post!

FormerLurker, 2020-05-08:
Forgot to close this, sorry!

closed 2020-05-08 by FormerLurker