FormerLurker/Octolapse #435: retract and detract all filament

3dhmpf, 2020-01-26:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: Octolapse 0.3.4

#### Version of OctoPrint

OctoPrint Version: Version 1.3.12

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: NO but serial logging (serial_log_and_settings.log)

#### What were you doing when the problem occurred

1. Trying to setup retract for stabilization timelapse 
2. Edited the Printer Profile Slicer/Octoprint Start Code in 8 different ways:

Octolapse Printer Settings:
X/Y/Z Axis Mode: Requires Explicit G90/G91
E Axis Mode: Default to Absolute
G90/G91 Influences Extruder: True

 Octolapse Printer Settings:
X/Y/Z Axis Mode: Requires Explicit G90/G91
E Axis Mode: Default to Absolute
G90/G91 Influences Extruder: False

 Octolapse Printer Settings:
X/Y/Z Axis Mode: Default to Absolute
E Axis Mode: Default to Absolute
G90/G91 Influences Extruder: false

 Octolapse Printer Settings:
X/Y/Z Axis Mode: Default to Absolute
E Axis Mode: Default to Absolute
G90/G91 Influences Extruder: True

 Octolapse Printer Settings:
X/Y/Z Axis Mode: Requires Explicit G90/G91
E Axis Mode: Requires Explicit M82/83
G90/G91 Influences Extruder: True

 Octolapse Printer Settings:
X/Y/Z Axis Mode: Requires Explicit G90/G91
E Axis Mode: Requires Explicit M82/83
G90/G91 Influences Extruder: False

 Octolapse Printer Settings:
X/Y/Z Axis Mode: Default to Absolute
E Axis Mode: Requires Explicit M82/83
G90/G91 Influences Extruder: false

 Octolapse Printer Settings:
X/Y/Z Axis Mode: Default to Absolute
E Axis Mode: Requires Explicit M82/83
G90/G91 Influences Extruder: True

See also:
https://gist.github.com/3dhmpf/789aaf8a791dfad1c7b810be8135ce0a#file-serial_log_and_settings-log

#### What should have happened?
The Printer should have done a 4mm retract before going to snapshot position. After taking a snapshot ist should have been detractet 4mm.

#### What happened instead?
The Printer is retracting the completed amount of printed filament. Afterwards detracting the completet amount of printed filament.

#### Operating System running OctoPrint and Octolapse

OS Name: OctopPi
Os Version: 0.15.1,

#### Printer model & used firmware incl. version

Printer Model: Geeetech A30 HW Version 3.2
Printer Firmware Version: 1.38.59.

#### Browser and version of browser, operating system running browser

Browser: Chrome
Browser OS: Win 10 

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: https://gist.github.com/3dhmpf/789aaf8a791dfad1c7b810be8135ce0a#file-gcode-gcode

#### Link to settings.json

Link to settings.json with all passwords removed: https://gist.github.com/3dhmpf/789aaf8a791dfad1c7b810be8135ce0a#file-setting-json

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://gist.github.com/3dhmpf/789aaf8a791dfad1c7b810be8135ce0a#file-plugin_octolapse-log

#### Link to octoprint.log

Link to octoprint.log:  https://gist.github.com/3dhmpf/789aaf8a791dfad1c7b810be8135ce0a#file-octoprint-log

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2020-01-26:
Thanks for submitting a complete issue!  Unfortunately the log for Octolapse you posted did not include any debug info (you have to switch your debug profile to include additional logging), but I do see this perhaps unrelated error indicating you are trying to run a DSLR script without having gphoto2 in your sudoers file:  2020-01-26 13:55:56,025 Error output was returned from the snapshot script: [sudo] password for pi: 

In any case, the first thing to do is to verify whether or not your printer's E axis is actually affected by G90/G91.  This has caused some serious confusion regarding setup of Octolapse, and unfortunately I cannot find anyway around it.  Slicers can easily work around this by controlling the extrusion parameters and adding extra M82/M83 and axis position resets (G92)s.  Fortunately, it's pretty easy to run a test to see what the correct setting for G90/G91 influences extruder.  Run the following script through OctoPrint's terminal AFTER unloading your filament:

```
G90 ; set XY axis to absolute
M82 ; Set extruder to absolute mode
G92 E0 ; reset extruder absolute position
G1 E10 F1000; Advance extruder by 10 at a fairly slow speed.  E coordinate should be 10
G91 ; Set XY axis to relative mode (maybe the extruder
G1 E10 F100 ; If G91 influenced your extruder, this command should take six seconds 
; to advance the filament.  If G91 does NOT influence your extruder, it should complete 
; instantly (no motion), so run to your printer within 6 seconds and check for motion
```

Notice that I've added a bunch of comments.  You don't have to send those to the printer, but they should explain what's happening.  The critical bit is the very last line (G1 E10 F100).  If that command causes your extruder to move, then G90/G91 influences your extruder.  You will have approx six seconds to move from your computer to your printer to check the extruder for motion.

Regarding the axis mode settings (X/Y/Z Axis Mode, E Axis Mode), set those to 'Default To Absolute'.  Those settings should be fine for your printer.

Here's my guess as to what's happening.  I think G90/G91 DOES influence your extruder, and that there's a bug in Octolapse v0.3.4 (corrected in V0.4) regarding this setting.  If this ends up being true, you should be able to re-slice using relative E (instead of absolute E) as a workaround, OR you could upgrade to Octolapse V0.4 alpha.   Let me know if you want to try the alpha after we verify your printer profile setting for the G90/G91 influences your extruder.

3dhmpf, 2020-01-28:
First of all thank you for your very fast reply.

I know about the errors produced by gphoto2, but the script works fine, i wasn't able to edit the "sudoers file", i wasn't able ta get root access. Needed to add the password to the snapshot script as described here: https://github.com/FormerLurker/Octolapse/wiki/Configuring-an-External-Camera

Anyway, it's not related to the topic I think.

I will have a further look into this issue about G90/G91 asap when the printer finished a multi day printjob. I wondered why the settings in Octolapse took no affect on the behaviour of the printer thats why I started this topic.

I would like to upgrade Octolapse V0.4 alpha - would be an honour to be part of it.

3dhmpf, 2020-01-28:
Okay... Here it is:

Send: G90
Recv: ok
[...]
Send: M82
Recv: ok
[...]
Send: G92 E0
[...]
Recv: ok
[...]
Send: G1 E10 F1000
Recv: ok
[...]
Send: G91
Recv: ok
[...]
Send: G1 E10 F100
Recv: ok
[...]

The Extruder does extrude as expected. G91 influence the Extruder.

3dhmpf, 2020-02-02:
Update to my issue... Installed the dev, made changes in Cura Start GCode, set the Printer Profile to autodetect, enabled xyz and e-Axis to "Default to Absolute" and G91/90 influences Extruder"true"... No Change in the behavior of over retracting :cry:
What did I missed?

Updatet the gist... Settings.json: https://gist.github.com/3dhmpf/789aaf8a791dfad1c7b810be8135ce0a#file-settings-json

FormerLurker, 2020-02-02:
Can you also attach your gcode?  Your slicer profile is set to automatic, but the gocde file I found within the gists has no embedded settings.  I do see that your z-lift speed is unusually low (4mm/sec or 240mm/min), but somehow I doubt this is involved.

3dhmpf, 2020-02-04:
Here is the Gcode in a new Gist, the GCode in the old gist was the initial GCode.
I will have a look into the Z-Hop settings, never changed them I think.

<script src="https://gist.github.com/3dhmpf/391df30cf5ea249dae5d07522fdcdb05.js"></script>

FormerLurker, 2020-02-04:
Hey, I thought you used the 'Automatic' slicer setting in your printer profile and added the automatic configuration script to your start gcode.  I don't see this in the gcode you just posted.

3dhmpf, 2020-02-05:
Sorry wrong link...
This should be the correct link[GCode](https://gist.github.com/3dhmpf/391df30cf5ea249dae5d07522fdcdb05)

3dhmpf, 2020-02-06:
Thank you so much for your help!
The main issue is solved now! 

Now there is something new, the Timelapse won't be created, I only changed this one setting mentioned above.

Will do some tests with logging and open a new issue if needed.

closed 2020-02-06 by 3dhmpf

FormerLurker, 2020-02-06:
@3dhmpf, so glad to hear this!  I am still confused as to what is going on with your firmware.  If you feel up to it, perhaps you can experiment switching axis modes and see how your printer responds.  It would be great to be able to pin that down so that I can get some insight and hopefully prevent this issue for others.  Though the axis mode issue went away for you, depending on the start gcode and extrusion settings (relative vs absolute), other issues may pop up.

When you post regarding your timelapse, can you include the snapshot files?  You should be seeing a red button on the Octolapse tab called 'Unfinished Renderings' where you can export your snapshots to a zip file:

[image]

[image]

I'm very glad now that I added the rendering recovery and archive features :)  BTW, I fixed a rendering issue very recently that can cause a failure if your camera delivers a snapshot of 0 bytes.  The image was being counted, but does not get extracted due to the way the ZipFile library operates.  You might want to check the zip file after you download it for any jpg files with size 0.  If you see any, try upgrading Octolapse to the latest devel branch via this link:  https://github.com/FormerLurker/Octolapse/archive/devel.zip

Anyway, it was a pleasure working with you!

3dhmpf, 2020-02-06:
The firmware is nearly closed source in that Printer, Marlin isn't available for this Mainboard. There are 4 different Mainboards, i got the one which is not supported by marlin right now. 
No chance to do changes in the Firmware except steps, dir etc... by a special "Motor Tool" which only works when the moon is in the right position ;-)

I would include the snpashot file, but there was no file, no unfinished Renderings, no error or something like this. However with the latest devel this behaviour disapears. Rendering works fine again.

Even the DSLR works 👍 

I just submitted the Printer Profile, I hope at the right location.

FormerLurker, 2020-02-06:
Hooray!  That make me very happy :)

I don't see the gist in your profile, or in the link you send earlier with your gcode file.  Do you mind posting a link to it?  Once I have it I'll upload it to the profile repository.

3dhmpf, 2020-02-06:
Here it is ;) 

[Printer Profile](https://gist.github.com/3dhmpf/f7530e1159977cdaba6dcdb318479c57)

I'm afraid i'm not familar with github.

FormerLurker, 2020-02-06:
Thanks!!

> I'm afraid i'm not familar with github.

I don't think you really want to be, lol!  It is a lot of work..

FormerLurker, 2020-02-06:
So, I added the profile.  If you click the gear icon in the Octolapse tab, followed by the **Update Profiles** button, the new profile can be added.  Then you can edit your geeektech profile and select your make/model to import.  Let me know what you think!

3dhmpf, 2020-02-06:
Will test it after the first full DSLR Timelapse print is finished tommorow... You're doing a awsome job!!!