FormerLurker/Octolapse #436: Error using Canon 80D

samd12, 2020-01-28:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.3.4

#### Version of OctoPrint 1.3.12

OctoPrint Version: 1.3.12

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: NO

#### What were you doing when the problem occurred

Followed online configurations of yours.  They are awesome thank you.  Webcam is working.  I was trying to implement Canon 80D for better quality...

#### What should have happened?
files downloaded

#### What happened instead?
error

#### Operating System running OctoPrint and Octolapse

OS Name: Raspberry
Os Version: updated

#### Printer model & used firmware incl. version

Printer Model: PRINTER_MODEL_GOES_HERE
Printer Firmware Version: PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the 
[Info.zip](https://github.com/FormerLurker/Octolapse/files/4120257/Info.zip)
project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

samd12, 2020-01-28:
Can not get files attached, log files are here https://drive.google.com/open?id=1dSggHTI29RrTcpn2gqzlYpGYt6NWWlks

samd12, 2020-01-30:
snapshot-thumbnail-create-error: Create Thumbnail - An unexpected exception occurred while creating a snapshot thumbnail. Check the log file (plugin_octolapse.log) for details. - Inner Exception: decoder libtiff not available

samd12, 2020-01-30:
All tests work fine from wiki, at a loss.  anyone??

FormerLurker, 2020-01-30:
Is your camera set to make raw images by chance?  It should be set to jpeg only.  Let me know!

samd12, 2020-01-30:
omg I would have sworn I checked that.  Working, Thanks!!

FormerLurker, 2020-01-30:
No problem!  Happy printing!

samd12, 2020-01-30:
Woohoo
https://www.youtube.com/watch?v=ogefQ0X0SzY

FormerLurker, 2020-01-30:
Wow, very good quality!  Nice setup :)

FormerLurker, 2020-02-02:
I hope things are still going well!  Since the issue appears to be solved, I'm going to close this issue down.  Thanks for posting!

closed 2020-02-02 by FormerLurker