FormerLurker/Octolapse #437: Position error at end of print

samd12, 2020-01-31:
If this is a feature request describe it here
FEATURE_REQUEST_DESCRIPTION_GOES_HERE

Version of Octolapse
Octolapse Version: 0.3.4

Version of OctoPrint 1.3.12
OctoPrint Version: 1.3.12

When you ran into the problem, did you have diagnostic logging enabled?
Diagnostic Logging was Enabled: NO

What were you doing when the problem occurred
Error comes at the end of the print.  Everything seems to work though
What should have happened?

What happened instead?
error

Operating System running OctoPrint and Octolapse
OS Name: Raspberry
Os Version: updated

Printer model & used firmware incl. version
Printer Model: Sovol
Printer Firmware Version: Copertec 1.5.0
[image]

Browser and version of browser, operating system running browser
Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

Link to the gcode file you were printing when the problem occurred
Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

Link to settings.json
Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

Link to plugin_octolapse.log
Link to plugin_octolapse.log: LINK_GOES_HERE

Link to octoprint.log
Link to octoprint.log: LINK_GOES_HERE

Link to contents of Javascript console in the browser
Link to javascript console output: LINK_GOES_HERE

Screenshots and/or videos of the problem:
Screenshot/Video Links: LINKs_GO_HERE

Please consider becoming a patron
If you like this project, please support my work by becoming a patron, and consider adding a 'star' to the repository. It takes a lot of time and effort to maintain the
Info.zip
project and respond to issues. The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various videos and tutorials by subscribing to my Youtube channel. You can also follow me on Twitter.

FormerLurker, 2020-02-02:
So, what do you have your build volume set to in your Octoprint printer profile.  I cannot tell because you didn't attach your settings as is suggested by the issue template.  I can see from your error that the build volume is at least 280mm wide, so ensure this is true in your Octolapse printer profile.

FYI, these position errors have been pretty much removed from the next version of Octolapse, though it's still important to get the build volume correct.

FormerLurker, 2020-03-26:
This issue has been open for a while with no response, so I'm going to go ahead and close it.  Please let me know if you want it reopened!

Thanks for your post!

closed 2020-03-26 by FormerLurker