FormerLurker/Octolapse #438: mjpegstreamer and Logitech C615 customizations

sbalick, 2020-02-02:
#### Version of Octolapse
Octolapse Version: 0.3.4

#### Version of OctoPrint
OctoPrint Version: 1.3.12

#### When you ran into the problem, did you have diagnostic logging enabled?
Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred
The camera resolution or frame rate is not capable of being a custom value in /boot/octopi.txt.  I have made sure to uncomment the last 2 lines and set them according to the instructions for this version.

1. Set the camera_usb_options="-r 1920x1080 -f 10"
2. Set the camera_usb_options="-r  1280x720 -f 10"
3. Set the camera_usb_options="-r 640x480 -f 10"
4. Set the camera_usb_options="-r 1920x1080"
5.  Set the camera_usb_options="-r 640x480"
6. set the additional_brokenfps_usb_devices=(046d:082c). --- this was reported by lsusb. -> NO DIFFERENCE here, still errors out when trying to enable the Enable and Apply Preferences Before Printing
7. I have consulted all the other similar reports and issues here on GitHub.  Issue 186 was similar, but my problem was not solved.

#### What should have happened?
All of the resolutions and frame rates above should have allowed the "Enable And Apply Preferences Before Print" to enable further settings and operate as discussed and described.

Additionally, the web page for the (camera url):8080 should display the MJPG-Streamer Demo Pages.  Instead, two images are presented only - a snapshot image, and below is a live view image.  These are properly formatted. However not consistent with all of the documentation.

#### What happened instead?
Upon test, an error message was reported that the server denied access to mjpgstreamer.
[image]

#### Operating System running OctoPrint and Octolapse
OS Name: OctoPi
Os Version: 0.17.0

#### Printer model & used firmware incl. version
Not relevant but for completeness:
Printer Model: Ender 3 Pro
[plugin_octolapse.log](https://github.com/FormerLurker/Octolapse/files/4143583/plugin_octolapse.log)

#### Browser and version of browser, operating system running browser
Safari and Firefox - latest
Browser: BROWSER_VERSION_GOES_HERE
Browser OS: Mac OSX Catalina - latest

#### Link to octopi.txt that works
https://gist.github.com/sbalick/58df73ea03c44af9193bb3896586da7a

#### Link to octopi.txt that doesn't work
https://gist.github.com/sbalick/5c771243fb02b540b0f1be28bd7b5181

#### Link to the gcode file you were printing when the problem occurred
Not applicable

#### Link to settings.json
Link to settings.json with all passwords removed: https://gist.github.com/sbalick/fdc3870eea582e9f9ad4d01d1161315b

#### Link to plugin_octolapse.log
Link to plugin_octolapse.log:  https://gist.github.com/sbalick/6e035abacf23a38f428d5cc92701a7f3

#### Link to octoprint.log
Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE
 ---- pic attached to the issue report ----

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2020-02-02:
This looks like an issue with mjpgstreamer.  Octolapse makes no modifications to mjpegstreamer other than sending values to the control.htm page.  The fact that you can't see the control page suggests something else is going on.  

You say you tried to reach the cameraurl:8080.  Have you tried cameraurl:8080/control.htm?  This page must be accessible in order to use the custom image preferences in Octolapse.

I see that you have a lot of plugins installed.  It would be a good idea to try disabling all non-bundled plugins except for Octolapse, and trying your various octopi.txt again.  You could also try killing the mjpgstream via ```sudo service webcamd stop```, then manually starting the stream with your desired resolution and framerate.  Here is an example using port 8081 with a resolution of 1080p at 15fps:

```
/home/pi/mjpg-streamer/mjpg_streamer -i "/home/pi/mjpg-streamer/input_uvc.so -r 1920x1080 -f 15 -d /dev/video1" -o "/home/pi/mjpg-streamer/output_http.so -w /home/pi/mjpg-streamer/www -p 8081 "
```
Once started, you can try to connect to control.htm and see if it works with your various settings.  This way you can try things completely outside of OctoPrint to eliminate any conflicts.  An added benefit is you'll see any errors outputted by mjpgstreamer that may help diagnose the problem.

You may also try updating mjpgstreamer (not sure what version you are running).

sbalick, 2020-02-05:
Thanks for reaching out on this.  I've taken a stepwise approach here and present my actions and results.

1) Action: Disabled all non-included plug-ins.
	Result: No change. Was not able to access cameraurl:8080/control.htm

2) Action: Set octopi.txt camera_usb_options to default: camera_usb_options-“r 640x480 -f 10”
	Result: No change.  Was not able to access cameraurl:8080/control.htm

3) Action: Set octopi.txt camera_usb_options as commented line - using hard coded defaults.
	Result: Able to access cameraurl:8080/control.htm.
	
4) Action: Re-enable all non-included plug-ins.
	Result: Able to access cameraurl:8080/control.htm
	Conclusion 1: Plug-ins not influencing issue.

5) Action: stop the camera service.
	Result:  Confirmed all cameras stopped.

6) Noted that /var/log/webcamd.log is not present. I got this from error statement in cameraDevice is listed in slush and devices /dev/video0 and /dev/video1 appear when camera is plugged into Pi.

7) In https://github.com/foosel/OctoPrint/wiki/Webcams-known-to-work, the Logitech C615 is known to work with Pi 3.  I have a Pi 4b.

8) Stopping the service and adding "-n" option to bypass mjpg-streamer reports errors relating to Pan/Tilt/Focus/Lights…   This was recommended by https://stackoverflow.com/questions/13437244/mjpg-streaming-with-a-raspberry-pi-and-a-webcam
	Results: Access via webpage at octopi.local/webcam/?action=stream operates and reports the correctly requested resolution. Frame rate cannot be confirmed. Can access /octopi.local:8080/control.htm.   * CANDIDATE SOLUTION *

9) Added resolution, frame rate and “-n” option to octopi.txt.  Command line interface reports no errors.  Modified /boot/octopi.txt with -n option.
	Results: 
		Can access cameraurl:8080.  
		Cannot access cameraurl:8080/control.htm. 
		cameraurl/webcam/?action=stream operates BUT the resolution has changed to 640x480 despite an octopi.txt configuration request of 1280x720 @15fps.
		/var/log/webcamd.log does not exist.
		Cannot confirm frame rate in window.
		Webcam view in Webcam tab in Octoprint works, but only at 640x480 size (I’m guessing this is normal) (octopi.local/webcam/?action=stream)
		Octoprint is still reporting same errors.

10) Removed frame rate and resolution request to octopi.txt.  Retaining “-n”.
	Results: No change from 9.

Conclusions:
* At least on the Pi 4, using 1280x720 @15fps, the OctoLapse capture is not allowing a test image capture nor providing a proper result to the octopi.local:8080/control.htm.  Only default values are allowed. (640x480 at unknown frame rate)
* There is something preventing a requested (and standard) resolution on the camera(perhaps the frame rate too?) from requested value to 640x480. Not sure if this is OctoPrint or OctoLapse using the octopi.txt for camera configuration.  mjpg-streamer is capable of running this resolution and requested frame rate to 30fps (no errors presented.  Excerpt:

pi@Octopi:~ $ /home/pi/mjpg-streamer/mjpg_streamer -i "/home/pi/mjpg-streamer/input_uvc.so -r 1280x720 -f 30 -d /dev/video0 -n" -o "/home/pi/mjpg-streamer/output_http.so -w /home/pi/mjpg-streamer/www -p 8080"
MJPG Streamer Version: git rev: 501f6362c5afddcfb41055f97ae484252c85c912
 i: Using V4L2 device.: /dev/video0
 i: Desired Resolution: 1280 x 720
 i: Frames Per Second.: 30
 i: Format............: JPEG
 i: TV-Norm...........: DEFAULT
 o: www-folder-path......: /home/pi/mjpg-streamer/www/
 o: HTTP TCP port........: 8080
 o: HTTP Listen Address..: (null)
 o: username:password....: disabled
 o: commands.............: enabled

Without the -r option:

pi@Octopi:~ $ /home/pi/mjpg-streamer/mjpg_streamer -i "/home/pi/mjpg-streamer/input_uvc.so -r 1280x720 -f 30 -d /dev/video0" -o "/home/pi/mjpg-streamer/output_http.so -w /home/pi/mjpg-streamer/www -p 8080"
MJPG Streamer Version: git rev: 501f6362c5afddcfb41055f97ae484252c85c912
 i: Using V4L2 device.: /dev/video0
 i: Desired Resolution: 1280 x 720
 i: Frames Per Second.: 30
 i: Format............: JPEG
 i: TV-Norm...........: DEFAULT
UVCIOC_CTRL_ADD - Error at Pan (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Tilt (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Pan Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Pan/tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Focus (absolute): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Tilt (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan/tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Focus (absolute): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at LED1 Mode: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at LED1 Frequency: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Disable video processing: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Raw bits per pixel: Inappropriate ioctl for device (25)
 o: www-folder-path......: /home/pi/mjpg-streamer/www/
 o: HTTP TCP port........: 8080
 o: HTTP Listen Address..: (null)
 o: username:password....: disabled
 o: commands.............: enabled

I would like to be able to control the camera to the extent possible, and have more reasonable resolutions provided in the output to see the detail.  I have not yet attempted to capture a time-lapse, but will be doing so shortly and report my findings. 

I'll continue to dig, but hopefully this gives you some information or be able to provide me some additional guidance.

Thanks again!

FormerLurker, 2020-02-05:
Thanks for the detailed report.  

So, for most of these issues Octolapse is not involved in any way.  Octolapse does not engage mjpeg-streamer or read anything from octopi.txt.  For this reason I really can't help you solve the resolution/framerate problem.  However, I can try to help you with the custom image preferences.  For help getting your resolution and framerate set in octopi.txt, I would advise that you post on the [OctoPrint User Forum](https://community.octoprint.org/).  I can almost guarantee someone there knows what's going on.  Provide all of your data, and you'll get a quick response.  Once the webcam works and is loaded, you can continue trying to get Octolapse all configured to work.

So, just to clarify what program is responsible for what, I'll outline a bit here.  OctoPi (not OctoPrint) actually deals with getting mjpeg-streamer up and running (hence the octopi.txt file).  Once it is running, OctoPrint uses the camera stream (/webcam/?action=stream) to show the camera stream in the interface.  Both OctoPrint and Octolapse use the snapshot urls (/webcam/?action=snapshot) for taking timelapse snapshots.  FYI, starting in Octolapse v0.4, a camera stream is shown to help see what changing your image preferences is doing to the actual image, which is really helpful.  If you want to update to this version after we get your camera preferences working, let me know!

Now, as far as I know Octolapse is the only plugin that interfaces with mjpeg-streamer's control.htm page.  Octolapse relies 100% on mjpeg-streamer to make settings changes.  Octolapse does this by sending commands to control.htm (they are just basic get requests).  If control.htm is not accessible, or if it does not actually respond to the commands that are sent, nothing will happen.  The error you posted originally indicates that Octolapse got either a 40X or 50X error (probably a 404) when submitting a request to control.htm, which means it either isn't accessible, or it isn't responding.

I hope the above helps to explain what program is responsible for what action, and how they interact.  Briefly restating the above, you need to make sure you can access control.htm, and that you can use control.htm to change your camera settings before attempting to use those features within Octolapse.  Here is what I see when I navigate to control.htm when using a picam:
[image]

You won't see all of those controls when using your camera.  Here is a page that is likely to be similar to what you will see when everything is working:

[image]

I saw that you attempted to manually start mjpeg-streamer.  That is good.  FYI, here's the output I get when manually starting mjpeg-streamer with my Logitech C920:
```
/home/pi/mjpg-streamer/mjpg_streamer -i "/home/pi/mjpg-streamer/in                                                                         put_uvc.so -r 1920x1080 -f 30 -d /dev/video1" -o "/home/pi/mjpg-streamer/output_                                                                         http.so -w /home/pi/mjpg-streamer/www -p 8081 "
MJPG Streamer Version: git rev: 821c330ea6bbb5fbed98d48e00aac156e923161b
 i: Using V4L2 device.: /dev/video1
 i: Desired Resolution: 1920 x 1080
 i: Frames Per Second.: 30
 i: Format............: JPEG
 i: TV-Norm...........: DEFAULT
UVCIOC_CTRL_ADD - Error at Pan (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Tilt (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Pan Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Pan/tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Focus (absolute): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Tilt (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan/tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Focus (absolute): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at LED1 Mode: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at LED1 Frequency: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Disable video processing: Inappropriate ioctl for dev                                                                         ice (25)
UVCIOC_CTRL_MAP - Error at Raw bits per pixel: Inappropriate ioctl for device (2                                                                         5)
 o: www-folder-path......: /home/pi/mjpg-streamer/www/
 o: HTTP TCP port........: 8081
 o: HTTP Listen Address..: (null)
 o: username:password....: disabled
 o: commands.............: enabled
```
Even with all the errors above (the author states they are harmless), I am able to control everything through Octolapse.

So, that brings me to some questions:

1.  Can you take a snapshot of what you are seeing in control.htm when it works (you stated that sometimes it works, and sometimes it does not)?  
2.  When control.htm works, IF you actually see controls, when you try to change settings, do they actually affect your live stream (/webcam/?action=stream)?  The results of settings changes should be nearly instantaneous, but if you use the text box and not the arrow buttons, remove your cursor  [... cut ...]
4.  IF control.htm is displaying but not working, you can check mjpeg-streamer to see if it is properly detecting the reported capabilities by navigating to this URL:
```
http://{ip_of_pi}/webcam/input.json
```
If you can't get to that URL, something is wrong (make sure the IP is correct, and if you are using a different port, 8081 in this case, it may look more like {ip_of_pi}:8081/input.json).  The json file should look a whole lot like the output of v4l2-ctl.

FYI, I have verified that you are using the most recent master release of mjpeg-streamer by comparing the commit guid to the repository version, which is good.

I hope this helps!

FormerLurker, 2020-02-26:
It's been a while since this thread was updated, so I'm going to go ahead and close it now.  If you are able to run the tests I mentioned in my previous post, please feel free to update the issue and I'll reopen it.  Thanks for posting, and good luck!

closed 2020-02-26 by FormerLurker