FormerLurker/Octolapse #439: Cannot get to the config page to change focus, contrast, etc.

Rivenin, 2020-02-02:
Hi! hopefully this is easy

octopi.txt has the following last lane 

camera_http_webroot="./www"
camera_http_options=""

rebooted the pi

tried to get to the following and all i get is the picture preview

octopi:8080
192.168.0.201:8080
octopi:8080/
192.168.0.201:8080/

How do i get to the m-jpeg camera settings??

FormerLurker, 2020-02-02:
Try http://192.168.0.201:8080/control.htm

If you don't get a config page that looks like this:

[image]

You might be using a picam.  Let me know if this is the case.  I added picam support (it's not perfect, but pretty good.  Requires some octopi.txt changes) in Octolapse V0.4 (alpha), and you can try that if you are using one of those.

However, you can also configure the setting within the Octolapse camera profile by enabling custom image preferences.  Again, this has been greatly improved in V0.4.

Rivenin, 2020-02-04:
That worked! thank you! 
I'm using a logitech series USB cam - going to tweak with this later.

FormerLurker, 2020-02-07:
I'm very glad, and you are certainly welcome!  Should I consider this issue closed?

Rivenin, 2020-02-14:
Yep! Everything is working. 👍👍👍

closed 2020-02-14 by Rivenin