FormerLurker/Octolapse #440: [Request] Add custom before and after gcode

danielfmo, 2020-02-02:
#### If this is a feature request describe it here

Be able to add custom GCODE to be executed before and after the screenshot.

Use case: I have my printer in an enclose and I'm using klipper to turn ON and Off the enclosure LEDs. I wand to be able to turn these ON before the screenshot is taken, and OFF after.
For that I need to be able to configure GCODE commands to be executed. e.g. M42 S1 Penclosure_leds

#### Version of Octolapse
 
Octolapse Version: 0.3.4

#### Version of OctoPrint

OctoPrint Version: 1.3.12

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES_OR_NO

#### What were you doing when the problem occurred

1. STEP_ONE_GOES_HERE
2. STEP_TWO_GOES_HERE
3. STEP_...

#### What should have happened?
PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: OS_NAME_GOES_HERE
Os Version: OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: PRINTER_MODEL_GOES_HERE
Printer Firmware Version: PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2020-02-02:
That's a good idea.  I will see if I can slip it in real quick.

FormerLurker, 2020-02-02:
Ok, these have been added to the devel branch.  Please give it a try by installing from the plugin manager with the following URL:  https://github.com/FormerLurker/Octolapse/archive/devel.zip

BEFORE installing doe these two things:

1.  Please read [this guide] about using v0.4 (a LOT has changed), and pay close attention to the section about automatic slicer settings.
2.  Download your settings.json file via Octoprint Settings (wrench/spanner icon)->Octolapse->Download Settings (or possibly export settings).  This will be useful if you want to revert to the previous version.

After you update and reboot, You will find the new script sections in the camera profile.  They are executed before any before/after snapshot bash/batch scripts run, and octolapse will wait for any moves to finish before continuing.  This may cause some issues on some printers, however, due to M400s being sent without any movement, and I may eventually deal with that.  For now let's see if it does what you want it to do.

Thanks!

FormerLurker, 2020-02-07:
@danielfmo, have you had a chance to take a look at this?  I'm curious if it meets your needs, and if it actually works for you.  I'd feel much better including this in the next RC if someone else gives these scripts a try.

Thanks!

FormerLurker, 2020-02-15:
Have you had a chance to check these out yet?  I've run some basic testing myself, but I'd love it if you could give them a try for your use case.

I'll give it a few more days before closing the issue.

FormerLurker, 2020-02-26:
@danielfmo, it's been a while since I've heard from you, so I'm going to assume the changes I made in response to your feature request are working.  I'm going to go ahead and close the issue, but would still love to hear from you regarding this addition.  I will reopen the issue if you run into problems.  Thanks for posting!

closed 2020-02-26 by FormerLurker