FormerLurker/Octolapse #441: Rendering Fails on Most prints

gdotp01, 2020-02-02:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
Octolapse Version: OctoPrint 1.3.12 running on OctoPi 0.16.0

#### Version of OctoPrint
OctoPrint Version:  0.4.0rc1.dev3

#### When you ran into the problem, did you have diagnostic logging enabled?
Diagnostic Logging was Enabled: YES(Live Print)

#### What were you doing when the problem occurred

1. Print Runs 
2. Only obvious issue is that the Octolapse screen display does not show a preview 
3. When finished I get the error message that the rendering has failed

#### What should have happened?
Renering

#### What happened instead?
Octolapse Rendering Failed
Rendering failed for camera 'Default Webcam'. render-error: Unknown render error. Please check plugin_octolapse.log for more details.. Inner Exception: [Errno 2] No such file or directory: '/home/pi/.octoprint/data/octolapse/tmp/octolapse_rendering_tmp/remixed-85_0.15mm_PLA_MK3S_1h54m000138.jpg'

I've downloaded the renders and unzipped and one o the files is blank, number 76, all the others are Ok so i presume that the rendering failed when it got to this file. I've dropped the zip file with the various logs here too (i've removed most of the photos die to file size restrictions but left the one that errored and the one either side)

[dd46e35c-50ab-4f7a-a90d-4c69e59b2a4f 3.zip](https://github.com/FormerLurker/Octolapse/files/4145628/dd46e35c-50ab-4f7a-a90d-4c69e59b2a4f.3.zip)

#### Operating System running OctoPrint and Octolapse
Raspberry Pi 3 Model B Plus Rev 1.3
OS Name: Raspbian GNU/Linux 9 (stretch)
Os Version: OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: Prusa Mk3S
Printer Firmware Version: 3.8.1

#### Browser and version of browser, operating system running browser
Browser: Safari 
Browser OS: Version 13.0.4 (15608.4.9.1.3)

#### Link to the gcode file you were printing when the problem occurred
Link to Gcode File: https://gist.github.com/gdotp01/1b02a68c6ea72f2217fd7be665c2ad26#file-remixed-85_0-15mm_pla_mk3s_1h54m-gcode

#### Link to settings.json
Link to settings.json with all passwords removed: https://gist.github.com/gdotp01/7829449a36afad32b2df404e48db11c6#file-settings-4-json

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

gdotp01, 2020-02-02:
BTW im running Pillow version 6.2.2, I was on 6.2.1 but when this error originally happened I updated it in the hope it might resolve it

FormerLurker, 2020-02-02:
Nice report 👍.  Can I ask when you installed this version?  I did fix a few bugs recently.  I will import and attempt to render your zip and will see what happens.

gdotp01, 2020-02-03:
Thanks I try. Great Software btw. I downloaded this on Friday last week but only because the previous beta was not working. 

The issue happens with most Gcodes, I’ve tried a few. It occasionally works. My son needed a video for his 3D school design project which uncovered the issue and after a couple of prints I managed to get a render to work with an identical file. It also seems to work better in Test mode without the printer printing anything but this could just be luck. 

It seems that sometimes (once every 150 shots or so) the jpg image is not written out and there is a blank file which then when rendering causes the error)

FormerLurker, 2020-02-03:
So, I took a look at your zip file, and I'm seeing a directory structure issue, but I have no idea how it could have happened.  Octolapse can import images from archives in two ways:

1.  All files in the root of the zip.
2.  Files that follow the Octolapse structure of JOB_GUID\CAMERA_GUID\*.JPG

Your zip file looks to the the structured kind, but the JOB_GUID directory has been altered somehow.  It looks like this in your zip:

dd46e35c-50ab-4f7a-a90d-4c69e59b2a4f 3

Notice the space at the end and the trailing 3.  I think this is the root of the problem.  Any ideas how this got there?

Also, I just learned about the __MACOSX directory from your zip.  This shouldn't have hurt anything, but it would have resulted in some junk files being extracted, so I have prevented this directory and anything inside it from being read.

For now I'm going to manually rename this directory, but will need to figure out what happened there.

FormerLurker, 2020-02-03:
OK, I've adjusted the code so that files of size 0, or any files that fail to convert to JPEGs are handled appropriately (not counted as snapshot images).  Your zip file, after I renamed that directory, imported and rendered successfully.  You will want to update to the latest devel branch of Octolapse from the plugin manager using this url:  https://github.com/FormerLurker/Octolapse/archive/devel.zip

Also, I noticed that you have not enabled the [automatic slicer settings](https://github.com/FormerLurker/Octolapse/wiki/V0.4-Getting-Started-with-Octolapse#use-automatic-slicer-configuration) within your printer profile.  I recommend you do this since all of your slicer settings are already embedded within your gcode file.  You might also want to take a look at [this guide](https://github.com/FormerLurker/Octolapse/wiki/V0.4-Slic3r,-Slic3r-PE-and-PrusaSlicer), which explains how to turn on the verbose gcode feature.  This will increase quality when using the *Smart Triggers*, which I also recommend you check out.

gdotp01, 2020-02-03:
Thanks. Im away until Thursday and will give it a go then. I normally have teh automatic slicer settings set but I'd turned it off trying to diagnose the issue. Thanks ever so much for such great and prompt support.

FormerLurker, 2020-02-03:
No problem, let me know how it goes!  I also added a popup with a help button in the case that no images are found within an archive.  The help link explains what can be done to fix the file (admittedly, it is a bit rough ATM).  When you get around to testing the fix, perhaps you could try to upload the zip file you sent to me, and see if the error message and help works and is reasonable.

gdotp01, 2020-02-06:
Hi, Ive updated and it did render the failed output I had successfully. I've also re-enabled Automatic Slicer settings so I'll give a real print a stab the next couple of days. 

Thx

gdotp01, 2020-02-07:
All looks good. I’ve done a few small prints now and they have all rendered ok. Thanks

FormerLurker, 2020-02-07:
Fantastic!  Thanks for bring this to my attention since I'm sure this issue has been causing problems for others for a while now.

Closed!

closed 2020-02-07 by FormerLurker