FormerLurker/Octolapse #442: Octopi camera setings revert to same bitrate in control.htm on reboot

manroop5, 2020-02-04:
followed guide at https://github.com/FormerLurker/Octolapse/wiki/Configuring-a-Raspberry-Pi-Camera
edit pencil step 8 not available Custom Image Preferences enabled and applied , camera working
Able to change camera settings in control.htm but the bitrate reverts back on reboot

#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: OCTOLAPSE_VERSION_GOES_HERE

#### Version of OctoPrint

OctoPrint Version: OCTOPRINT_VERSION_GOES_HERE

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES_OR_NO

#### What were you doing when the problem occurred

1. STEP_ONE_GOES_HERE
2. STEP_TWO_GOES_HERE
3. STEP_...

#### What should have happened?
PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: OS_NAME_GOES_HERE
Os Version: OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: PRINTER_MODEL_GOES_HERE
Printer Firmware Version: PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2020-02-04:
What version of Octolapse are you using?

manroop5, 2020-02-04:
0.3.4

FormerLurker, 2020-02-05:
So V0.3.4 will apply your camera settings before the print starts, but not on reboot.  V0.4 has an option to apply the settings on reboot.  Would that solve your issue?  If so, I will provide instructions for trying out v0.4.0rc1.dev3.

manroop5, 2020-02-05:
I am unable to change camera settings from octolapse interface. None of those settings are available. The pencil shown  (on guide) on the side of snapshot does not show up in my octolapse.  I use the mpeg streamer control page to change bitrate,settings erc. It stays the same even if I stop and restart camera service. One I reboot raspberry pi all the settings are gone ie bitrate, contrast brightness etc and have to be changed again.  
Also the camera zooms in automatically once I set the resolution to 1920x1080. Zoom in octopi interface does not work either. 
Thx

FormerLurker, 2020-02-05:
In the very top of the guide for enabling custom image preferences for the raspberry pi camera there is a warning stating that the guide is for v0.4.  the pencil icon doesn't exist in v0.3.4, nor does full support for the picam.  The image settings will stay the same till you reboot, even in v3.4.  also, the picam does not support zoom.  The camera interface for the picam in octolapse v4 only shows controls the picam supports.  If you want to use the custom image preferences and all of the associated goodies for the picam, you will have to upgrade.

manroop5, 2020-02-06:
How do I upgrade? Also will the preferences stick around with updated version. 

Thanks

FormerLurker, 2020-02-07:
Hey, sorry I didn't get back to you earlier, it's been a bit busy.  I actually typed a response on my phone yesterday afternoon, but the github app I was using crashed right at the end and I had to take off, doah!

Anyway, read [this guide](https://github.com/FormerLurker/Octolapse/wiki/V0.4-Getting-Started-with-Octolapse) all the way through before you attempt to upgrade.  It may save you some grief.  

Then, follow the instructions, making sure to use this URL in the plugin manager when you actually install Octolapse:  https://github.com/FormerLurker/Octolapse/archive/devel.zip

Once it installs, take a look at the guide you originally tried to follow for getting the picam to work.  You've already done most of the things necessary to make it work, so it shouldn't take much.  In fact, if you open your camera profile and attempt to enable custom image preferences, it just may start working straight away!

Let me know how it goes.  I'll be around for most of the day if you have issues.

manroop5, 2020-02-08:
Thanks 
Everything works with this new update including the preview screen but the settings still do not stick on reboot. Sometimes they do and other times they revert back. For eg screen rotation, bitrate etc.

FormerLurker, 2020-02-08:
Double check and make sure you have this setting checked:

[image]

manroop5, 2020-02-08:
Its checked and I restarted multiple times after changing the settings and camera inversion. Changes back on reboot.

FormerLurker, 2020-02-08:
When you say, "Changes back on reboot", do you mean the *Apply Preferences When Octoprint Starts* setting or both that setting and your camera preferences.

In any case, please export your octolapse settings (gear icon on the Octolapse tab followed by *Export Settings*), and paste the resulting file to gist.github.com.  Paste a link to the gist in here and I will take another look.  Thanks!

manroop5, 2020-02-08:
Revert to what it was before the settings changed. 
My camera is upside down by default. So it goes back to being upside down, bitrate goes back to 1000000 etc .
File uploaded at 
https://gist.github.com/manroop5/04b6d630122c2a4a99a8fbbb8e804c2c

FormerLurker, 2020-02-09:
So, I've imported your settings with my pi, and they all get applied correctly at reboot (bitrate, flip, etc) :(

However, one thing I noticed is that the camera you are using hasn't been detected as a picam, which might be playing a role.  What picam version are you using?  Also, if you could navigate to this address and paste the results into gist.github.com, that would be very helpful:  http://192.168.1.70:8080/input.json

If you have any trouble getting input.json to show up, let me know.  If I can't figure out what's going on, I may ask you to produce a logfile for me at reboot.  It's kind of a pain, but if you want to do it preemptively, here are the steps:

1.  Just to make sure you are using the most recent version of Octolapse (I've been pushing lots of updates), reinstall from the Plugin manager by using this URL, and reboot:  https://github.com/FormerLurker/Octolapse/archive/devel.zip
2.  Edit the 'Logging' profile, click 'Clear All Logs', then configure the profile so it looks like this:
[image]
Pay special attention to the octolapse.camera **Module to Log**, making sure the **Logging Level** is set to **Verbose**.
3.  Reboot the Pi by clicking the power button and selecting 'Reboot System'.
4.  After it has rebooted, edit your debug profile and click **Download Log**.
5.  Paste the log into the gist where you added your input.json file I requested above.

Hopefully, from all that I can figure out what is going on.  I have two theories:
1.  Mjpeg-streamer hasn't fully loaded when Octolapse is sending the control changes.
2.  The order of the camera settings requests is having an impact.  However, I don't think this is the case since the bitrate shouldn't be affected at all, but I could be wrong.

manroop5, 2020-02-09:
Its raspberry pi 2 camera. 8 MP, from element14
How do you know it's not recognized as picam. Is is because I mounted it as usb cam per instructions? I tried mounting it as picam with previous version of octolapse 0.3.4 and controls didn't work. 
Once I go home I will send you input.json file. 
Thx

FormerLurker, 2020-02-09:
Thanks, I will lookup the model and see what i can find.  Regarding how I know it wasn't detected properly, Octolapse looks at that input.json file at all of the available settings and default, and compares it to a list of known settings to see if there is a custom control page.  I have to physically add each variation and direct Octolapse to the appropriate control page based on the specific camera's reported capabilities.

After I loaded your settings, Octolapse told me it had detected a new type of camera, which tells me the capabilities differed from my picam.  It doesn't mean you did anything wrong when you set your camera up, buy that your system reported different settings.  Once I get your input.json file, I can see what the difference is.  It could be something as simple as slightly different firmware or drivers.  We shall see.

Thanks for sticking this process out.  I think we will get to the bottom of the sooner rather than later.

manroop5, 2020-02-09:
Couple of observations
When I flip the camera back to normal (in regular control screen) from the webcam settings the octolapse camera screen shows it still upside down but the settings stick after multiple reboots.(Maybe that settings are interfering with octolapse settings.)

Then I try to change to normal from octolapse settings and the one on control switches to upside down. Also the jpeg settings and compression settings also revert to default after reboot if you already seen that. 

Thx

FormerLurker, 2020-02-09:
The control.htm mjpegstreamer page should not stick ever.  There is no mechanism that Im aware of that should cause that to happen.  This suggests something deeper is going on.

When you say reboot, which option in octoprint are you picking?  There is an option to reboot the pi, and another to reboot just the octoprint server.  Ive only been testing a full reboot of the pi.  Perhaps there are behavioral differences between the two?

manroop5, 2020-02-10:
No matter what i do the controls in octolapse JPEG Compression Controls/bitrate stays the same as soon as i restart. Even if I save the settings and open them up again they revert back to the same.

FormerLurker, 2020-02-10:
Try following the steps I posted.  I really just need to see the logs and that json file.  Without them I can only guess.

manroop5, 2020-02-10:
input.json
https://gist.github.com/manroop5/c279804e76f1063448017bf5cccd73cf

FormerLurker, 2020-02-10:
That was fast, thanks!  I will dig in tomorrow.  If you can get the log file too, that would be helpful.

manroop5, 2020-02-10:
Strangely after I took the above steps and applied above update (even though its the same version of octolapse  Octolapse v0.4.0rc1.dev3) the settings started to stay the value I set before reboot. ie camera does not go upside down and bitrate seems to be fine but the sliders in octolapse control  (bitrate/jpeg compression controls) stay at their default values. Bitrate slider remains at 10000000, JPEG Compression (Compression Quality  30 JPEG quality 0), unable to save the manual exposure (reverts to automatic ) all after reboot. 
Log file uploaded at 
https://gist.github.com/manroop5/d66a38c7ff24003754192b7deec20a56

FormerLurker, 2020-02-10:
Hmm..  So, depending on when you installed Octolapse, the update may have fixed things.  I've been working on v0.4.0rc1.dev3 for quite a while now, and there have been a ton of updates.

I looked at your log, and it looks like all of the requests to change settings went though, which is good.  The only thing left to figure out is why the camera detection didn't work as expected.  Again, it's possible that this was fixed in some commit that was applied after you upgraded.  If you have a few seconds, maybe you could re-export your settings and post them again.  I can then import those and see if the camera detection sticks.  If it does, I think we can call this issue closed :)

manroop5, 2020-02-10:
new settings 
https://gist.github.com/manroop5/cded212ff02a53dc1e59f4572991bfe8
So the jpeg compression/quality as well as auto exposure are going to stick to defaults for now? I am not sure if the jpeg quality is going up with the slider even though it sticks to default like bitrate which changes despite slider going back to default.

manroop5, 2020-02-11:
The inverted camera/bitrate thing started to happen again. Error log from octolapse attached. 
https://gist.github.com/manroop5/be1ab21b15a5f80297739f72881ccda1

manroop5, 2020-02-11:
The problem with the inverted image after reboot started again. Attached log file. 
https://gist.github.com/manroop5/eb2058382c21b9940dd62ccbc0a524ab

FormerLurker, 2020-02-15:
The only thing I can think of is that your camera stream isn't fully started when Octolapse tries to update your settings.  I'll be working on this today, but this is a tricky and time-consuming issue to test and debug.

FormerLurker, 2020-02-15:
OK, I have altered the apply settings after reboot function so that it retries 9 times over a period of approximately 60 seconds.  I'm going to give it a go on my Pi now, with both my logitech and RPI camera.  Hopefully if my hunch about delayed camera start is correct, this should solve that issue.  I will let you know.

FormerLurker, 2020-02-15:
So, my testing on the pi was successful!  I was able to the the rpi camera to apply the settings consistently, and as as an extra hard test, I configured a camera with no mjpeg-streamer stream started, so that on reboot I had to manually start the mjpeg-streamer stream.  Even with this delay (I did it fast, in under 60 seconds), the settings were applied correctly.

Fyi, here is how the retry works.  If the first connection fails, the system waits 100MS and retries up to 9 times.  However, on each retry, the delay doubles, so that the final retry delays about 30 seconds.  However, in order to make this work well for ALL settings change requests, I have changed octolapse so that if any of the settings fail to apply, the routine stops trying.  This could lead to some strange behavior if a single control fails to set, but I figure I'll wait and see if that causes any issues.  If it does, I may have to alter this feature somewhat.  For now, I'm pretty confident it will work for you.

To test, make sure you install from the latest devel branch (https://github.com/FormerLurker/Octolapse/archive/devel.zip).  Next, reboot via the Octoprint **Reboot System** option.  Once Octoprint comes back up, take a look at your camera screen and see if your settings are applied.  If the settings are NOT applied when you look, please wait up to 120 seconds to be absolutely sure it doesn't get applied on the final retry.  Then maybe try to reboot a few more times, just to make sure it works EVERY time.

Thank you so much!

FormerLurker, 2020-05-08:
Update.  Another issue was found when editing the camera image preferences from the camera profile (as opposed to the edit pen under the timelapse preview to the right of the camera drop down).  It was just too complicated to work out, so I just removed the edit button in the camera profile.  The controls in the tab are better anyway.  I'm considering this closed now.  Thanks!

closed 2020-05-08 by FormerLurker