FormerLurker/Octolapse #443: [Request]

ShamusTheTinker, 2020-02-11:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

I'd like to request an option to include text in the video produced.  I'd like for it to include a time stamp and the current layer the image is taken at.  I had a couple of pieces fail and being able to tell exactly what layer and time the problem occurred would be useful.

#### Version of Octolapse
 
Octolapse Version: OCTOLAPSE_VERSION_GOES_HERE

#### Version of OctoPrint

OctoPrint Version: OCTOPRINT_VERSION_GOES_HERE

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES_OR_NO

#### What were you doing when the problem occurred

1. STEP_ONE_GOES_HERE
2. STEP_TWO_GOES_HERE
3. STEP_...

#### What should have happened?
PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: OS_NAME_GOES_HERE
Os Version: OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: PRINTER_MODEL_GOES_HERE
Printer Firmware Version: PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2020-02-11:
This already exists :)

Check the rendering profile and look for text overlay.  You might need to install fontconfig first, because it isn't installed by default on octopi for some reason.

FormerLurker, 2020-02-11:
Actually, I see you are using the latest development rc.  I'm about to push dev3 out, and it includes one font, so fontconfig won't be required anymore (though you still may want it so you have access to more fonts)

ShamusTheTinker, 2020-02-12:
Awesome, I'm playing with it now.  Thanks

FormerLurker, 2020-02-13:
Hey @ShamusTheTinker, I took another look at this issue, and realized that the {snapshot_number} token won't always be the current layer!  I'd been meaning to add more replacement tokens anyway, so I went ahead and added the following tokens:  {layer}, {height}, {x}, {y}, {z}, {e}, {x_snapshot}, {y_snapshot}

Layer is always an integer.  Height, x, y, z, e, x_snapshot, and y_snapshot are shown as absolute values (regardless of the axis mode) and are all floating point values.  E will display 5 decimals while x, y, z, x_snapshot, and y_snapshot will display 3, which is pretty typical within gcode.  

You can reinstall using this URL in the plugin manager (from url...) to add the new tokens:  https://github.com/FormerLurker/Octolapse/archive/devel.zip

At some point I may add more specific formatting options for each, but for now I think that's pretty good.  What do you think?

ShamusTheTinker, 2020-02-14:
I think that's great, I'm going to install immediately and run a test on it tonight.

ShamusTheTinker, 2020-02-15:
Installed correctly and got a test print off last night.  I love the additions.

FormerLurker, 2020-02-15:
Fantastic!  I'm going to close the issue, but if you run into any problems I will reopen it.

Closed!

closed 2020-02-15 by FormerLurker