FormerLurker/Octolapse #444: I can not configure Octolapse without a pause

Goodfeat, 2020-02-13:
#### If this is a feature request describe it here

Adding the absolute “no pause” function without the intervention of octolaps in the G-code

#### Version of Octolapse
 
Octolapse Version: Octolapse v0.4.0rc1.dev3

#### Version of OctoPrint

OctoPrint Version: OctoPrint 1.3.12 running on OctoPi 0.16.0

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred

Hello. I'm trying to get pictures through octolapse. 
But no matter what filters I choose, my printer still does a micropause before the snapshot

I tried to use smart filters, which in principle should take a screenshot without a pause,
changed the camera delay to zero, but it also causes micro-pauses.

I realized that octolapse processes its actions interfering with the g-code.
I noticed that before the picture, the octolapse empties the command buffer. and only after the next snapshot in the firmware continues to receive the commands of the g-code. this causes the printer to micropause.

Question: is it possible to add a filter that does not empty the printer buffer to take a picture?
It would be nice if you added a filter that takes pictures in parallel with printing.

Standard timelapse works on this principle. but it lacks some of the settings that are in octolapse

#### What should have happened?
the picture should be taken without emptying the printer buffer

#### What happened instead?
the picture is taken with the buffer underrun

#### Operating System running OctoPrint and Octolapse

OS Name: raspberry pi 3
Os Version: OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: Delta
Printer Firmware Version: Repetier 1.03

#### Browser and version of browser, operating system running browser

Browser: Chrome, Operas
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

[image]

Screenshot/Video Links:  https://youtu.be/k_168XJamKY

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2020-02-13:
I actually couldn't see the pause from your video, but I have no doubt there is a brief delay.  Before I address your request, I want to explain where the delay you are seeing comes from.  Here are the major sources:

1. Framerate delay. The average delay added by the snapshot itself is 1/(2*FPS) unless my math is wrong.  It ranges from 0-1/FPS seconds.  Without accounting for this, the printhead will not be where it is expected to be in the acquired image.
2. Image download -  the request itself takes some time.  I have actually added an option to stream most of the download.  This delay could approach zero, so it would be the easiest to eliminate.
3 M400 + M114 command pair. As you mentioned, this clears the buffer. That is how Octolapse knows when the extruder and bed are in position. Without this command, It is difficult to figure out where the printhead actually is.  
4.  Then there is the time it takes for Octoprint/Octolapse to process the printers response, and to send the next gcodes.  This delay is quite small, and could approach zero.  It may be possible to eliminate/nearly eliminate.

There is also camera delay, but that can be set to 0.  I may be missing something else, but I'd wager that is the vast majority.

So, If I removed the M400 command today, and completely eliminate the rest of the delay, the effect would be a non-stabilized timelapse.  It would probably look just like the stock timelapse function.

Now, that's not to say it would be absolutely impossible to do what you are asking.  I've thought a lot about this, actually, but haven't figured out a way to know for sure, or even estimate where the printhead will be without m400.  Even if I did, there would still be uncertainty regarding the snapshot itself (due to frame rate).  However, if the frame rate were the only consideration, I would have already added this feature.  I have thought about trying to use a very accurate print time estimation, and using that to guess where the print head is, but there are a lot of problems with that.  I haven't stopped thinking about this though.

If you (or anyone else reading this) have any ideas regarding how to maintain a stabilized or quasi-stabilized time lapse without using M400, please let me know.  I would LOVE to figure out a way to eliminate (or greatly reduce) all delay.   I have spent months getting it as low as it is now.  Perfectly stabilized images can pretty easily be sub 100MS.  My record is around 60MS, though there was some stuttering occasionally.  I think there is still room for improvement.

Goodfeat, 2020-02-13:
The video shows that before creating a screensaver, the printer buffer is empty

I realized what the problem is

There is such an idea.
The new version of the program has a smart script that analyzes the code before sending the print.

I'm not quite sure how it works, but I think that after analyzing the g-code in advance, you can make a plan for creating screenshots.
When printing starts, your smart script will simultaneously search for triggers for the code to automatically take a screenshot without the M400 command.

In this case, the screenshot will be taken with some delay, but it will be taken approximately at the place where it should be

can this be realized?

In this case, there is no need to stabilize the image. you just need to take a screenshot approximately at the place where it should be. Let this happen with a delay, but without interrupting the print itself and emptying its buffer

FormerLurker, 2020-02-13:
>  you just need to take a screenshot approximately at the place where it should be.

The problem is I don't think it will be close enough to call approximate.  However, I will run some tests and see what happens when I disable M400, and just go ahead and take a picture.  I will post a video of how it turns out when it is complete.

Goodfeat, 2020-02-13:
> Я опубликую видео о том, как это получается, когда это будет завершено.

OK,
if it works in much the same way as a regular time-lapse, will it be possible to add a separate experimentally function "disable the M400 command"?

FormerLurker, 2020-02-13:
Yes, it would be an option somewhere.  I'm not quite sure where the setting should go though.  I will think about that too.

Goodfeat, 2020-02-13:
I think this option should be in the "Trigger" section, since this is an alternative method of creating a screenshot

Goodfeat, 2020-02-13:
I experimented a bit. I tried to exclude the m400 commands from the files (I don’t know which files are specifically responsible for the buffer underrun, so I changed all of them).

received the following results, which were published on YouTube

One tower test octolapse witout m400 (no extrude)
stabilization - disable
trigger - smart to print
https://youtu.be/Mqsv7AT621E

Two towers test octolapse witout m400 (no extrude)
stabilization - disable
trigger - smart to print
https://youtu.be/Hu7a-QhBzOA

Gear test octolapse witout m400 (no extrude)
stabilization - disable
trigger - timer 0.05 sec
https://youtu.be/o1ZosmA6u1g

The last video is very good if you use octolapse for the "time"

for "time" to take screenshots without the m400 command is best suited.
and I was mistaken when I suggested adding the option "without the m400 command" to the trigger section.

It’s best to place it in the “stabilization” command with a separate checkbox in the menu

FormerLurker, 2020-02-13:
So, you edited the source?!  Good for you!  Thanks for the proof of concept.  I'm going to give it a shot on my end now.

FormerLurker, 2020-02-13:
Ok, so I've added this option in an 'Advanced Settings' section within the stabilization settings.  I've also added a stabilization profile you can import called 'Octoprint Style - No Stabilization' that has both axes disabled with wait for moves to finish disabled (no m400).  The tab will show a blue info panel on the top when this mode is active (it says 'Wait for moves to finish has been disabled within your stabilization profile.') since otherwise I fear it will confuse users.  There is a help link next to this message explaining the setting and where to find it.

Also, I've prevented all stabilization actions when wait for moves to finish is disabled.  You can use any trigger you like, and any stabilization you like, but Octolapse will refuse to lift, retract, change feedrates, axis modes, or move to the snapshot position.  It should behave as though it's printing without octolapse enabled at all.

See what you think.  you can re-install from this URL within the plugin manager to see the new setting:  https://github.com/FormerLurker/Octolapse/archive/devel.zip

If you'd like to import the new stabilization profile, make sure you first click on the 'Gear' Icon within the octolapse tab, and click 'Update Settings'.  After that you should be able to import the 'OctoPrint Style - No Stabilization' profile, which behaves, I believe, how you want it to.

I'm just now installing this on my printer and will run a test.  I will let you know how it turns out.

FormerLurker, 2020-02-13:
OK, I had to fix a few glitches.  I pushed the changes to the devel branch.  If you've already installed, please re-install from the same URL I posted above.  I'm going to keep running tests, and will let you know if anything else breaks :)

Goodfeat, 2020-02-13:
Hello, I have not experienced, I just woke up.  Today I will check the octolaps and unsubscribe about the results

Goodfeat, 2020-02-14:
I tested a new octolapse.
I think it's perfect. this function was so lacking for this plugin

[image]

test video https://youtu.be/8vwdwDnwJZc

Probably it seemed to me, but I have
Post-Roll 1 - 1 second,
but he doesn’t seem to generate it now.
when I tested at myself I noticed it too

Is this possible somehow connected with the lack of the M400?

FormerLurker, 2020-02-14:
Well, I'm glad it is mostly working!  Not sure what is going on with the post roll.  I dont see how it could be related to the new "wait for moves to finish" setting, but who knows. Try this:

Edit your rendering profile and enable the snapshot archive setting.

Now run another print in test mode.

When your test print is finished, click on the "Videos and Images" button.  Then click on the snapshot archive tab and download the zip file.  Post it somewhere and paste in a link.  I should be able to figure out what is happening by rendering your timelapse in my debugger.

Goodfeat, 2020-02-14:
ok, try to do it

Goodfeat, 2020-02-14:
[M_0.8_kvadrat_20200214034121.zip](https://github.com/FormerLurker/Octolapse/files/4202568/M_0.8_kvadrat_20200214034121.zip)
I did a post roll for 4 seconds, but for some reason there’s still no generation of an additional 4 seconds

FormerLurker, 2020-02-14:
Thanks!  I'll take a look at it first thing tomorrow.

Goodfeat, 2020-02-14:
Ок

Goodfeat, 2020-02-14:
I tested the print with a new function here and noticed a very unpleasant thing.
the function "without M400" works as expected. but there were difficulties with the drawdown of octoprint performance.

I started printing gear, but during printing I noticed the following
From time to time, new g-code commands are no longer sent to the board after the screenshot. This causes the printer to freeze for 2-4 seconds.

I assume this happens when you take a screenshot while processing commands. But that is not all.
[image]

Most likely, the preview slows down which is processing frames in real time.

here is another bug
in addition, if you put a "trigger  classic - every layer", then at the end of printing, the octolapse continues to work, regardless of whether the print has ended or not. as I understand it, the script ceased to understand when printing ended. (the test was conducted with the mode without M400)

I understand that the timelapse preview does not turn off at all. so i cant check for sure?
Is there a screenshot quality setting in the program?

FormerLurker, 2020-02-14:
I will look into it.  I ended up having to work most of the day.  It was supposed to be my day off :( Anyway, I wanted to let you know that I haven't gotten to your issue yet, but I will soon.

FormerLurker, 2020-02-14:
Oh, also, you can prevent auto update of the latest snapshot in the octolapse main settings.  The image processing is all done on a separate thread, but python threading is notoriously inefficient.  In any case, I will look into it!

Goodfeat, 2020-02-15:
Hello, I'm sorry I ruined your day off.
I will try again to look for the settings for disabling previews and once again test in real print mode.
I could not find this setting yesterday

If I don’t find it, I’ll try to look for a way to disable it in the code.

P.S. If I honestly do not really understand the python code, but often I have to analyze and edit the code to achieve a good result.

FormerLurker, 2020-02-15:
Lol!  It wasn't you that ruined my day off :) I was looking forward to working on Octolapse.  My real job had an emergency unfortunately.

I think that setting is called 'auto reload' or something similar.  I will take a screenshot in I get back home early enough.

Goodfeat, 2020-02-15:
I will try to find this myself again,
but if this is not found, your screenshot will help)

Goodfeat, 2020-02-15:
I seem to have found the setting
[image]

however, after turning off the option, nothing has changed
some lags still appear

[image]
when octolapse is disabled, this defect ceases to appear

The defect appears periodically when the next photo is created by the octolapse. I noticed that some photos are created spending a little more time than others

[test model.zip](https://github.com/FormerLurker/Octolapse/files/4208555/test.model.zip)
I used this model for tests. The model is made with smooth turns to maximize the load on my delta printer.
I use a 32-bit board and under normal conditions it copes with such models without problems

I hope this information helps you somehow.

FormerLurker, 2020-02-15:
Thank you, that will be helpful.  I just woke up, and will start working on this very soon!

FormerLurker, 2020-02-15:
I found and fixed the problem with the post-roll frames.  Turns out it would only fail if there were no pre-roll frames.  If you reinstall from this url, that problem will be solved:  https://github.com/FormerLurker/Octolapse/archive/devel.zip

Now I'm going to look at the M400 problem.  This issue might take a while to fix.

FormerLurker, 2020-02-15:
So, I may have fixed the issue you were seeing, but I had to guess what was causing it.  My theory is that there were occasional delays  fetching a snapshot from your webcam.  The old snapshot routine would wait for a snapshot to be received, so if your buffer ran out while waiting, your printer would pause (unretracted), which is not good.

I have removed the wait (a thread join/event wait) from the snapshot routine.  This could cause issues with camera scripts (the before/after snapshot scripts), since they will both be executed at the same time with no wait now, but it's better than leaving blobs on the print.

Here is an image from a print of the STL you send:

[image]

It looks pretty good, though there is one area that doesn't look as good as the rest.  I'm running a print with Octolapse disabled to see if that problem goes away, or if it's just some minor issue with my printer (I just got a new nozzle).

You might want to reinstall and try again.  I'll let you know about the second test as soon as it's finished.

FormerLurker, 2020-02-15:
So, I ran the test long enough to see that the slight blemishes I saw in one area also appeared with Octolapse disabled.  Give the new build a try and let me know if it works for you!

Thanks!

Goodfeat, 2020-02-16:
Hello
I tried installing the updated octolapse plugin, but it seems nothing has changed.

It seems to me that the link installed the same version that was installed previously with me.
at least the revision I still have Octolapse v0.4.0rc1.dev3
but the plugin installed successfully.
[image]

I don’t know why it happened

FormerLurker, 2020-02-16:
I only increment release version numbers, sorry :( working on that too.

Have you tried printing that model with octolapse enabled AND the M400?  Maybe it is some other problem with Octolapse?  Im not convinced I understand what the  problem is since I can't replicate it.

Maybe you can send me the gcode you are printing?  Perhaps i can see something new?

Goodfeat, 2020-02-16:
I just checked the files that are installed on my raspberry pi.
all changes inside the files correspond to the history of changes on the site
So the update was successful.

[image]

I don’t understand why the print is still lagging.

I print so that the m400 is not used

[image]

I used the same stl file as before. when the octolapse is on, there are defects,
when the octolapse is off, there are no defects at all

Goodfeat, 2020-02-16:
This is the same g-code that I print
[M_test.zip](https://github.com/FormerLurker/Octolapse/files/4209256/M_test.zip)

post roll works now, four seconds are added to the video

Goodfeat, 2020-02-16:
Please tell me which firmware is installed on your printer.
I have a repeater 1.03. the display clearly shows how the command buffer is squeezed. I have 64 commands installed.
when the octolapse takes a picture, it is clear that there is a buffer underrun. sometimes he manages to empty out, sometimes not.
but I can definitely say that the buffer is empty for only 1-2 commands if you turn off the octolapse

Perhaps you have different firmware and kinematics and most likely you do not notice a buffer drawdown.
Perhaps the pictures manage to be processed at you before the buffer is empty

FormerLurker, 2020-02-16:
I'm printing with a prusa MK2.5S with the latest firmware.

If you are getting blobs like that in the regular mode (where M400 is used), I think the problem is not related to M400 at all.  It sounds like the added cpu load of running Octolapse might be causing delays.  Do you have any other plug-ins installed?  If so, try disabling everything but octolapse.

I will look at your gcode tomorrow.  I really want to get the bottom of this.

Goodfeat, 2020-02-16:
[image]

I will try to disable additional plugins, but I have few of them. not sure if this will help, but I will try

I am now dealing with the camera, I have a logitech с270 on my camera, some settings are not available.

FormerLurker, 2020-02-16:
Thanks!  I wish I had a lot more hardware to test.  I know it takes a lot of effort to run all of these tests remotely, so thank you for helping!

Goodfeat, 2020-02-16:
happy to help you i will test the plugin

Where parallel lines are, this is normal, because my rollback is turned off when a picture is taken.
[image]

drops marked in a circle - this is not normal, and it just happens at that moment when the camera’s picture is taken longer than usual. 
it seems to me that time even goes beyond the timeout

FormerLurker, 2020-02-16:
Im familiar with the effect.  Usually it is settings related (retraction usually, or g90-91 influences extruder setting, but I don't believe that is the problem in this case.  I may have you run a test with full logging.  It will create a massive log file.  The 'log everything' profile should work.  It would help to know exactly which layer had a problem.  Maybe I can find something with that.  It takes forever to review a full log, but I'm happy to do it.

FormerLurker, 2020-02-16:
I've been looking at your gcode.  So far everything looks OK.  I did notice that your gcode settings aren't included in the file.  Please follow [this guide](https://github.com/FormerLurker/Octolapse/wiki/Automatic-Slicer-Settings#if-you-are-using-cura-follow-these-steps) for enabling automatic slicer settings detection.  Then re-slice your model and re-post the gcode so I can see and use your slicer settings to run a test.

Also, can you export your current octolapse settings and post them?  You can find them by clicking on the gear icon in the Octolapse tab, then clicking the export settings button here:

[image]

One final thing.  I think I do need a full log file.  First, clear your current log by editing your logging settings:

[image]

and then clicking the Clear Log... button:

[image]

Then switch to the 'Log Everything' profile:

[image]

and save your settings.  Then run a print until you see a blob.  Stop the print, and download your log by editing the logging profile and clicking on the Download Log button:

[image]

Then post the log somewhere and link to it here.

Goodfeat, 2020-02-17:
Hello, I didn’t manage to test octolaps yesterday. Today I will continue the experiment

I made automatic settings detection on the manual and now I will test printing with logs.
I want to ask only a question.
in the script, before each tincture there is a ";" Will the script work if these characters are removed and does this not affect the behavior of the octolapse script?

[new test octolapse +start script.zip](https://github.com/FormerLurker/Octolapse/files/4211313/new.test.octolapse.%2Bstart.script.zip)

FormerLurker, 2020-02-17:
The ";" is necessary.  If you remove them, the automatic settings detection will not work.  It looks like the start script you added worked perfectly!  Octolapse should be able to extract your settings just fine when using "automatic settings detection"!

I'll run your code through the debugger tomorrow.  If you get the log file to me by then, I will review it also.  

Again, thank you so much for your help here!

Goodfeat, 2020-02-17:
I'm afraid the whole log is a complete mistake.

when the log is started, the command buffer does not fill up and the print lags
the command buffer is hardly filled into two teams and immediately drops to one command.
printing stops for a few seconds when a photograph is taken.
I didn’t even understand when to stop him.

[plugin_octolapse.zip](https://github.com/FormerLurker/Octolapse/files/4211342/plugin_octolapse.zip)

FormerLurker, 2020-02-17:
It could be the logging.  Sounds like it may be an IO problem.  What kind of SD card are you using?

Try disabling logging and see if it starts working better.

Goodfeat, 2020-02-17:
I'm starting to suspect something. It seems to be a problem.

[image]

FormerLurker, 2020-02-17:
Looks like a class 2 SD.  Class 10 is recommended.  If you upgrade, I bet you will have much better performance!

Goodfeat, 2020-02-17:
Yes, I’ll make an image of this memory card and try to re-upload the image to another class 10 memory card. I’ll definitely write about the results

Goodfeat, 2020-02-17:
I had to install octoprint on a new tenth grade card.
Now everything is working well. At least printing does not stop if you disable the M400 command.
the command buffer drops from a maximum of 64 to 50 when creating a screenshot

I think it's great

But one detail haunts me.
How did regular timelapse take pictures without interrupting printing? I had a second class memory card and I didn’t observe any performance drawdowns.

maybe he doesn’t save the pictures on the memory card but keeps them in the RAM until the printing process is completed?

FormerLurker, 2020-02-17:
Fantastic!  What a mystery that was!  So glad we finally finished it out.

Regarding the stock timelapse plugin working fine...  Octolapse wasn't designed to work like the stock timelapse plugin.  Normally it has a lot of work to do to figure out how to stabilit, rewrite gcode, execute synchronous gcode, etc.  To get it running without the M400 command, I forced the stock time lapse behavior into the Octolapse process.  That was the easiest way to get it working AND to support most of the other features.  I hope to improve things in the future, but it would take a lot of work to do it efficiency.   TLDR: I did it the fast, but inefficient way, and octolapse has more overhead.

By the way, Octolapse does run OK on my pi2, as well, so IO must be a bit more important than I thought.  I need to run some more code profiling I think.

Goodfeat, 2020-02-17:
It is excellent,
it will be great if you continue to develop octolapse, improving its work including the mode without the M400 command

I will continue to test the plugin, if I find errors, I will let you know

FormerLurker, 2020-02-26:
So, it's been a while since this thread has been updated, so I'll assume things are going well.  If you have any additional problems, please let me know.  Thanks for posting here, and good luck!

Closed!

closed 2020-02-26 by FormerLurker