FormerLurker/Octolapse #445: Smart - Snap-To-Print - Trigger prefers front of model

EleRas, 2020-02-13:
#### If this is a feature request describe it here

The current (v0.4.0rc1.dev3) Smart - Snap-To-Print - Trigger seem to prefer the "front" of a model to create snapshots / stabilizing the timelapse. For some printers it would be better if the bed would be as far in the front as possible, meaning the hotend should be as far back on the model as possible. Taking snapshots with the hotend at the front of the model covers most of the model by the hotend / extruder.

#### Version of Octolapse
 
Octolapse Version: v0.4.0rc1.dev3~e19bb6b

#### Version of OctoPrint

OctoPrint Version: 1.3.12

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES_OR_NO

#### What were you doing when the problem occurred

1. STEP_ONE_GOES_HERE
2. STEP_TWO_GOES_HERE
3. STEP_...

#### What should have happened?
PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: OS_NAME_GOES_HERE
Os Version: OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: PRINTER_MODEL_GOES_HERE
Printer Firmware Version: PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2020-02-13:
You can select the stabilization point yourself.  Choose the rear center of your bed (you can import a profile for this) to position the extruder nearest to the back.  This may or may not work when using the 'smooth' option, fyi.  It's somewhat unpredictable and depends on the model.  Give it a try and let me know how it goes.

Thanks for you post!

EleRas, 2020-02-13:
Yes I can choose the stabilization point with "Stabilization", but the description says to choose "Disabled" for a better print. As far as I understood it the printer just takes the snapshot without moving the head to a pre-determined position (i.e. in the back of the printbed), just based on a chosen point in the model.
This chosen points seems to always be in the front of a model.
My wish would be that I can get the "Smart - Snap-To-Print - High Quality" option with stabilization "Disabled", but the chosen point as far in the back of the model as possible.

FormerLurker, 2020-02-13:
> but the description says to choose "Disabled" for a better print

Where does it say that?  Can you paste in a screen cap for me so I can see it?  This is probably a typo of some kind.

With stabilizations disabled, the snapshot will be taken immediately on layer change, and your current model is probably changing layers at the front of your model.  Just enable stabilization, and select the back center to achieve the effect you are looking for.

EleRas, 2020-02-13:
I can do even better: https://github.com/FormerLurker/Octolapse-Profiles/blob/master/0.4.0rc1.dev3/trigger/smart_snap_to_print_high_quality/profile.json#L24 ;)

FormerLurker, 2020-02-13:
So, the text says this about disabling stabilization

> Consider trying this trigger with the 'disabled' stabilization profile for an interesting effect

It doesn't say that disabling stabilization improves quality at all (which it doesn't).  Your extruder will not move from your print when using snap to print no matter what your stabilization setting is.  When using snap-to-print, octolapse chooses a point nearest to the stabilization point to take a picture.  Just set your stabilization to 'Back Center' and keep your trigger profile set to 'Snap to Print'.  You can see exactly where your extruder will be in the snapshot preview screen (hit the play button for an animation).

Please try out my suggestion and let me know if it works for you.  you can enable 'test mode' if you don't want to waste any plastic trying out these settings (I recommend that).

EleRas, 2020-02-13:
Ahh, thank you very much, that actually cleared a lot of my confusion.

I seem to remember testing it with different stabilizations and it moved the hotend to the chosen stabilization - point, moving away from the print. Maybe I had a wrong selection of the "Trigger" while testing this. I will test it again during the upcoming weekend.

FormerLurker, 2020-02-13:
Sounds great!  I'm glad I could help, and I'm glad someone is interested in the snap-to-print setting :)  It is one of my favorite new features.

EleRas, 2020-02-15:
I apologize, it seems during my previous tests I didn't have the Smart - Snap-To-Print - Trigger selected. Of course the Stabilization-Point would be at the "real" end of the bed in this case. My tests this morning shows that it indeed stays at the model. 

The Snap-To-Print trigger seems to work as designed, thanks for the patience & plugin!

closed 2020-02-15 by EleRas

SPeters1979, 2023-01-24:
This has been very informational.  I have used Snap To Print - Smooth with stabilization disabled and it seemed there was a slight pause by the print head during the snap.  When printing a smooth cone, this pause would cause a second line up the side of the cone, similar to the seam line.  With manipulating stabilization to line up with my seam line (back center) I can effectively combine the seam line and snap to print line into one.  Then as a bonus, I can also get the print head out of the way for the snapshots as well!