FormerLurker/Octolapse #446: octoprint pausing SD prints with M25's when octolapse is enabled

wpegden, 2020-02-15:
#### If this is a feature request describe it here
not a feature request

#### Version of Octolapse
Octolapse Version: 0.3.4

#### Version of OctoPrint
OctoPrint Version: 1.4.0rc5

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: NO

#### What were you doing when the problem occurred
Printing from the SD card, starting from the printer.

#### What should have happened?
Print should have completed as usual.  Octolapse should have done nothing as it does not support SD prints.

#### What happened instead?
M25 is sent to the printer, pausing the print.  If I disable octolapse, this doesn't happen.

#### Operating System running OctoPrint and Octolapse

OS Name: Ubuntu
Os Version: 16

#### Printer model & used firmware incl. version

Printer Model: Prusa MK3S
Printer Firmware Version: latest

#### Browser and version of browser, operating system running browser

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: ommitted since I was printing from SD card

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log
Link to plugin_octolapse.log: https://pastebin.com/AV4CrUcW

#### Link to octoprint.log

Link to octoprint.log:  https://pastebin.com/MjZYhZRd
also: terminal contents (you can see the M25 commands):
https://pastebin.com/vuLgB0ui

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron

FormerLurker, 2020-02-15:
Octolapse definitely isn't sending an M25, and wont start if it is enabled during an SD print.  However, it is possible that some plugin is causing some kind of conflict.  I see lots of exceptions in your log.  Will dig in asap.

wpegden, 2020-02-16:
It did seem strange to me that Octolapse would ever send M25.

Let me know if I can help somehow with the diagnosis.

Thanks!
Wes

FormerLurker, 2020-02-16:
Octolapse is definitely NOT sending an M25.  It doesn't have that code anywhere within the source.  The only it could send that code is if it appears within your gcode file, and that would be unlikely.

Also, Octolapse does not operate during an SD print.  What I expect is happening is some kind of gcode parsing error due to a slightly buried setting in your Octolapse printer profile called 'Suppress Snapshot Command When Disabled'.  Uncheck this option and see if it helps.

Alternatively, I have fixed the possibility of running into this error in the alpha version.  I've been preparing to drop v0.4.0rc1.dev3 all week (hopefully will today or tomorrow).  If you want to try it out, you can install within the plugin manager using this URL:  https://github.com/FormerLurker/Octolapse/archive/devel.zip

Be sure to read this guide all the way through **before** installing:  https://github.com/FormerLurker/Octolapse/wiki/V0.4-Getting-Started-with-Octolapse

I also recommend you go to OctoPrint Settings (wrench/spanner icon)->Octolapse and **download settings* as a backup in case you want to downgrade later.  Unfortunately, in v0.3.4 (the version you have) you will have to manually copy your settings to recover them.  V0.4 allows you to import settings files directly.  I doubt you will want to downgrade, but having the settings file from v0.3.4 will make it easier to debug any upgrade problems.

Let me know how it goes, and thanks for you offer to assist!  I appreciate you taking the time to crate an issue.

wpegden, 2020-02-18:
Disabling this seems to have worked!  I will report back if I see it again.  Thanks!
Wes

FormerLurker, 2020-02-18:
That's great!  I'm going to close this issue for now, since this problem has been fixed in the next version.  Please do let me know if you have any additional issues.

FormerLurker, 2020-02-26:
I'm going to assume that things continue to work for you, and will now close this issue.  If you see it again, let me know and I'll reopen it.

Thanks for posting!

closed 2020-02-26 by FormerLurker