FormerLurker/Octolapse #447: Oсtolaps refuses to take a picture in the start / end g-code.

Goodfeat, 2020-02-18:
#### If this is a feature request describe it here

Please add a hybrid mode for creating a picture by layers + in an arbitrary place on the command in the g-code

#### Version of Octolapse
 
Octolapse Version: Octolapse v0.4.0rc1.dev3

#### Version of OctoPrint

OctoPrint Version: 1.3.12

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred

1) I tried to add a command to create a screenshot to the beginning of the g-code using SNAP commands (alternative commands were also used, but this did not bring any result)
2) I also tried to add the command at the end of the g-code, but it also did not work
3) I used different triggers to take a screenshot before printing and after printing, including the trigger by g-code, smart g-code, smart layer, classic-layer, classic-g-code.
it all didn't work properly
I even tried to trick the program by generating a small section of the g-code (different commands were used, including g90, g91, g92 E0 + d0 15), according to which the head moves up after printing.
But this either does not cause any effect, or for some reason gives the command to render the video when the printer head has not yet parked

I did all this without stabilization, and without the M400 team

As a result:

The screenshot command for the g-code in the initial and final g-code does not work.
There is not enough trigger that will make the image using all the commands, including the g-code + layer \ height

#### What should have happened?
required in the initial and final g-code to take a screenshot by command

#### What happened instead?
none of the triggers allows this

#### Operating System running OctoPrint and Octolapse

OS Name: Raspberry Pi
Os Version: Pi 3

#### Printer model & used firmware incl. version

Printer Model: Delta
Printer Firmware Version:  repetier 1.03

#### Browser and version of browser, operating system running browser

Browser: Chrome, Opera
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2020-02-18:
So, this will be a very difficult feature to add.  I've been considering adding a feature like this for some time, but have not yet due to time constraints.

There is a really old version of octolapse out there (I think before 0.3) that allowed the classic layer and classic timer trigger to be combined with the gcode trigger, so adding this capability for those two triggers wouldn't be too bad.  With the addition of the smart triggers, this is complicated.

I will mark this as a new feature request, and will save it for after the release of 0.4.0.  Since Octolapse 0.3.4 is NOT compatible with the new release of OctoPrint, it's very important that I get this released soon.

Thanks for another good suggestion!

Goodfeat, 2020-02-19:
I will really wait for this release

FormerLurker, 2020-05-08:
Just wanted to give you an update.  I'm getting close to releasing v0.4 (trying hard as hell to get the next release candidate out).  There are a few features that will go in next, and i'm considering this one.