FormerLurker/Octolapse #448: RC3 JS error

steveheinsch, 2020-02-18:
Just upgraded to the newest RC3 (OctoPrint 1.3.12 on OctoPi 0.17). On the main octolapse tab/page, when I click the pencil icon to edit my camera settings, I now get a JS error and no popup for camera settings. All other icons correctly produce the popup.

The error:
TypeError: self.get_src is not a function - packed_plugins.js:5863:40

Using current Firefox on Mac

Also, at the bottom of the page (below About/Github/Youtube links), it says "Loading Octolapse" with an animated spinner that just sits there spinning forever.

FormerLurker, 2020-02-18:
Thanks for reporting this!  I don't have access to a Mac, but will see if i can get browserstack working.  I believe they give a free subscription for open source projects.

Here is the code that is failing:

https://github.com/FormerLurker/Octolapse/blob/d1aae6046913b249b16cbfd2ef2d6e133216d83a/octoprint_octolapse/static/js/octolapse.js#L889

Looking at it now, I'm not sure why it's working on Chrome and FF for windows.  The 'self' variable looks to be out-of-scope.

steveheinsch, 2020-02-18:
@FormerLurker  For giggles, I tried on Safari as well. Same issue. I don't allow Chrome on my machine so can't test that. I went down a just tried on my win8 box (current firefox). Same issue/error, so it doesn't seem specific to the Mac. I've cleared browser cache/forced reloading, etc.

I am a web dev, but this is my first stint into the Pi so am unfamiliar with how everything is set up and working as far as toolchains, etc.

FormerLurker, 2020-02-18:
> I am a web dev
I'm always happy to have feedback from a pro!

>  I went down a just tried on my win8 box (current firefox). Same issue/error, so it doesn't seem specific to the Mac.

That's very odd, especially since I'm using FF for windows, FF developer Edition and Chrome for testing.  Lots of ppl have been using the image preferences, which uses that failing code to load the camera stream.  I wonder what's going on!?

For now I'm going to move all of the references to 'self' (set to this in init), which should not be working anyway, outside of the  binding handler.  I'm going to push the changes to the devel branch when I'm done.  I am working today, so unless I can get it done over lunch, it will be a while until I can make the changes.

FormerLurker, 2020-02-18:
Ok, so I think I might have fixed your issues, maybe?  I deployed to the devel branch.  Except for the flask_principal issue, I wasn't able to replicate the js errors (odd).  However, I think the js error is fixed.

The flask_principal issue was tricky.  I have some conditional requirements now in setup.py that will only install flask_principal if the OctoPrint version is < 1.4.0.  Installing it in versions > 1.4.0 caused issues in that version.  I am pretty sure this will fix it.  I was able to do a fresh install and an upgrade to 1.4.0rc5, and it worked in both cases.  Fingers crossed!

Unfortunately, to properly test the flask_principal issue, you'll need to uninstall Octolapse.  First, make sure you export your settings, which will save you effort later.  Click on the gear icon within the Octolapse tab, then click *Export Settings*.  Next, uninstall flask-principal from the venv.  Finally, reinstall Octolapse from the plugin manager with this url:  https://github.com/FormerLurker/Octolapse/archive/devel.zip.  After you reboot, click the gear icon within Octolapse and import your previously exported settings.

Also, I don't think I fixed the skinning error you saw with the 'Loading Octolapse' spinner yet.  I need to get access to browserstack or a mac or something, and see what exactly is going on there :)

Let me know if I managed to fix any issues, and what is still broken.

steveheinsch, 2020-02-19:
Awesome, I'll be able to test later tonight when this print is done!

FormerLurker, 2020-02-19:
Perfect 👍 I anxiously await your results.

steveheinsch, 2020-02-19:
That did solve it, along with the spinner. I was thinking that the spinner could be the contents of the popup or something, but since it was breaking due to the other error it never got hidden/content replaced/updated. Anyway, both issues here are fixed! Thank you! :)

closed 2020-02-19 by steveheinsch