FormerLurker/Octolapse #449: Ability to set steps for slider values?

steveheinsch, 2020-02-18:
Is there a place/setting where you can set the steps for the sliders for various settings? Some, like focus, are set to 5 where I'd like to set it to 1. Is this customizable? I couldn't find anything related to it.

On the topic of sliders, I really don't like them. That's a personal thing, I know. They are very hard to just increment/decrement by the next value depending on the input device you're using. One thing I'd really appreciate is the addition of just a text box next to each parameter to quickly set a value instead of having to use the slider. Or an option to use one/the other. I realize that's a bit of work to do, but was wondering if the sliders are hard to use for others as well? It's not specific to this app...it's sliders in general, especially with a wide range of values and a fairly narrow slider, which makes it more "sensitive" and harder to be precise.

FormerLurker, 2020-02-18:
You are not the first one to ask about this.  So, Octolapse is set to only allow values that are supported by your camera.  1 is likely not a valid value for focus.  You can check this out by looking at http://{ip_of_pi}/webcam/input.json, where you'll see something like the following for the focus control:
```
{
"name": "Focus (absolute)",
"id": "10094858",
"type": "1",
"min": "0",
"max": "250",
"step": "5",
"default": "0",
"value": "0",
"dest": "0",
"flags": "16",
"group": "1"
},
```
Notice the *min* and *step* value.  According to the control definition only values of 0, 5, 10.. 250 are allowed, at least by my camera.  The behavior of a value of 1, in this case, would be undefined.

FormerLurker, 2020-02-18:
Also, you can switch to 'single column mode' at the top to make a narrow slicer long.  Others have asked about a text box.  I've considered this, and your reasoning about why I didn't include them was correct:  additional work, not enough space to fit them in :)  I may still do this, but I still would disallow values that aren't allowed by the camera (like 1 in your example).

closed 2020-02-19 by steveheinsch

FormerLurker, 2020-02-19:
Are you sure you don't want me to leave this open as a feature request?

steveheinsch, 2020-02-19:
I can if you'd like, or if it would be better I can make a new issue so it's separate from the steps question?

FormerLurker, 2020-02-19:
Yeah, now that I think of it, a new issue would be great!  I won't be working on it until v0.4 comes out, but I would like to get this done since I've gotten this question and request a few times now.