FormerLurker/Octolapse #451: Print head stops on the way back after image has been taken

nulaft, 2020-02-19:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.3.4

#### Version of OctoPrint

OctoPrint Version: 1.3.12

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: Yes

#### What were you doing when the problem occurred

1. I start a print with octolapse enabled.
2. After a layer has been printed, the print head moves to the side, as configured.
3. The camera takes an image.
4. The print head starts moving back to the printing position, but then stops for a short moment before continuing to the correct position.
5. Repeat from step 2.

#### What should have happened?
After the picture has been taken, the print head should move without hesitation back to the printing position.

#### What happened instead?
On the way to the printing position, the print head stops for a brief moment.

#### Operating System running OctoPrint and Octolapse

OS Name: Raspbian GNU/Linux
Os Version: 10 (buster)

#### Printer model & used firmware incl. version

Printer Model: Prusa i3 MK3s
Printer Firmware Version: 3.8.1-2869

#### Browser and version of browser, operating system running browser

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: https://gist.github.com/nulaft/91e41a8fc4204fcfac5aad8e40b6427d

#### Link to settings.json

Link to settings.json with all passwords removed: https://gist.github.com/nulaft/1af4a643abb46e96e924256c0f21791c

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://gist.github.com/nulaft/cd616b0a8b1e380d741d8cce826eabfd

#### Link to octoprint.log

Link to octoprint.log:  https://gist.github.com/nulaft/e47136e16cb6ad54f9d78b4abd8a0337

#### Link to contents of Javascript console in the browser

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: https://www.youtube.com/watch?v=pneZl3fv-nA

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2020-02-19:
I have an Mk2.5 and it does the exact same thing.  It doesn't really stop, but decelerates in the middle.  I looked at the log, and can see the codes for traveling to and from the stabilization point, and it's just two gcodes (to/from) separated by an M400.  I have no idea why it does that, but I suspect it's something to do with the M400.  Unfortunately, that gcode is absolutely needed to get a stable timelapse :(

Also, I just started a test to double check this behavior, and notice the same thing when my printer is traveling from the priming line to the first line of the print (thought it was much less pronounced), even though no snapshot was taken there.  Perhaps it is something odd with the firmware?  If you have any ideas, I'd LOVE to hear them!  If I could fix this, it would mean that Octolapse would add less time to the print!

FormerLurker, 2020-02-19:
This sounds similar, and it looks like the same thing that happens to me just trying a single G0 command:  https://forum.prusaprinters.org/forum/original-prusa-i3-mk2-s-others-archive/stuttering-travel-moves/

nulaft, 2020-02-21:
I've done another recording that also shows the beginning of the print and the first two minutes of printing. I've lowered the traveling speed in octolapse to 60 mm/s to see if the stuttering also happens at slow moving velocities, which it does.

https://www.youtube.com/watch?v=kj4RUyXb4rU

Meshbed leveling is done at 0:28. It then draws the first half of the priming line. Then it moves to take the first snapshot, and it stutters on the way to and from that!

After finishing the second half of the priming line (0:44), it stutters on the way to the first line of the print! Very interesting, I haven't seen that happen before. I switched off octolapse and started the print multiple times, the stuttering didn't happen anymore.

Upon arriving at the position of the first printing line at 0:46, it interrupts for the second snapshot. It stutters in both directions again.

For the travel moves for the snapshots taken between printing the layers, it's a bit random if it stutters on the way to the snapshot position, but it always stutters on the way back:
1:03, after layer 1, it stutters on both moves
1:14, after layer 2, it stutters only on the return move
1:24, after layer 3, it stutters only on the return move
1:32, after layer 4, it stutters only on the return move
1:40, after layer 5, it stutters only on the return move
1:48, after layer 6, it stutters only on the return move
1:55, after layer 7, it stutters on both moves
2:04, after layer 8, it stutters on both moves
2:12, after layer 9, it stutters only on the return move
2:20, after layer 10, it stutters only on the return move

I ran the same test multiple times. Stuttering on the way to the snapshot does not seem to depend on the layer that has just been printed, it is random if it will stutter on the way to the snapshot after a specific layer. It consistently stutters on the travel move after the snapshot.

I would be glad to help solving this issue. I currently don't have any good ideas what could be causing the stuttering. Since it happens randomly on the way to the snapshot but consistently on the way back, maybe it's something related to timing? And the M400 command synchronizes something?

FormerLurker, 2020-02-21:
So, [I made a video](https://youtu.be/2jif0zRQwC0) showing what I'm seeing while only sending gcode through the terminal.  I got stuttering from every command, sometimes only at the beginning of the movement, and sometimes both at the beginning and the end.  Since I saw this every time, and was not using Octolapse, I'm going to blame the firmware.

I do have a theory about why we both see this more often when using Octolapse than when not using Octolapse (though I have no idea why I only sometimes see it on the travel right after the prime line is laid down).  I think there is some problem with acceleration/jerk when there are no gcodes in the planner.  This would explain why we see this issue every time when Octolapse is returning from the snapshot position, and why I see it every time I send a command via the terminal.  If I send a gcode from the terminal when the printer is idle, it obviously has no gcodes in the queue.  The M400 command that Octolapse sends is used for synchronization, as you suspect. It forces the printer to reduce the gcode buffer to 0 before responding.  That's how Octolapse knows when to take a snapshot.  The actual sequence of events is this:

Octolapse decides it's time to take a snapshot, so it prevents new gcodes from your file from being sent to the printer.  It then injects some gcode to prepare for the snapshot (lift, retract, ect...).  Next, it sends the travel command to move to the snapshot position, followed by an M400 and an M114 command.  The M400 causes the printer to stop responding to gcode commands until the buffer is empty and all moves have completed.  Once this happens, the printer responds with its current position (from the M114), which tells Octolapse it's time to take a snapshot.  After the snapshot is finished, octolapse sends another chunk of gcode.  It sends a travel command back to the starting position, and undoes all of the other things (deretract, delift, revert axis modes and speed to what they were before Octolapse started the snapshot process).

I think the M400 command is causing the printer to act just like it does in the video where I sent commands from the terminal.  For whatever reason the first travel seems to be very jerky when the machine is idle and the buffer is clear.

I may put a gcode file together to test all of these behaviors both with and without gcodes in the buffer.  This would be a pretty good test of my hypothesis.  Something like this might work:
```
G28 X Y; home x and y
G1 X0 Y0; move to origin
; test with some gcodes in the buffer by sending a series of fast travel movements in 10mm increments
; in theory the printer should accelerate smoothly from the x=0 to x=250
M117 Test buffered travel
G1 X10 
G1 X20 
G1 X30 
G1 X40 
G1 X50 
G1 X60
G1 X70 
G1 X80 
G1 X90 
G1 X100
G1 X100
G1 X100
G1 X100
G1 X100
G1 X100
G1 X100
G1 X100
G1 X100
G1 X100
G1 X200
G1 X200
G1 X200
G1 X200
G1 X200
```

FormerLurker, 2020-02-21:
OK, I created [some gcode](https://gist.github.com/FormerLurker/9773e2fbcc9dfdb275a8f091a2d1c900#file-uneventravelaccelerationtest-gcode) to test this out, and I get way different results.  The buffering travel definitely causes some stuttering, but not too much.  Sending a single travel command is much smoother.

I'm going to try adding some pauses and see if that changes things.

https://gist.github.com/FormerLurker/9773e2fbcc9dfdb275a8f091a2d1c900#file-uneventravelaccelerationtest-gcode

nulaft, 2020-02-22:
I replicated your tests on my printer:
https://www.youtube.com/watch?v=hlXVmhO4aQU

When I send gcode g1 commands through the terminal, I do not get any stuttering.

I also ran your buffering test gcode, and I get stuttering on the buffered moves but not in the single travel command moves.

My conclusion from these tests is that (at least for these test cases) my printer behaves exactly like it should. There is no stuttering on single travel commands. The small amount of stuttering at the start of the buffered travel commands could happen because the second travel command is not yet fully processed while the print head has to decellerate because the first travel command is halfway executed already. How many gcode commands can the printer buffer? Is there a way to tell the printer to "listen and think until the buffer is full, then execute"?

FormerLurker, 2020-02-22:
>I replicated your tests on my printer:

It doesn't look like your extruder is moving nearly as fast as mine, despite you having set the correct feedrate.  Perhaps something is limiting the maximum speed, like M220 Sxxx (control feedrate position) or adjustments via your printer's LCD screen?  When I ran my tests, the acceleration looked better the slower the extruder was moving, so you might want to re-watch my video and compare the extruder speeds to make sure yours is not being reduced.

> Is there a way to tell the printer to "listen and think until the buffer is full, then execute"?

Not that I'm aware of, though I don't know why it would help since we're both seeing delays on the acceleration as the plan is built up.

Additionally, I added M400 commands in strategic places within the gcode you tried, and it didn't seem to have any effect.  I'm not sure what I can do about this, since we still haven't even figured out what causes this.  However, I'm pretty sure this is not an Octolapse issue, and that it is a Prusa Firmware issue.  It doesn't seem to be causing any problems, other than taking a bit longer than it should to travel back from the stabilization position.  That being said, I believe it's worth looking into this issue a bit more.  Perhaps you can add an issue to [prusa firmware repository](https://github.com/prusa3d/Prusa-Firmware)?  If I were you I wouldn't mention Octolapse, and would only show them the uneven acceleration that can be seen just sending commands from the terminal.  They understandably are not interested in solving issues that might be caused by someone else's software :)  What do you think about this?

nulaft, 2020-02-22:
I've done the same test again (sending single travel commands from the terminal), this time the extruder was moving at the correct speed. I'm not sure what was limiting the speed last time, but it doesn't really matter. There was still no stuttering. I can not create stuttering with sending a single travel command from the terminal. There seems to be a bug in the firmware of your printer that causes this. This issue would be easy to show prusa and they should really fix that!

We still need to find a way to induce the stuttering that happens with octolapse on my printer, without using octolapse. Could you create a minimalistic gcode file that does exactly what octolapse does? We should be able to get the stuttering with that, right? We could then see if the same happens if the gcode is run from an SD card. And we could try to simplify the gcode to have the minimal amount of commands while still getting the stuttering. If we have that, we can add an issue to the prusa firmware repo.

FormerLurker, 2020-02-23:
> Could you create a minimalistic gcode file that does exactly what octolapse does?

I will do this.  Octolapse can log all of the gcode produced during the stabilization by enabling the *octolapse.stabilization_gcode* logger.  I will create a small demo as soon as I have a bit of time.

nulaft, 2020-02-26:
Thank you, looking forward to that!

nulaft, 2020-03-01:
I've experimented a bit more with sending gcode commands from the terminal. I've used pronterface, just to be sure it's not octoprint.

I have this very simple gcode sequence:
G28 W
G80
G1 X240 F10800
G1 X10 F10800
M84

If I send these commands one after another through the terminal, the printer will stutter reliably on both G1 moves. It will also stutter on other G1 moves, such as moving y or z axis.
However, if I leave out the G80  (meshbed leveling) command, the G1 moves will not stutter!

I created two gcode files, one with and the other without meshbed leveling. I put them on the SD card and ran them directly on the printer. The results are almost the same:
The one with meshbed leveling has stuttering on the first G1 move, but not on the second.
Without meshbed leveling there is no stuttering.

Now we have a simple setup to reproduce the stuttering. The setup is rather different from what is happening when using octolapse. Maybe the root cause for the stuttering is the same in both cases.

FormerLurker, 2020-03-01:
Fantastic!  Great work, thank you.  Been playing around with this for the last hour and had no luck.  I will test your code and see what happens.  Link to this issue if you creat an issue on the prusa firmware github please.

nulaft, 2020-03-02:
I was about to create a new issure on the prusa firmware github. They asked me to search for similar issues first, so I did. I found this: https://github.com/prusa3d/Prusa-Firmware/issues/1406
OP was having a different issue, stuttering during lots of fast and short movements, but quickly the focus is shifted to exactly the problem I'm on about. They already found out that it has to do with meshbed leveling and the move command queue being empty. Currently they are waiting for some other changes to be merged before addressing this issue.

FormerLurker, 2020-03-02:
Yes, looks like the same issue.  I too looked at their issues, but wasn't able to find that thread.  I guess we just need to wait for this to be fixed.

So, should I consider this issue closed since we've verified it is a prusa firmware issue?

nulaft, 2020-03-02:
Yes!

closed 2020-03-02 by nulaft