FormerLurker/Octolapse #454: [Request] Octolapse wait to start time

franferri, 2020-02-22:
#### If this is a feature request describe it here
I have octolapse with the default plugin settings, the relevant configuration for this request is:

* Stabilization: Animated - Orbit
* snapshot: Layer Change - Compatibility
* Rendering: MP4 Fixed length 00:10

My printer is Prusa MK3S

When the printer starts printing, it draws a first line in the bed to stabilize the nozzle pressure, is the very first thing the printer prints before each print.

Since I have enabled octolapse, with the previous setings, the stabiliztion line stops at the middle, moves the header to the center, to tacke the "empty snapshot" and then continues with the stabilization line.

This is very very wrong, the stabilization line must be printed at once.

Having a way of telling to octoprint to skip that first layer snapshot with a timer (like first 10 seconds dont do anything) or just a check to (Skip during nozzle stabilization) will be great.

I know I can do this in my gcode but I will rather have it as an option.

#### Version of Octolapse
Octolapse Version: 0.3.4

#### Version of OctoPrint
OctoPrint Version: 1.3.12 running on OctoPi 0.17.0

FormerLurker, 2020-02-22:
So, Octolapse v0.3.4 (your version) has a feature you can use to prevent this.  It's called position restrictions and they can be found in the *snapshot profile*.  You can add a *rectangle* that *must be inside* with the following values:

```
X - 0 to 250
Y - 0 to 210
```
That will prevent any snapshots on the priming line.

Octolapse V0.4+ (currently in dev, but it's getting close to release) has two additional features that make this even easier.  Within the printer profile you can enable an option called *restrict snapshot area* that is pre-configured for your printer when you select the proper make/model:

[image]

Finally I've added ```@OCTOLAPSE``` commands that can be used to turn snapshots on and off.  This has the added benefit of being able to prevent snapshots during the end gcode.  Here is how the two command (start and stop snapshots) would be used in your start and end gcode:

```
; start gcode
@OCTOLAPSE STOP-SNAPSHOTS ; Don't take any snapshot during the start gcode
; other start gcode here
...
; end start gcode
@OCTOLAPSE START-SNAPSHOTS; Start taking snapshots now
; model printing gcodes
...
; end gcodes
@OCTOLAPSE STOP-SNAPSHOTS
```

So, take your pick!  However, if you want to try the dev version out I'd be appreciative.  The more people who test it the fewer issues I'll discover on release day!  You can upgrade within the plugin manager by clicking *get more* and entering the following url into the *from url...* box:  https://github.com/FormerLurker/Octolapse/archive/devel.zip

Be sure to read [this guide](https://github.com/FormerLurker/Octolapse/wiki/V0.4-Getting-Started-with-Octolapse) all the way through before you attempt to upgrade.  Pay special attention to the section on automatic slicer settings detection and verbose gcode, depending on which slicer you are using.

franferri, 2020-02-23:
Testing now for my version the restriction, many thanks!

FormerLurker, 2020-02-26:
@franferri, did the position restrictions work for you?  I'm going to mark this issue as waiting for response, and will give you some time to respond.  Thanks!

franferri, 2020-03-01:
I'll try tomorrow to do a timelapse and confirm

FormerLurker, 2020-03-26:
Have you been able to try this out?

franferri, 2020-03-26:
yes, and the camera went on on the right rectangle constrains but  the
movements of the extruder were the same. So not really fixing the issue.

FormerLurker, 2020-03-26:
Maybe it is time to try v0.4.0rc1?  The *restrict snapshot area* setting should do the trick.  I will help you setup and configure it if you wish.  Let me know, and thanks for responding!

franferri, 2020-03-27:
Due corona i ran out of pla. Ill try when i get it