FormerLurker/Octolapse #455: Feedback for V0.4.0rc1.dev4

FormerLurker, 2020-02-22:
Please see the release notes [here](https://github.com/FormerLurker/Octolapse/releases/tag/0.4.0rc1.dev4). This includes a list of enhancements, bug fixes, and issues known at the time of release.

All feedback is welcome here!  Let me know what you like, what you hate, what works, what is broken, etc. 

THANK YOU for helping me iron out all of the kinks and to get v0.4.0 out the door!

gege2b, 2020-02-25:
hi there
great progress !

I didn't test dev4 deeply enough yep, but my first encountered problem is related to the timelapse folder

As a timelapse is finished rendering, the file appears in octolapse's own timelapse browser, but not in the octoprint timelapse tab, I had to ctrl+F5 for it to appears

I don't really know if it's related to dev4, but it doesn't behave like this before

simpat1zq, 2020-02-25:
I think I've got a problem that started with dev3, and is still there with dev4. I am not using stabilization. I'm using Smart Snap to Print - Smooth. I just disabled OL, and did a print, and the print went off just fine. I re-enabled the plugin, and as soon as I start the print, I get the following error, and the print goes into 'cancelling' state, it runs the first few commands of the print, and then officially fails:
`Octolapse Startup Failed
Manual Slicer Settings Missing
Your printer profile is missing some required slicer settings. Either enter the settings in your printer profile, or switch to 'automatic' slicer settings. Missing Settings: Is Vase Mode`

I don't know how OL causes a print to cancel, but I literally did not change anything else between the first print and second, except for the fact that OL was disabled for the first print. 

What should I try? Please let me know if you need anything more than that error message and I can get it for you.

simpat1zq, 2020-02-25:
After looking at that error a little closer, I decided to go into the printer profile settings page, and make a change to save it and then put the change back to save it again. That did something, because I don't get that same error any more. I get this one:

`Octolapse Startup Failed
Printer Not Supported
Your printer does not support the M114 command, and is incompatible with Octolapse`

I have an Ender 5 with an SKR Mini E3 board. Not sure what to do about this error.

FormerLurker, 2020-02-25:
So, two things:

1.  Export your settings.json file by clicking the gear icon in the upper left hand corner of the Octolapse tab, then clicking 'export settings'.  Then paste the file into gist.github.com and paste a link in here so I can review the settings.  
2.  Edit your printer profile and save it.  That should create the missing setting.

It looks like another settings migration bug for Cura.  I've had a lot of difficulties getting that right.  You could follow the suggestion from the error and setup automatic slicer configuration.  It's awesome, and will save you tons of time.  Click the help link next to the 'slicer type' for more info.

FormerLurker, 2020-02-25:
@simpat1zq, lol!  Jinx :)  Looks like our posts crossed paths.  And, it also looks like you solved the issue in the way I'd expected.  Perhaps you can do me a solid and download this file from you pi (you can use any sftp client, like filezilla, to connect): /home/pi/.octoprint/data/octolapse/settings_backup{some version here}.json

Not sure what version you upgraded from, but when in doubt, send all of the backups.  I can run these through the debugger, which will let me replicate and fix the issue.

Thanks!

simpat1zq, 2020-02-25:
I change the name of the file to .txt because GitHub doesn't do .json files apparently. 
And as you just post, yes, our paths just crossed :)

Also, just to be clear, I still can't use OL. It causes the print to fail. It says because of the M114 command. But with Snap to Print, should it even be using that?

[settings.txt](https://github.com/FormerLurker/Octolapse/files/4251787/settings.txt)

FormerLurker, 2020-02-25:
@simpat1zq, yes, M114 is required to work for all stabilizations.  Can you go into the OctoPrint terminal, send M114, and let me know how your printer responds please?  It is entirely possible that the check for M114 isn't working 100% (it is new).  I encountered some printers that did not support either M114 OR M400 recently, and added that test.

Another question.  When it says there is a problem with M400, is there a help button on the popup?  That is a new feature too.  I'm trying to add specific help popups for every common error. but I have to add them one at a a time for various reasons (the overwhelming effort, lol).

Meanwhile I'll try to fix the settings migration issue for the 'vase mode' option (should be false after migration, not null).

FormerLurker, 2020-02-25:
@simpat1zq, Can you also upload the settings file that looks like this ```settings_backup_0.4.0rc1.dev3.json``` (it could have a different version number)?  You'll have to get it off of your pi, which can be done many ways (my favorite is an SFTP client like FileZilla).  It will be in this directory:  you can use any sftp client, like filezilla, to connect): ```/home/pi/.octoprint/data/octolapse/```

Also, I will be creating an issue for the cura settings migration issue, and depending on how your printer responds to the M114 command, I may create one for that as well.  I will edit this comment, and will link to those issues if/when they are created.  That will keep this thread clean.

Thanks!

simpat1zq, 2020-02-25:
`Send: N6 M114*33
Recv: X:220.00 Y:220.00 Z:10.00 E:-1.00 Count X:17600 Y:17600 Z:16000
Recv: ok`

`Send: N7 M400*32
Recv: ok`

Yes. There was a help button. I don't remember the exact wording, but it basically said to flash firmware that supports m114. My firmware already does though. It's the latest Marlin. Just a shot in the dark, but I'm wondering if your code is not liking that '-1.00' for the extruder. I do a retract after a print is finished, so that might be why it's negative. 

Still working on getting the backup settings files off the box.

FormerLurker, 2020-02-25:
@simpat1zq , ok, I'm creating an issue for both of your problems.  That response looks perfectly OK to me, so it must be a problem with the test itself somehow.

simpat1zq, 2020-02-25:
Sounds good. Thanks for the help with this. I've attached my settings backups in case you need those. 
[settings.zip](https://github.com/FormerLurker/Octolapse/files/4251871/settings.zip)

FormerLurker, 2020-02-25:
Ok @simpat1zq, see #458 and #459.  We can debug this issue there.  I believe I have enough to fix the settings migration issue.  I will need some additional information to solve the M114 test issue.  I'm going to make an attempt to fix it, but if I can't I'll just remove the test.  I added it because that problem is particularly difficult to debug (just an OK response to M114, no position at all), and would much rather have that there than not.

ciordia9, 2020-02-25:
Is the preview window supposed to be dark until you click on it?
[image]

FormerLurker, 2020-02-25:
@ciordia9, no it is not.  It looks like the CSS is messed up somehow. Maybe whatever plugin you are using to use icons for the tab?  Can you try disabling that and see if your Octolapse tab looks more like this:

[image]

ciordia9, 2020-02-26:
> @ciordia9, no it is not. It looks like the CSS is messed up somehow. Maybe whatever plugin you are using to use icons for the tab? Can you try disabling that and see if your Octolapse tab looks more like this:

Removed the tab plugin and instead of a square in the middle it's now the enlarge symbol. 

Went back and pulled out a few more things--I was impatient so I popped out anything that could add controls to the UI.

- Fullscreen Plugin
- Resource Monitor
- Tempsgraph Plugin
- LayerDisplay

I'll circle back and add one at a time and see what triggers the bad behavior. My gut for some reason is betting its the fullscreen one.

FormerLurker, 2020-02-26:
Thanks!  I know that takes forever.  Once you figure out which one is causing it, I will install it and will try my best to fix it, either via a pull request, or by changing Octolapse code (probably the latter)

ciordia9, 2020-02-26:
Decided to go ahead and do it. Activated everything back one by one leaving the **Fullscreen** one for last. It causes you problems. I don't need it or use it though so I've delete/cleaned it. They never got out of a decimal release so it might just be handling things really poorly. I've noticed quite a speedup on page rendering since it's gone.

FormerLurker, 2020-02-26:
Thanks for sticking it out!  I had full screen until recently (just didn't reinstall), so I will take a look.  Octolapse is still in decimal release too, lol!

FormerLurker, 2020-02-26:
@ciordia9, I've created issue #460 to deal with the fullscreen webcam plugin.  Thanks for the report!

FormerLurker, 2020-03-07:
@gege2b, I believe issue #468 should fix your problem with the timelapse tab.  Note that the two tabs will not sync if you delete files as of now, though I will probably make it work if files are deleted through Octolapse.  Making it work the other way (when files are deleted from the OctoPrint timelapse tab) will be much more difficult, and might require changes to the stock timelapse plugin.

FormerLurker, 2020-03-07:
Thanks to everyone who posted in here!  I've just released V0.4.0rc1.dev5, which includes bugfixes to the vast majority of issues I've heard about in this version.  There are still some unresolved issues, but since OctoPrint v1.4.0 dropped, it's extremely urgent that I get the next version of Octolapse out there.  Octolapse v0.3.4 will stop working if you upgrade, so time is running out.

Please see [this issue](https://github.com/FormerLurker/Octolapse/issues/470) to report feedback regarding the next release.

Thank you all so much!

closed 2020-03-07 by FormerLurker

gege2b, 2020-03-07:
thx @FormerLurker , I would like to try this out next week, but unfortunately, both my printer's PSU and his replacement blew up yesterday... (maybe Murphy is angry about me)