FormerLurker/Octolapse #456: Video not playing after render

iskyguy, 2020-02-23:
#### If this is a feature request describe it here

#### Version of Octolapse
 
Octolapse Version:   v0.4.0rc1.dev4

#### Version of OctoPrint

OctoPrint Version: 1.3.12 

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: yes

#### What were you doing when the problem occurred

1. Everything is working
2. print finished 
3. rendered 
4. downloaded but cant open file, issue is that it is in a non supported format

#### What should have happened?
video opens and plays normally

#### What happened instead?
movies and tv errors out and cant view video even though it is an mp4 file

#### Operating System running OctoPrint and Octolapse

OS Name:OctoPi
Os Version: 0.17.0

#### Printer model & used firmware incl. version

Printer Model: Prusa I3 mk3 
Printer Firmware Version: 3.8.1

#### Browser and version of browser, operating system running browser

Browser: 80.0.3987.116
Browser OS: Chrome

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: https://drive.google.com/file/d/1wS-opXwdQivfZS1L4jgs0dtoKPiKG2z9/view?usp=sharing

#### Link to settings.json

Link to settings.json with all passwords removed: 

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log: 

#### Link to octoprint.log

Link to octoprint.log:  

#### Link to contents of Javascript console in the browser

Link to javascript console output:

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: https://drive.google.com/file/d/1806FRUfjyNQvKxrUNckfwSGTZgovIOZ7/view?usp=sharing
https://drive.google.com/file/d/1cL9DIYowTGmuTqYBHIIPLbRW9mNBjNXJ/view?usp=sharing

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2020-02-23:
Hey, sorry I didn't respond yet to this issue (working my way towards it). I saw you just closed it, and am wondering if you solved the problem.  If so, what did you do?  Your explanation may prove useful to me or others who search the issues.

iskyguy, 2020-02-23:
Yeah, its fine I figured it out after posting issue. I had set my dslr to shoot raw (6000x4000) and the interface with octoprint can read the raw image but movies and TV, vlc, or windows media player cant. I imported the file into primere and discovered that i had been shooting it in raw, simple problem.

FormerLurker, 2020-02-23:
This actually may be a bug.  I added you image conversion to dev3, so unless Pillow isn't correctly determining the file type, it should have worked.  However, I believe that resolution may not be supported fully by the encoder.

Would you be willing to run a test mode print with raw files, and enable snapshot archiving (either disable rendering or enable archiving in the rendering profile), stop the print after 5 or so layers, and upload the archive file (videos and images button->archive tab) so I can run it through the debugger?

iskyguy, 2020-02-23:
ok here is the archive. 

https://drive.google.com/file/d/1q8yaNqYb2QwXnGsFcJ8W6U5s8c-bQD3J/view?usp=sharing

FormerLurker, 2020-02-23:
You are awesome!  Will run this through the debugger later tonight or tomorrow.

iskyguy, 2020-02-23:
awesome, let me know what happens!

FormerLurker, 2020-02-26:
So, those images ARE jpegs, but they are special lossless jpegs.  The video I generated played just fine in VLC, but not in some other players (which was I think similar to your experience).  I will see what I can do to fix this, either via the rendering process, or through a conversion process.

iskyguy, 2020-02-26:
OK thanks, I have started having issues even being able to get the rendered file. when i download the archive to put it into Premier for a manual time lapse, windows cant open the zipped file. The rendered file Octolapse produces also isn't readable by anything I have. 1080 photos work.

iskyguy, 2020-02-26:
my bad, the working resolution is 3984x2656

iskyguy, 2020-02-26:
ok, i got the render working in vlc, but the zip archive download still isn't working.

closed 2020-02-27 by iskyguy

Apexpredation, 2021-09-25:
I don't know if this is allowed, but I'm jumping in here to say i just started having this exact problem. Video will render but not be playable. I found that I can upload to .MP4 to .gif converter sites and it will make a gif, so the frames are there. It also fails to render in 60fps only 30 fps works.

aw-jansen, 2023-02-28:
@gavmeister did you find a solution for this?

gavmeister, 2023-03-02:
@aw-jansen it was browser dependent. For example it played if I browsed on my iPhone but not with windows. 
anyway I’ve moved to Klipper and mainsail on the raspberry pi so not using octoprint or octolapse.  
you should try it.

aw-jansen, 2023-03-02:
Thanks for getting back to me. Much appreciated. Def going to check out Klipper and mainsail though I see support for my Sovol sv06 is limited.