FormerLurker/Octolapse #457: Don't use absolute paths

EleRas, 2020-02-24:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: v0.4.0rc1.dev4~f98c787

#### Version of OctoPrint

OctoPrint Version: 0.3.12

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: No (not relevant)

#### What were you doing when the problem occurred

1. Configured haproxy to serve OctoPrint via <domain>/printer1/, <domain>/printer2/, ... due to multiple OctoPrint - instances running on the same machine.
2. Trying to download a timelapse fails with 400 Bad request, since octolapse tries to download from a hardcoded path (http://<domain>/plugin/octolapse/downloadFile?type=timelapse_octolapse&name=<filename>.mp4). This path should be relative to the installation (http://<domain>/printer1/plugin/octolapse/downloadFile?type=timelapse_octolapse&name=<filename>.mp4
2a. Any help - icon fails with 400 Bad Request as well
3. Downloading the file via the normal timelapse - list in OctoPrint works.

#### What should have happened?
Timelapse download / help icons should work

#### What happened instead?
400 Bad request, see above description

#### Operating System running OctoPrint and Octolapse

OS Name: Modified OctoPi 0.17.0
Os Version: 0.17.0

#### Printer model & used firmware incl. version

Printer Model: PRINTER_MODEL_GOES_HERE
Printer Firmware Version: PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2020-02-24:
I must have missed that one.  I implemented new tornado routes for all downloads to improve performance, and must have goofed.  Will look into it.

FormerLurker, 2020-02-24:
Ok, I fixed them all, and found one other place where I missed the leading period.  Whoops :)

Try updating via the plugin manager using this URL in the *from url,...* box: https://github.com/FormerLurker/Octolapse/archive/devel.zip

Thanks for catching this!

EleRas, 2020-02-26:
Sorry for the late response. I tested this with v0.4.0rc1.dev4~ca1352b and downloading the timelapses as well as the help - buttons work again, thank for the fast fix!

closed 2020-02-26 by EleRas