FormerLurker/Octolapse #462: Prusa Mini extruder stops extruding after the first layer with Octolapse turned on

BradMcGonigle, 2020-02-26:
#### Version of Octolapse
Octolapse Version: 0.3.4

#### Version of OctoPrint
OctoPrint Version: 1.3.12

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: No

#### What were you doing when the problem occurred

1. Start print with Octolapse on
2. Print starts and the first layer prints perfectly
3. After the first layer, the extruder moves to the snapshot position and when it begins printing again, filament is NOT being extruded.

#### What should have happened?
The extruder should  extrude filament as usual after the first layer.

#### What happened instead?
After the first layer, the extruder moves to the snapshot position and when it begins printing again, filament is NOT being extruded.

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: 0.17.0

#### Printer model & used firmware incl. version

Printer Model: Prusa Mini
Printer Firmware Version: 4.0.3

#### Browser and version of browser, operating system running browser

Browser: Safari 13.0.5
Browser OS: macOS

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: https://gist.github.com/BradMcGonigle/4606cf689c1b41cc54ef92feb98049e4#file-3dbenchy_0-2mm_pla_mini_1h31m-gcode

#### Link to settings.json

Link to settings.json with all passwords removed: https://gist.github.com/BradMcGonigle/4606cf689c1b41cc54ef92feb98049e4#file-settings-json

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://gist.github.com/BradMcGonigle/4606cf689c1b41cc54ef92feb98049e4#file-plugin_octolapse-log

#### Link to octoprint.log

Link to octoprint.log:  https://gist.github.com/BradMcGonigle/4606cf689c1b41cc54ef92feb98049e4#file-octoprint-log

FormerLurker, 2020-02-26:
Hi, thanks for the report!  I've been waiting for someone with a prusa mini to post for a while now.  I will review the issue after work, and will get back to you.

FormerLurker, 2020-02-26:
So, apparently gist cut your gcode off, and doesn't seem to include the other files.  Can you try this:

1. Truncate all but the first two layers from your gcode file, but include your end gcode and all of the comments at the end of the file.  Upload this as a separate gist.
2. Re-upload your settings.json file as a separate file.
3. Read [this guide](https://github.com/FormerLurker/Octolapse/wiki/V0.4-Getting-Started-with-Octolapse) for installing and configuring the pre-release version of Octolapse.  Pay extra attention to the bits about PrusaSlicer (enable verbose gcode).
4. Then, install the latest pre-release version of Octolaspe by using the plugin manager and pasting this into the *from url...* box after clicking *get more*:  https://github.com/FormerLurker/Octolapse/archive/devel.zip

Let me know if you have any issues following the above steps.  Once you have the new version installed, you might want to test out my hunch about G90/G91 by changing this setting in your printer profile:  

[image]

You might not have heard about my hunch yet since my comment on the Prusa forum is *awaiting moderation*.

BradMcGonigle, 2020-02-28:
So sorry for the delay but I install the 0.4.0 pre-release version, enabled Verbose GCode in PrusaSlicer and set G90/G91 Influences Extruder to `True` and viola - it works perfect on the Mini now!

I'm happy to still provide the GCode and `settings.json` if needed.  Just let me know.

BTW, 0.4.0 is amazing for me so far.  I love the Camera Image Settings modal with the live preview.  It makes fine-tuning the camera settings so easy!  Amazing work!

FormerLurker, 2020-02-28:
Click on the 'printers' link on the Octolapse tab, and then export the mini profile (you can export/import individual profiles now) and I will add it to the repository so you can test it out.

Thank you so much!

BradMcGonigle, 2020-02-28:
No problem at all.  I created a gist with the file.

https://gist.github.com/BradMcGonigle/0902a38bb3d494cbd7233616ee0d6cde

Thanks for your help again and let me know if you need me to test anything regarding the Mini for you.  I'm happy to help out.

FormerLurker, 2020-02-29:
Hi @BradMcGonigle!  I am looking at your settings and notice that the build volume was set to use your Octoprint printer volume.  For the profile library, I always want to set these values explicitly.  I'm wondering if you can check your octoprint printer profile and verify that what I have come up with is accurate:

[image]

I was looking through the firmware source for the mini, and can't for the life of me find where the build volume is defined.  I see a manual X homing position of 180.4, so I may be off somewhat.  Hopefully we can get this figured out.

I also restricted the snapshot area to prevent snapshots when the printer is priming.  I just set all axes to 0-180mm, which should be correct.

I've already pushed a beta profile (v0.1).  To see it, you'll need to click on the gear icon in the octolapse tab:

[image]

then click on the *Update Profiles* button:

[image]

FYI, I have set the profile to use automatic slicer configuration.  I recommend you enable [verbose gcode comments](https://github.com/FormerLurker/Octolapse/wiki/V0.4-Slic3r,-Slic3r-PE-and-PrusaSlicer) within PrusaSlicer (or Slic3rPE if you are using that) so that the new smart triggers will reduce artifacts automatically.  If you are using a different slicer, let me know.

Thanks!

jettoblack, 2020-03-04:
Hi, by the way, this same problem is also affecting the Prusa MK3(S) printer when using firmware 3.9.0-RC1, because Prusa changed the G90/G91 gcode parsing to affect the extruder absolute/relative positioning mode which did not occur in previous firmware versions. (However apparently this is correct behavior according to the Marlin description of these gcodes).

I have Octolapse 0.3.4 and changing the "G90/G91 Influences Extruder" setting to TRUE did NOT fix the problem for me. Do I need to update to Octolapse 0.4.0 to get this setting to work correctly?

Attached is a serial log of the start of a print with Prusa firmware 3.9.0-RC1 and Octolapse 0.3.4 with this setting set to TRUE. I see that G90/G91 commands are issued by Octolapse, however it is not sending a subsequent M83 to restore the extruder mode, so nothing is extruded after the first snapshot is taken.

[octolapse 0.3.4 prusa fw 390rc1 extruder mode problem serial.log](https://github.com/FormerLurker/Octolapse/files/4288151/octolapse.0.3.4.prusa.fw.390rc1.extruder.mode.problem.serial.log)

FormerLurker, 2020-03-04:
Ug...  Why would they make such a fundamental change in the way gcode is interpreted?  This is going to cause all kinds of havoc with start/end gcode, OctoPrint installations, plugins, old gcode, etc...  Plus, it now means that my pre-configured prusa profiles will work for SOME prusa printers and will totally break others.  I'm not willing to add a firmware version drop down as it would explode the number of profiles I have to maintain (though at some point I may be forced to...).
/endrant

Anyway, yes, you should upgrade to the pre-release due to a bug in handling G90/G91 influences extruder in some situations (relative extrusion + g90/g91 influences extruder) depending on your slicer and slicer config.  I believe the pre-release (since dev3) handles this correctly.

There are instructions in a previous comment, but make sure after you select the MK3 that you change the G90/G91 influences extruder setting to 'true'.  Once that firmware drops, if the behavior doesn't change back to the old way, I'll update all prusa profiles.  It's unfortunate because that will 100% break all prints for people who don't upgrade their firmware right away :(

FormerLurker, 2020-05-08:
So, this issue has been resolved.  The default G90/G91 behavior has been restored for Mk2 and Mk3, and there is now a prusa mini profile with the correct settings.  Excellent :)

Closed!

closed 2020-05-08 by FormerLurker