FormerLurker/Octolapse #463: Printer Not Supported - M114

SamKemp, 2020-02-27:
#### Version of Octolapse
 
Octolapse Version: v0.4.0rc1.dev4

#### Version of OctoPrint

OctoPrint Version: 1.4.0rc6

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES_OR_NO

#### What were you doing when the problem occurred

1. Installed OctoLapse plugin
2. Setup Octolapse for my printer... setting Printer, Stabilization, etc
3. Clicked Print

#### What should have happened?
Print should of started

#### What happened instead?
Error, Printer not supported
[image]

#### Operating System running OctoPrint and Octolapse

OS Name: Linux, Raspbin
Os Version: OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: Creality - Ender 3 Pro
Printer Firmware Version: Marlin 2.0.3_C 

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: [CE3_RING_OCTOLAPSE.gcode.txt](https://github.com/FormerLurker/Octolapse/files/4261339/CE3_RING_OCTOLAPSE.gcode.txt)

#### Link to settings.json

Link to settings.json with all passwords removed: [settings.json.txt](https://github.com/FormerLurker/Octolapse/files/4261345/settings.json.txt)

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  [plugin_octolapse.log](https://github.com/FormerLurker/Octolapse/files/4261348/plugin_octolapse.log)

#### Link to octoprint.log

Link to octoprint.log:  [octoprint.log](https://github.com/FormerLurker/Octolapse/files/4261351/octoprint.log)

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: 
[image]
[image]

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2020-02-27:
Can you try installing from the devel branch, i may have fixed this already.

SamKemp, 2020-02-27:
That seems to have worked... I have only just started my first print after updating

I did have an issue prior regarding Octolapse not saving my timelapse, I had reinstalled Octolapse to try and solve that issue and thats when I came across this M114 error. Will let you know how both these issues pan out

SamKemp, 2020-02-27:
Thank you so much for the help. My Octolapse is working great now.

No worries on the Dropbox plugin, my flatmate saw the plugin and wanted it to work with Octolapse so I made my quick and dirty fork but its great that we have a good solution

closed 2020-02-27 by SamKemp

FormerLurker, 2020-02-27:
Fantastic!  I will push a new rc this weekend.

Have your flatmate respond to the Dropbox plugin issue.  I am guessing he/she was the OP?

SamKemp, 2020-02-27:
No, my flatmate is @TWIISTED-SHOPS , I had asked him to test your version

FormerLurker, 2020-02-27:
Ok, great!  Tell him I said hi and thanks!