FormerLurker/Octolapse #464: Zip Archive Download Problems

FormerLurker, 2020-02-27:
User @iskyguy reported issues downloading zip archives of snapshot images.  Anyone else having problems?

@iskyguy, can you give me some details, like how large is the zip, and can you manually download it via sftp from /home/pi/.octoprint/data/octolapse/sanapshot_archive (unless you changed the folder in the main settings)?  I'm wondering if this is an issue with the creation of the zip file, or with the download.

iskyguy, 2020-02-29:
I can download it fine onto my computer but it seems like the bigger zipped files are having trouble being opened by windows. It also gives me a progess attempt error after a second file download attempt. The first attempt downloads fine but i have to reset the Raspberry pi to try again, even when i confirmed that nothing else was occuring.  anything over 2.0 Gb doesnt download with the progress event error.

iskyguy, 2020-03-01:
now having issues with downloading files at all, no archive or renders. ill test another pi tomorrow and do some other troubleshooting.

FormerLurker, 2020-03-01:
I've added a patch to the devel branch to enable zip64.  My pi ran out of disk space while I was trying to create a huge zip file, lol.  Was at 24GB when it died, but I was able to retrieve the images via sftp and delete them.  I just ordered a massive SD card so hopefully that doesn't happen again.

Anyway, check your disk space on your pi, try upgrading from the devel branch (https://github.com/FormerLurker/Octolapse/archive/devel.zip), and see if that helps.

Regarding the downloads, not sure what's going on there.  Set your logging profile to *Debug* and try some downloads.  You can clear the Octolapse log file within the logging profile:

[image]

Also check the octoprint.log file and see if any errors appear there.

Let me know what you find!  As soon as we get this worked out I plan to push a new devel rc.

iskyguy, 2020-03-02:
Regarding the download i can download any lossless render below about 30 MB. Ill do some testing at 2.5k and see what happens there, anything above about 30 MB With the lossless image gives the error [object: eventProgress]. Sorry for the delays in my replies, schools ramping up for finals and dont have much time for camera/printer testing.

iskyguy, 2020-03-24:
Now when i try to render something it doesnt render giving me the
 error code -9, it has something to do with ffmpeg. i made sure it was updated and then restarted.
[error help github.txt](https://github.com/FormerLurker/Octolapse/files/4376730/error.help.github.txt)

FormerLurker, 2020-05-08:
@iskyguy, so sorry it's been so long.  So, I think I fixed large file download (it was a bear), but upload is broken for anything over 1GB, and won't be fixed any time soon due to the complexity involved.

Regarding the rendering error, I usually see that with the H264/H265 codec.  If you have a pi4 you could probably use those profiles, but the Pi3 just doesn't have the mem.

Let me know if you are still having problems, or if it has resolved it self.  If it hasn't, you might want to try upgrading to the code in the devel branch.  

Thanks!