FormerLurker/Octolapse #467: How to deal with some starting gcode that wipes nozzle, and includes a z raise?

ForceConstant, 2020-03-05:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: Latest

#### Version of OctoPrint

OctoPrint Version: 1.3

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred

My start Gcode inclues a raise of Z to 10, but then when it actually starts printing octolapse doesn't start taking snapshots until Z>10.  I even tried using the Z-hop trigger, and same thing.

#### What should have happened?
Snapshots starting at first layer.

#### What happened instead?
Snapshots starting at Z>10

#### Operating System running OctoPrint and Octolapse

OS Name: octopi
Os Version: jessie

#### Printer model & used firmware incl. version

Printer Model: PRINTER_MODEL_GOES_HERE
Printer Firmware Version: PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2020-03-06:
There is only 1 good way to deal with this in the current master release:  Take a look at the priming height in your printer profile and make sure that is correct.  This should be set below 10MM (try 0.5mm).  I could take a look, but you'll need to post your settings and a sample gcode file for me to say anything definitive.

ForceConstant, 2020-03-07:
Thanks, I will check out the prime option, but here is my settings.json and starting gcode.

[Settings.zip](https://github.com/FormerLurker/Octolapse/files/4301195/Settings.zip)

```

M201 X500.00 Y500.00 Z100.00 E5000.00 ;Setup machine max acceleration
M203 X500.00 Y500.00 Z10.00 E50.00 ;Setup machine max feedrate
M204 P500.00 R1000.00 T500.00 ;Setup Print/Retract/Travel acceleration
M205 X8.00 Y8.00 Z0.40 E5.00 ;Setup Jerk
M220 S100 ;Reset Feedrate
M221 S100 ;Reset Flowrate

G28 ; home all axes
G29 ;

G1 Z10
G1 X0 Y0 Z10  F3000
G1 X0 Y0 Z0.2 F3000 ; get ready to prime
G1 X80 F5000 ; Wipe against brush.
G1 X0  F600 ; Wipe slow.
G1 Z10
G1 E10 20
G92 E0 ; reset extrusion distance

```

FormerLurker, 2020-03-07:
This gcode looks wrong:
G1 E10 20

What is the 20?

Also, here:
G1 X0 Y0 Z10
The Z10 is redundant
Other than that, it looks fine, though I dont claim to understand the purpose of this (you jave a brush?? Interesting)

Make sure your xyz and e axis are set to default to absolute, and set your prime height to 0.5.  that should do it.