FormerLurker/Octolapse #470: Feedback for V0.4.0rc1.dev5

FormerLurker, 2020-03-06:
Please see the release notes [here](https://github.com/FormerLurker/Octolapse/releases/tag/0.4.0rc1.dev5). This includes a list of enhancements, bug fixes, and issues known at the time of release.

All feedback is welcome here!  Let me know what you like, what you hate, what works, what is broken, etc. 

It is **very** likely that dev5 gets promoted to v0.4.0rc1, barring any serious bugs.  I would like to verify that nothing obvious is broken with OctoPrint v1.4.0, and if possible *Python 3*.  If anyone adventurous has *Python 3* installed and wants to help test I would be most appreciative.

steve220, 2020-03-07:
Installed this under 1.4.0. Installed fine and ran well on a short test. The only issue I had is that the custom camera setting for focus is not "sticky." All of the other camera settings are retained, but when I set a manual focus, save, then subsequently go back into the custom settings, the checkbox for autofocus is active and the focus slider is back to 0.

FormerLurker, 2020-03-07:
@steve220, you had this issue in the last version too, right?  I will come up with some steps to debug, and will create a new issue for this.  Will either be this evening or tomorrow AM.  Thanks for your report!  We will get this straightened out.

steve220, 2020-03-07:
Yep, same issue, but it's not that big of an issue.  I've installed the latest RC on 3 different Octoprint 1.4.0 installations. All working well and no hiccups during the installs.

FormerLurker, 2020-03-08:
@steve220, can you take a look at Issue #471?  Maybe we can get to the bottom of it.

Archer36, 2020-03-10:
Just installed it. Upgraded from dev4 using the GUI on OctoPrint 1.4.0 Python 2 using RPi Cam v2. No issues on upgrade. 

I upgraded to dev3 directly from v0.3.4 using the GUI as well and no issues there either. I am using TheSpaghettiDetective which seems to be coexisting fine with Octolapse (custom image settings disabled). I did a 30+ hour print on dev4 that completed today, seems like it worked well. Not sure if its to be expected but it seems like the time lapse had a few seconds of no movement at the start and end of the clip. 

I should be printing a few things over the week for the kids, so let me know if there is anything specific you want me to test! Thanks for the excellent plugin and hard work!

FormerLurker, 2020-03-10:
@Archer36, the start/end pause is configured in the rendering profile as pre/post roll.  Edit your rendering profile and double check for me will ya?

Archer36, 2020-03-12:
You are 100% correct sir! FYI I printed a 12ish hour long print and Octolapse captured the whole thing perfectly as far as I can tell.

FormerLurker, 2020-03-21:
Thanks @steve220 and @Archer36 for contributing!  I'm in the process of pushing out the first maintenance release (0.4.0rc1) now.  I have to create new issues and such first.

If you'd like to install or comment on the new release, check out [this issue](https://github.com/FormerLurker/Octolapse/issues/475).

Thanks!

closed 2020-03-22 by FormerLurker