FormerLurker/Octolapse #471: Camera Focus Not Saving - Custom Image Preferences

FormerLurker, 2020-03-08:
One user reported that the focus settings aren't saving within the custom image preferences for webcameras.  As of now I cannot replicate this issue.  However, I am guessing this has something to do with the 'Apply Preferences when Octoprint Starts' setting:

[image]

To debug this, first I'll need the camera profile settings.  These can be exported via the Octolapse settings pages:

[image]

A log would be helpful too.  This can be accomplished by following these steps:

1.  Set your *logging* profile to *log everything*:
[image]

2. Edit the logging profile and clear the current log file:
[image]

3. Replicate the issue if you can, by setting the focus and make sure it does not save.
4. Edit the logging profile and click *Download Log*:
[image]

5. Post the log file to gist.github.com and leave a link in here for me to review.

Any other info you can provide, such as steps to replicate, would be greatly appreciated.

Thanks!

steve220, 2020-03-09:
Please see the attached.

[Octolapse Troubleshooting-Steve.zip](https://github.com/FormerLurker/Octolapse/files/4306410/Octolapse.Troubleshooting-Steve.zip)

FormerLurker, 2020-03-10:
@steve220, thanks for the files!  It definitely looks like the control is being reset, but I'm having some trouble figuring out how that could happen.  Can you check your javascript console (F12) when you save the second time (in your example you changed brightness and saved after setting the focus)?  Perhaps there are some errors in there?

steve220, 2020-03-10:
Turns out I am getting an error in the console:
Autoclose property not set!
I'm not sure what to do about this. I've attached two images since I couldn't fit the listing for the error on one screen. 
Hope this helps.
[Octolapse-JavaConsole.zip](https://github.com/FormerLurker/Octolapse/files/4315187/Octolapse-JavaConsole.zip)

FormerLurker, 2020-03-10:
Hmm.  I dont think that is related unfortunately.  Can you remind me what browser/os you are using?

steve220, 2020-03-12:
I'm using the latest Chrome browser on Windows 10.  
On a separate note, I've had several great time lapses, but tonight I came home to download the latest one and after about 12 seconds, the quality drops dramatically. You can see it here:
https://youtu.be/ip8c2X3UP0E

Any ideas what might have happened?

FormerLurker, 2020-03-12:
It looks like the camera reverted back to auto white balance.  Either that or maybe the sun was shining through a window, or some low frequency LED lighting was turned on (that causes flicker too sometimes). 

Also, I was wondering where you are changing your camera settings (there are two ways).  Are you going into tye camera profile, or are you using the edit pen next to the timelapse preview?  Whichever one you are using, try switching and see if the problem goes away.

steve220, 2020-03-13:
I have been using the edit pen next to the camera profile. When I use the edit pen next to the timelapse preview, the behavior is correct and the focus setting is retained. Thanks.

FormerLurker, 2020-03-17:
@steve220, this is very useful information.  I think I've narrowed down the issue to a few lines of code.  Will continue to research and hopefully have a fix before I promote the current release.  I will keep you posted.

FormerLurker, 2020-03-18:
@steve220,

I have tried very hard to make things work 100% both when editing the camera settings inside of the camera profile AND by clicking the edit pen on the Octolapse tab, but it is just too complicated to finish right now.  Since I'm trying to get the next release out, my compromise solution was to remove the button within the camera profile and replace it with some instructions for how to edit the preferences from the tab.  I"m about to commit the changes, and will test on my pi along with some other minor changes that should go into rc1.  I'll let you know when I think it's good enough to test.

Thanks!

steve220, 2020-03-19:
I think removing the button is an excellent idea and could just be the long term solution. Not sure why we would need both ways to adjust things. Thanks for all you do for the printing community!!!

FormerLurker, 2020-03-21:
This has been fixed in v0.4.0rc1, which I am getting ready to release.  Thank you so much for your help!  your contribution is appreciated.

Closed!

closed 2020-03-21 by FormerLurker