FormerLurker/Octolapse #472: Failed trying to detect camera gphoto2 --auto-detect

KillerPrints, 2020-03-14:
#### Version of Octolapse
 
Octolapse Version: Octolapse v0.4.0rc1.dev5

#### Version of OctoPrint

OctoPrint Version: OctoPrint 1.4.0 running on OctoPi 0.17.0

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: NO

#### What were you doing when the problem occurred

I have followed the guide: https://github.com/FormerLurker/Octolapse/wiki/Configuring-an-External-Camera
Everything is working until I try to run step 4 which is testing the camera. I have it plugged into the Pi and also into my camera. The camera is turned on and it is a Nikon D3400. When I run the command gphoto2 --auto-detect it does not detect anything.

#### What should have happened?
It should have detected my camera

#### What happened instead?
It does not detect my camera

#### Operating System running OctoPrint and Octolapse

OS Name: Mac OctoPi
Os Version: 0.17.0

#### Printer model & used firmware incl. version

Printer Model: Ender 3 Pro
Printer Firmware Version: 1.1.5.2

#### Browser and version of browser, operating system running browser

Browser: Sfari 13.0.5
Browser OS: Mac OS Catalina

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2020-03-17:
Hi! 

So, I'm not sure how much help I can be if gphoto2 isn't working since I don't see your camera [in the list of compatible cameras](http://www.gphoto.org/doc/remote/).  You might want to create an issue in the [gphoto2 repository](https://github.com/gphoto/libgphoto2).  I'm betting the developer, who is quite active, will know if it can be made to work or not.

If you do create an issue there, paste a link in here please.  Thanks!

FormerLurker, 2020-05-08:
I'm hoping you got your issue worked out, and am going to close this now.  Let me know if you want me to reopen it.  Thanks!

closed 2020-05-08 by FormerLurker