FormerLurker/Octolapse #473: [Request] Printing from SD

kelemenlajos, 2020-03-21:
Dear @FormerLurker, 

[I read the previous request for this idea](https://github.com/FormerLurker/Octolapse/issues/18), but...

I think printing from SD option with a timer based expo would be a great feature. I understand why octoplase is based on checking and controlling every aspect of the gcode, and printing, and because of this the results are amazing! But some users (me included) would use a more simple timelapse solution for debugging some printer based printing issues using octolaps flexible rendering and capturing engine (advanced camera settings can be a great tool to highlight printing issues!). Also some Octoprint / Octolapse settings, if not set correctly, can cause problem with the print results. For crosscheckin these settings, i think an sd only timlapse would be a great tool.

br,
Lajos

FormerLurker, 2020-03-22:
Thank you for your suggestion, I will take it into account.  I can see that this could be useful for many users, but need to think about how it can be integrated with the current settings so that it is not confusing.  I will look into this more after v0.4.0 drops.  It's quite urgent that I get this new release out since Octolapse V0.3.4 isn't compatible with Octoprint 1.4.0.

Thanks for your post!

kelemenlajos, 2020-03-22:
Thanks for looking into this ;)