FormerLurker/Octolapse #474: Not taking snapshot at correct position V4.0rc1.dev5

davidms12, 2020-03-21:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.40rc1.dev5

#### Version of OctoPrint

OctoPrint Version: 1.3.12

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred

1. running a test print from cura
2. made sure extra gcode script had been added
3. made sure all configs under octolapse matched the recommended settings

#### What should have happened?
only take a snapshot when the extruder has moved to the fixed position (back left or right)

#### What happened instead?
takes snapshot while the head is in motion to the fixed position resulting in a jumpy difficult to watch timelapse

#### Operating System running OctoPrint and Octolapse

OS raspberry Pi 3+ b raspbian GNU
Os Version: 10 (buster)

#### Printer model & used firmware incl. version

Printer Model: Anycubic Mega S
Printer Firmware Version: 1.1.2 or (1.1.5 marlin) tried both

#### Browser and version of browser, operating system running browser

Browser: Chrome
Browser OS: 80.0.3987.132

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: https://gist.github.com/davidms12/6823c940873931fc7e6e8ab8df7ab323.js

#### Link to settings.json

Link to settings.json with all passwords removed: https://gist.github.com/davidms12/105bb49124eb210ddc4330c84df48129.js

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://gist.github.com/davidms12/57a76a81e715dc225ff7ca59c96be325.js

#### Link to octoprint.log

Link to octoprint.log:  https://gist.github.com/davidms12/77a917b513091f9c7f9c95e33fec45ac.js

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: 
https://www.amazon.com/photos/shared/4QyYsVbpSuOs55l9EfGmBw.oQ2mfMdisoSEdRtOJ640hS

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2020-03-22:
So, I looked through the log files you supplied, and noticed that debug logging wasn't enabled for Octolapse V0.4.0rc1.dev5.  I see lots of logs from the old version (0.3.4), but not enough to go on for your issues with the currently installed version.  Here's what you can do to fix this:

1.  Upgrade to Octolapse V0.4.0rc1 from the plugin manager using this url:  https://github.com/FormerLurker/Octolapse/archive/v0.4.0rc1.zip
2.  Reboot.
3.  Go to the Octolapse tab and change the 'Logging' profile to: Debug
4.  Click the 'Edit' pen next to the 'Debug' logging profile and click Clear Log.
5.  Run a short test print (you can stop after 4-5 layers if you'd l ike).
6.  Click the 'Edit' pen next to the 'Debug' logging profile and click 'Download Log'
7.  Post the log to gist, and leave a link in your response.

Please make sure that you are linking to the gist raw text view and not .js file, as you linked to earlier.  The link should look similar to this:  https://gist.github.com/davidms12/77a917b513091f9c7f9c95e33fec45ac

It's not a huge deal, as I can find the gists through your username, but it's simpler if you provide a direct link.

Thank you for posting your issue!  I'm looking forward to getting things straightened out.

davidms12, 2020-03-23:
Thanks FormerLurker. 

Octolapse upgraded, logs cleared, test run and here is the resulting log:

https://gist.github.com/davidms12/8b67d8a7e7f864d963579f497bd3eb3c#file-octolapse-log

hopefully I linked the file correctly this time.

Your help is greatly appreciated.

davidms12, 2020-03-26:
Any advancement on this?

thank you

FormerLurker, 2020-03-27:
Not yet.  I will be working on it tomorrow though.  Am planning to debug the more difficult issues then.  Thanks for checking in!

davidms12, 2020-03-28:
Ok thanks FL. Looking forward to making some cool timelapse videos :)

FormerLurker, 2020-03-29:
So, there are some pretty massive differences in the wait time for M400 on your printer.  It's hard to tell if that's really the issue, but it definitely could be.  Octolapse sends an M400 then an M114 and waits for the position to be returned from the printer.  It should not return the location until after all moves have completed.  If it's not doing this, then a stabilized timelapse is not possible without a very long camera delay.

The other thing I see is this: http://192.168.2.4:8888/out.jpg

Is that yawcam?  Is it possible that there is a delay here?  Maybe the camera is delivering a stale frame?  You could increase the camera delay to like 500MS and see if that helps.  You could also increase the framerate to something like 30fps and see if that helps.

FormerLurker, 2020-05-08:
It's been a while since this issue has been updates, so I'm assuming you got it worked out.  If not let me know and I'll be happy to reopen it.  Thanks!

closed 2020-05-08 by FormerLurker