FormerLurker/Octolapse #475: Feedback for V0.4.0rc1

FormerLurker, 2020-03-21:
Please see the release notes [here](https://github.com/FormerLurker/Octolapse/releases/tag/v0.4.0rc1). This includes a list of enhancements, bug fixes, and issues known at the time of release.

All feedback is welcome here!  Let me know what you like, what you hate, what works, what is broken, etc. 

I would like to verify that nothing obvious is broken with OctoPrint v1.4.0, and if possible *Python 3*.  If anyone adventurous has *Python 3* installed and wants to help test I would be most appreciative.

You can install this version of Octolapse using this URL in the OctoPrint plugin manager:
```
https://github.com/FormerLurker/Octolapse/archive/v0.4.0rc1
```
Please [read this guide completely](https://github.com/FormerLurker/Octolapse/wiki/V0.4-Getting-Started-with-Octolapse) before installation.  

Thanks!

FormerLurker, 2020-03-22:
FYI, issue #477, which was preventing OctoPrint from shutting down, has been fixed.  This issue doesn't seem to affect the shutdown commands in windows, or when using Rasbian.  However, It's hard for me to understand how shutdown was working at all in any Linux build.  The issue was caused by a thread that did not specify daemon=true, which was causing the OS to attempt to join to a thread which would never close.  I will push this out soon when RC2 drops.  I'm hoping to get a bit more feedback on this RC, specifically relating to Python3 (if possible), and regarding the new versioneer integration, including the custom version comparisons that I implemented.

tphandley, 2020-03-23:
Easy to install (octoprint 1.4.0) due to incompatibility with octolapse 0.3.4. A much easier interface. Time lapse worked beautifully first time. Only limiting factor is my hardware Raspcamera V2 and lack of zoom. Need to look into how to improve raspcam settings now.

lbussy, 2020-03-23:
Octoprint 1.4.0 - no issues installing.  Prusa MK3S, RPi camera.  I will be playing with this today.

lbussy, 2020-03-28:
Several days of printing.  Sometimes I get in a strange situation where the printer stays at 99% but the printer is done.  I'm not sure what to look at on that.  Other than that it's been working well.

FormerLurker, 2020-03-28:
@lbussy, thanks for the info.  I haven't experienced the 99% complete think yet, but I will keep my eyes peeled.  If I had to guess, I would suspect an octoprint issue of some kind.  However, if it is Octolapse, i definitely want to fix it.  Tha k you for your feedback!!!

lbussy, 2020-03-28:
I've got about a bazillion things going on so it's tough to zero in on a minor issue.  I'll try to pay attention next tim it happens.

FormerLurker, 2020-03-28:
@lbussy, you and me both!  Thanks for keeping your eyes peeled!  I will too.

tphandley, 2020-03-28:
No issues for me. Working perfectly.

Archer36, 2020-03-29:
Hi again @FormerLurker I successfully upgraded from V0.4.0rc1.dev5 using the GUI. Still running OctoPrint: 1.4.0 on Python 2 on a RPi 3B+ (PoE HAT). Will be printing tomorrow and this week and will let you know how things are going.

jneilliii, 2020-03-31:
@FormerLurker, I see the rc version has Python 3 compatibility. Just a reminder that you will also want to update the Plugin Repository (plugins.octoprint.org) with the new compatibility information. Additional information can be found [here](https://docs.octoprint.org/en/master/plugins/python3_migration.html#telling-octoprint-your-plugin-is-python-3-ready).

FormerLurker, 2020-03-31:
@jneilliii, Good call.  I didn't notice that it also needed to be stated in tye plugin repo!!  You probably saved me some time when I release on master.

ciordia9, 2020-04-06:
Semantic but your URL you reference in the beginning of this thread is missing the .zip extension otherwise it's not found. 

> You can install this version of Octolapse using this URL in the OctoPrint plugin manager:
> https://github.com/FormerLurker/Octolapse/archive/v0.4.0rc1

 https://github.com/FormerLurker/Octolapse/archive/v0.4.0rc1.zip

FormerLurker, 2020-04-06:
@ciordia9, good catch!  Fixed.

ciordia9, 2020-04-06:
Updated due to Python 3 migration and I wasn't smart to check incompatibilities. Testing this new version and I get this set of errors on the build:

**Start**
` Building wheel for Octolapse (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /home/pi/oprint/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-req-build-k69hxsr_/setup.py'"'"'; __file__='"'"'/tmp/pip-req-build-k69hxsr_/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-o9ks_qfx
       cwd: /tmp/pip-req-build-k69hxsr_/
  Complete output (551 lines):
  Found packages: {'octoprint_octolapse.test', 'octoprint_octolapse', 'octoprint_octolapse_setuptools'}
  /home/pi/oprint/lib/python3.7/site-packages/setuptools/dist.py:454: UserWarning: Normalizing 'v0.4.0rc1' to '0.4.0rc1'
    warnings.warn(tmpl.format(**locals()))`

**End**
`arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.7m -I/home/pi/oprint/include/python3.7m -c octoprint_octolapse/data/lib/c/gcode_position_processor.cpp -o build/temp.linux-armv7l-3.7/octoprint_octolapse/data/lib/c/gcode_position_processor.o -O3 -std=c++11
    In file included from octoprint_octolapse/data/lib/c/gcode_position_processor.cpp:23:
    octoprint_octolapse/data/lib/c/gcode_position_processor.h:30:10: fatal error: Python.h: No such file or directory
     #include <Python.h>
              ^~~~~~~~~~
    compilation terminated.
    error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1
    ----------------------------------------
ERROR: Command errored out with exit status 1: /home/pi/oprint/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-req-build-k69hxsr_/setup.py'"'"'; __file__='"'"'/tmp/pip-req-build-k69hxsr_/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-ci5b6sa2/install-record.txt --single-version-externally-managed --compile --install-headers /home/pi/oprint/include/site/python3.7/Octolapse`

Pastebin: https://paste.centos.org/view/a136290e

FormerLurker, 2020-04-06:
@ciordia9, Hmmm..  Looks like the linker can't find Python.h.  That is definitely a problem.  Would you mind doing a search on your PI to see if you can find Python.h anywhere?

This is exactly the kind of issue I was hoping for people to find, THANK YOU for reporting it.  I will create an issue for this later (probably after work) and will post a link.

ciordia9, 2020-04-06:
(oprint) pi@octopi:~ $ find . -name 'Python.h'
--nothing--

(oprint) pi@octopi:~ $ locate Python.h
/usr/include/python2.7/Python.h

Happy to help, now I head to work. :D

lbussy, 2020-04-06:
@ciordia9 you sort of need Python 3 in order to run Python 3 scripts.  :)

ciordia9, 2020-04-06:
I am running python 3. /usr/bin/python3 -> python3.7

FormerLurker, 2020-04-06:
Hmmm..  maybe it needs to be the dev version??  I will try a fresh install from the latest octopi using the official upgrade guide.  Im not sure how octoprint itself could have installed without Python.h, but it is possible if no extensions are being compiled.

ciordia9, 2020-04-06:
I just followed the migration guide from 2 to 3: https://docs.octoprint.org/en/master/plugins/python3_migration.html

> If you want to migrate your existing OctoPrint install on OctoPi to Python 3, I suggest to first make a backup, then move the existing venv /home/pi/oprint out of the way and create a new one based on Python 3 (which should already be present on current OctoPi images):

> mv ~/oprint ~/oprint.py2
> virtualenv --python=/usr/bin/python3 oprint
> source ~/oprint/bin/activate
> pip install "OctoPrint>=1.4.0rc1"
> sudo service octoprint restart

ciordia9, 2020-04-06:
I need to wait for a print to stop but I installed python3-dev:

```
(oprint) pi@octopi:~ $ locate Python.h
/usr/include/python2.7/Python.h
/usr/include/python3.7m/Python.h
```

I'll report back in a bit on whether that solves it.

ciordia9, 2020-04-06:
That solved it. So, to install, if you migrate you need to make sure to install python3-dev. Guess the rest of the plugins I use were not being built--which is odd due to the number I use. I'll keep testing this path and let you know if I see more issues.

FormerLurker, 2020-04-06:
Thanks for verifying.  It would have taken me a few days!

Also, fyi, octolapse is special in that it has to compile a python extension.  Plugins that don't need to do that probably would install fine.  I could work around this by packaging the extension, but it will take some effort to adjust the code.

ciordia9, 2020-04-06:
I like that you're special. 😀 

So testing! 

- My old profiles and settings carried over, which is always nice.
- Executed 15m print and it recorded it with overlays as usual.
- Dropbox plugin hasn't caught up to p3 yet so that's broken for the time being.

So good on first pass here. I'll do more in the coming days and let you know if it goes off the track somehow. Thanks as always for your work here. 👍

FormerLurker, 2020-04-06:
@ciordia9, thank you so much for your feedback!  Can you also check your printer profile and see if your make/model is listed?  If you haven't tried it yet, you might want to also switch your slicer type to 'automatic' while you're in there.  Read the help regarding automatic slicer settings, some minor adjustments need to be made for Cura, and there is one suggested change for Slic3r variants (Simplify 3d works out of the box).

ciordia9, 2020-04-07:
@FormerLurker I've been a 0.4 user for quite a while, just not with python3.

- Currently the Artillery Sidewinder X1 is not a profile that is available so I make my own
- Automatic is how I use it w/ the gcode additional info/template in Cura
- While it's not so useful for Octolapse I do run 2 cameras (C270 & Endoscope) -- the C270 is the main timelapse runner but I've tested it with the Endo and both at the same time and they work
- I've run most of the triggers but I still get confused on what to run for Vase mode items--everything I seem to pick still pauses the print at some point which creates holes so I gave up on that for the moment
- Rendering I run a 30FPS H264 variant

Anything else you want noted or tested?

Keichide, 2020-04-11:
I found something new in the rc1.. I somehow can't save the Main Settings it always drops some of Octoprints Info Popups with "Unable to save the main settings. Status: error. Error: INTERNAL SERVER ERROR" up.. But there is absolutly nothing in any Log File! And i already set everything on Verbose already... 

And Octoprint crashes every time i try to load a Gcode file while it's postprocessing the Data... Again no Errors inside the Log file and according to the Octolapse File he processed the gcode flawlessy right before Octoprint crashed.

I'm using the latest Stable Octopi Image with Octoprint 1.4.0 just to test what would happen if you actually put Octolapse 0.4 in the Stable Branch right now. So it's still using Python 2.6 while Octoprint also installed Python 3 with the 0.14 Update on the Pi.  But since i did'nt found any info in your release notes about changing the behaviour on Octoprint i thought i let everything like it is. 

Also the Build Plate Sizes for the Creality Ender 3 Profile on the Server are wrong, its 235x235x250

FormerLurker, 2020-04-11:
@Keichide, good catch regarding the main settings!  Not sure how I missed that.  I added the github commit version to create a direct link to the github commit, but forgot to update that routine.  I guess I don't change those settings often enough :)  I also found an error that was preventing the client versions (octolapse + git version) from being correctly saved after the main settings are updated.  I fixed that too.  I went ahead and pushed an update, which you can install from this plugin manager with this URL: https://github.com/FormerLurker/Octolapse/archive/163be9f62949d182659de8c774be2817a3444040.zip

Let me know if that fixes your main settings save issue.

> And Octoprint crashes every time i try to load a Gcode file while it's postprocessing the Data

Can you elaborate a bit here?  Do you mean you are trying to upload gcode while postprocessing is happening?  Also, when you say OctoPrint crashes, you mean the entire server locks up and needs to be rebooted?  I just need some idea how to recreate this.

> But since i did'nt found any info in your release notes about changing the behaviour on Octoprint i thought i let everything like it is.

Are you questioning whether or not Octolapse will run on python 3?  It should.  I will be doing more testing on this before release.  I just got another microSD so I can run a python3 and a python 2 version.

> Also the Build Plate Sizes for the Creality Ender 3 Profile on the Server are wrong, its 235x235x250

So, I noticed that there were differences between what the MFG reports and what some users are using for the print volume.  Is there ANY danger at all in setting it to 235x235?  If not, why does Creality report 220x220?  If 235x235 is safe, I'll use that instead.  If/when I push the settings updates, eventually you will be notified of an update (7 days or so) IF you did not check the customize profile box.  Otherwise you can uncheck that box and click the 'check for updates' button within the octolapse settings after I push the changes.

FormerLurker, 2020-04-11:
@ciordia9,

Have you tried the 'snap to print' stabilization for vase mode?  If you use it, make sure you set your camera delay to 0MS in the camera profile, and I recommend using 30FPS for your camera stream (lower and the delay is increased waiting for the next frame, higher and FireFox seems to crash the mjpegstream for some reason).  You may still get a seam, but I don't get any holes when I try it.  I'm sure it's quite model/printer dependent.  Vase mode is the most difficult thing for Octolapse to handle.

If you want to try upgrading to python 3, that would be excellent.  You do need to install python-dev, however, since the Python.h files weren't included in the latest octopi release for some reason.  I will have to post an issue about that, else users will have to install that manually.

Keichide, 2020-04-11:
> @Keichide, good catch regarding the main settings! Not sure how I missed that. I added the github commit version to create a direct link to the github commit, but forgot to update that routine. I [...]

Yes, that fixed it. 

> Can you elaborate a bit here? Do you mean you are trying to upload gcode while postprocessing is happening? Also, when you say OctoPrint crashes, you mean the entire server locks up and needs to be [...]
Strangely it seems that just Octoprint Crashes and not the whole Pi, also Python is'nt running anymore when i log into ssh, so it's probaly Python 2 related.. Anyway.. I think it has something todo with the Vase Mode in Cura, because i just printed an XYZ Cube just fine and eben Octolapsed processed the gcode without any problems.. So, it's probaly related that i did'nt set the Cura into the "Surface" mode and let the setting on "Normal" there.. But still.. It's a bit extrem that a single option in the Slicer crashes the whole Octoprint instance.

> Are you questioning whether or not Octolapse will run on python 3? It should. I will be doing more testing on this before release. I just got another microSD so I can run a python3 and a python 2 [...]

No, i was just a bit confused because all the people here, set it already on Python 3.

> So, I noticed that there were differences between what the MFG reports and what some users are using for the print volume. Is there ANY danger at all in setting it to 235x235? If not, why does [...]

Yeah, totally safe.. In every single tutorial Video (or website) it is mentioned that you set the Build Plate Size for an Ender 3 and Ender 3 Pro to 235x235 and i think even the Ender 3 v2 which is coming out in June will have the same size.  Just take a look on Amazon what size the Plate has.. The Ender 3 can use every mm of it and even has room for an bigger Heatbed.

FormerLurker, 2020-04-11:
Can you post your vase mode gcode?  It is entirely possible that python is crashing due to my gcode processor extension (they are tricky to get right).  I will run it through the debugger and will see if it crashes for me too.

Also, I will adjust the ender 3 build plate size appropriately, and will let you know when that is finished.

Keichide, 2020-04-11:
> Can you post your vase mode gcode? It is entirely possible that python is crashing due to my gcode processor extension (they are tricky to get right). I will run it through the debugger and will see [...]

Sure, here it is. 
[CE3_Julia_Vase_011_-_Heatwave_-_Solid.zip](https://github.com/FormerLurker/Octolapse/files/4465207/CE3_Julia_Vase_011_-_Heatwave_-_Solid.zip)

Keichide, 2020-04-11:
OH! Snap-to-print seems to be the culprit here.. I just used the same file just with Surface mode enabled (and changed the first layer Height also to 0.16mm) and changed the Trigger in Octolapse back to "Smart - High Quality" and it was working and i got to the Smart Preview Window and could start the Print.. (Which i of course did'nt let run because i wanted to actually test the Vase Mode Setting in Octolapse) 
[CE3_Julia_Vase_011_-_Heatwave_-_Solid_with-surface-mode-set.zip](https://github.com/FormerLurker/Octolapse/files/4465258/CE3_Julia_Vase_011_-_Heatwave_-_Solid_with-surface-mode-set.zip)

FormerLurker, 2020-04-11:
@Keichide, excellent!  That should help me debug, thanks!

FormerLurker, 2020-04-11:
FYI, I just pushed the new ender 3 profile.  Uncheck the 'custom' checkbox at the top, and click 'Update Profiles' near the octolapse main settings.  You should then be prompted to update your profiles.

Keichide, 2020-04-11:
> 
> 
> FYI, I just pushed the new ender 3 profile. Uncheck the 'custom' checkbox at the top, and click 'Update Profiles' near the octolapse main settings. You should then be prompted to update your profiles.

Thanks, but it's still the same Profile for me... Tryed it 4 Times now, with no sucess.

FormerLurker, 2020-04-11:
@Keichide, try the profile update again, I forgot to bump the version numbers.

FormerLurker, 2020-04-11:
Also, @Keichide, I created issue #495 to deal with the python crash problem.  That could get a bit hairy, so let's continue the discussion over there.  Can you export your settings.json (octolapse main settings->export settings) and post them?  I noticed you are not using automatic settings extraction, so I don't have your exact cura settings, which seem to be necessary to replicate the issue.  Or, you could enable automatic slicer settings extraction and then upload the re-sliced gcode.  [See this guide](https://github.com/FormerLurker/Octolapse/wiki/Automatic-Slicer-Settings) for enabling that, and please respond on the other thread.  Thanks!

Keichide, 2020-04-12:
Another thing came up.. I just tried to add a Watermark into my Videos, and it just said "Failed" without any Info why for most of my Images with a Transperany Layer in the Background. I think you should add a bit more Info there. Also they are not appering in the Preview Image and i really would like to have a way to position them somehow, just like the Text Overlay.

Keichide, 2020-04-12:
https://youtu.be/oeo8APl3-88 Here is an actual Timelapse shot with the Snap-to-print High Quality Profile. Which looks pretty good, but what i was actually expecting from Octolapse in a Vase Mode was a thing that happend at 0:22, where it began to slowly go around the Print. It looks like it's actually Printing it there right now, which is an awesome effect. 

https://youtu.be/TQnJIeZ35Cs It looks like the Plugin kind of tried the same thing here. 

But know i at least know the Limitations off having the Webcam on the Z-Axis and have to get another Webcam with an actual Zoom and a 3 or 4 axis Servo Controled holder. So that the whole Object is always inside the Picture.

Which remindes me of one limitation Octolapse currently has.. There is no way to add any Dynamic to the Stabilazation, for people like me, that use there Cam on the Z Axsis... Say for Instance Z ist at 10 but the Bed would still be at the same Position, so the Object would slowly move out of the Picture while the Camera is moving up, so Y would have to compensate for that, and also begin to move. Or the Webcam would have to begin Zooming in or Zooming out of the Picture, which would work better with Snap to Print.

tonysurma, 2020-04-22:
Install on 1.4 with zero issues.  This change made everything almost plug and play (compared to past one) - no playing with slicer settings, can now modify camera settings directly etc.  full success no issues on my side.  thank you1

FormerLurker, 2020-04-22:
@Keichide, your response got lost in my inbox, and I just noticed it while reviewing @tonysurma's response (thanks for that!).

Regarding snap to print, take a look at the 'smooth' option in the trigger profile (there are already downloadable profiles containing this setting).  What that option does is force the extruder to stabilize at the point closest to the stabilization point, but only for the very first stabilization.  From there, it tries to stabilize to the closest point to the PREVIOUS snapshot point, giving you a nice wandering effect.  A similar effect can be achieved by disabling stabilization entirely, though the first snapshot point is not controllable in that case.

That being said, I cannot view your videos because they are private.  Maybe set them not unlisted but not public?

Also, regarding 'dynamic' stabilization, you can specify a list of coordinates in the stabilizations to overcome your issue, though it will need to be done manually for every print with this issue.  Not a perfect solution, but I'm not sure how it could be automated without some serious knowledge about the model and some pretty sophisticated algorithm to control the stabilization.  When using snap to print, there are fewer choices about where to stabilize (it must be over the print), so that would further complicate any solutions.

Regarding changing camera settings, this could be accomplished via the snapshot scripts.  It wouldn't be simple, but it may be possible depending on the camera.

I will look into the watermark issue.  I actually did not add that feature and haven't messed with it much (it's extremely rare that anyone uses it).  I'll take a look and see if there are any obvious problems.  The text overlay was also a user submission (thanks again @Shadowen!), but I've made extensive modifications to it over the past year(s).  Due to all the modifications, I'm much more familiar about how that works.

Adding the watermark to the preview may be possible, but since we are currently adding the image in via ffmpeg (not manually constructing the frame via pillow) we are a bit limited in that regard in terms of accuracy.  I may have to move the watermarking overlay to the image conversion pre-processing routine that does image format detection and conversion (for DSLRs or uploaded images) , various transformations (flip, mirror, rotate, etc...), and adds the text overlay.  It will actually be a pretty big modification to the rendering pipeline, but not too terrible that it couldn't go into the next release (0.4.1) after this.  Maybe create a feature request for that?

Also, I haven't had time to look at the python crash yet (ug....) but I am REALLY going to try to get that dealt with so I can push out r2.  Since it's the worst possible kind of problem (crash) it must be fixed before the next release provided I can reproduce it.

FormerLurker, 2020-04-22:
@tonysurma, thank you for your feedback!  I'm so glad you found it mostly plug-and-play.  A LOT of emphasis on this release was to improve user friendliness without sacrificing configurability.  That is a difficult game to play with apps far less complex than Octolapse, so I'm glad all of the effort is paying off!

lbussy, 2020-04-22:
I'll put this in a new issue if you like.  I'm just not sure I am using it "properly."

When doing a lapse, the point at which the snap is taken seems rather arbitrary WRT the center point of the shape.  Some shapes (like a box) would benefit from being able to set a point at which to take the snap.  Is this possible with the current capabilities, or am I speaking of a new feature?

FormerLurker, 2020-04-22:
@lbussy, I need a bit more info to answer your question.  What trigger profile are you using, and what stabilization profile?

For the classic layer trigger, the point at which the snapshot is taken is usually very close to the start of the layer (depending on the extruder requirements).  That would make it seem random, though it is not.

For the 'Smart' triggers, the points are carefully chosen to minimize artifacts and the travel distance.  A lot of thought and experimentation went into the process, and it is pretty complicated.  The smart trigger prefers quality to performance unless the 'fast' option is selected, which can cause additional quality issues depending on the model, the placement, and the stabilization settings.  In general, the smart trigger will try to start a snapshot while over a prime tower, infill, or an interior perimeter (in that order), and dislikes taking snapshots over exterior perimeters.  If it finds multiple points of equal quality (say several over infill), it selects the closest point to the stabilization point.  This is a simplified explanation, as there are a lot of technical details.

The gcode trigger does allow you full control over where snapshots are taken, both in the 'classic' and 'smart' modes.  The main advantage of using the 'smart' gcode trigger is that you get a preview of all of the snapshot and stabilization points.  Perhaps this is the feature you are wanting?

lbussy, 2020-04-22:
I think so .... you used terms I've seen in there but did not understand until now (or at least I understand a little more.)

Let me poke at it and see what I can figure out.  You've helped by asking questions,. :)

Pavulon87, 2020-04-23:
Missing dot before "/plugin/" in 3 places.
For example, the export of settings does not work.
templates/octolapse_profiles_logging.jinja2:
`data-bind="attr: {href: '/plugin/octolapse/downloadFile?type=log&apikey=' + UI_API_KEY}" download="">`
templates/octolapse_settings.jinja2:
`data-bind="attr: {href: '/plugin/octolapse/downloadFile?type=settings&apikey=' + UI_API_KEY}"`
templates/octolapse_profiles.jinja2:
`data-bind="attr: {href: '/plugin/octolapse/downloadFile?type=profile&apikey=' + UI_API_KEY + '&guid=' + guid() + '&profile_type=' + $root.profileTypeName()}" download=""></a>`

Keichide, 2020-04-26:
> @Keichide, your response got lost in my inbox, and I just noticed it while reviewing @tonysurma's response (thanks for that!).
>
> Regarding snap to print, take a look at the 'smooth' option in the [...]

Yeah, that happens. Ayway, i put both Videos now on "Unlisted" because if thought it was the same as Private Videos on Youtube.. I think i got the Explanation wrong there. However one of the Two Videos was in "Smooth" mode but the Script did'nt quite get the Positions for it right.

> Also, regarding 'dynamic' stabilization, you can specify a list of coordinates in the stabilizations to overcome your issue, though it will need to be done manually for every print with this issue. [...]
It's more like an impossible Solution this way.. I already tried that. What i was thinking about is more like a Percent based movement, since the Plugin always knows the Coordinates of the Print, so if Z is at say 5% of the total Height, move the Bed 5% more from it's starting position to get the Image back into the Webcam view. It should actually be a simple Implanation since all Octolapse has to do is shift the starting Coordinates every time a little bit more, according to the Total Height of the Print. 

> Regarding changing camera settings, this could be accomplished via the snapshot scripts. It wouldn't be simple, but it may be possible depending on the camera.
> 

Oh yeah.. I totally forget that you always had something buildin for external Scripts

vajonam, 2020-04-28:
I was seeing this error in the logs.

```
2020-04-28 12:17:23,859 - octoprint.server.api.settings - ERROR - Could not load settings for plugin Octolapse (0.4.0rc1+u.163be9f)
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/site-packages/octoprint/server/api/settings.py", line 288, in _get_plugin_settings
    result = plugin.on_settings_load()
  File "/usr/local/lib/python2.7/site-packages/octoprint/plugin/types.py", line 1506, in on_settings_load
    if self.config_version_key in data:
TypeError: argument of type 'NoneType' is not iterable
2020-04-28 12:17:23,921 - octoprint.server.api.settings - ERROR - Could not load settings for plugin Octolapse (0.4.0rc1+u.163be9f)
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/site-packages/octoprint/server/api/settings.py", line 288, in _get_plugin_settings
    result = plugin.on_settings_load()
  File "/usr/local/lib/python2.7/site-packages/octoprint/plugin/types.py", line 1506, in on_settings_load
    if self.config_version_key in data:
TypeError: argument of type 'NoneType' is not iterable
``` 

Has this already been addressed?

FormerLurker, 2020-04-28:
I think so.  You could install from the devel branch and test.  There are a few fixes in there.

Ant8490, 2020-05-04:
I am using a DSLR camera with an external camera script to capture snapshots.  So far everything seems to be working great with the auto-import Cura settings and overall rendering.

Two questions:

1. Is there an option in this version to keep the snapshots that are saved rather than delete after a successful render?  I thought I seen an option for this in the non-devel branch?

2. Is there an option to disable ffmpeg rendering and just capture/save the images for later use?  I know the gphoto2 script can be modified to save photos on the camera, but was looking to see if there was a quick option in the settings.

FormerLurker, 2020-05-04:
1 and 2 have similar answers:  Yes.  Edit your rendering profile and uncheck enabled, or there is a disabled profile you can import (create new rendering profile and select disabled). This will cause a zip archive of your images to be created.  You can also render as normal and enable the archiving option in the rendering profile.  There you get the best of both worlds: a rendered video and a zip of the images for download.  You can even re render with different settings directly from the archive.  You can find the zip by opening up the files and videos dialog from the octolapse tab.  Enjoy!

Ant8490, 2020-05-04:
> 1 and 2 have similar answers: Yes. Edit your rendering profile and uncheck enabled, or there is a disabled profile you can import (create new rendering profile and select disabled). This will cause [...]

Perfect! Exactly what I was looking for.  Awesome job!