FormerLurker/Octolapse #477: Octolapse keeps Octoprint from shutting down

jangxx, 2020-03-22:
#### Version of Octolapse
 
Octolapse Version: 0.4.0rc1

#### Version of OctoPrint

OctoPrint Version: 1.4.0

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: doesn't matter

#### What were you doing when the problem occurred

1. Press Ctrl-C while Octoprint is running or send SIGTERM if Octoprint is running via systemd

#### What should have happened?
Octoprint shuts down.

#### What happened instead?
Octoprint just hangs. The webinterface is not accessible anymore and no log output is produced.
Attaching gdb to the python process and requesting the backtrace with `py-bt` reveals, that the process is waiting for a thread to join, but it never does.

```
(gdb) py-bt
Traceback (most recent call first):
  Waiting for the GIL
  File "/usr/lib/python2.7/threading.py", line 340, in wait
    waiter.acquire()
  File "/usr/lib/python2.7/threading.py", line 940, in join
    self.__block.wait()
  File "/usr/lib/python2.7/threading.py", line 1097, in _exitfunc
    t.join()
```

#### Operating System running OctoPrint and Octolapse

OS Name: Ubuntu
Os Version: 18.04.4 LTS

#### Printer model & used firmware incl. version

Printer Model: doesn't matter
Printer Firmware Version: doesn't matter

#### Browser and version of browser, operating system running browser

Browser: doesn't matter
Browser OS: doesn't matter

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: doesn't matter

#### Link to settings.json

Link to settings.json with all passwords removed: https://gist.github.com/jangxx/d7b1a4f13614b415a41fa8ab34e51107/raw/bcef529ba46d0a27fdba522f181c2255c46f15a1/settings.json

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://gist.github.com/jangxx/d7b1a4f13614b415a41fa8ab34e51107/raw/bcef529ba46d0a27fdba522f181c2255c46f15a1/plugin_octolapse.log

#### Link to octoprint.log

Link to octoprint.log:  https://gist.github.com/jangxx/d7b1a4f13614b415a41fa8ab34e51107/raw/0cc989e154c9c210b1191d4da0175bd0022e22ea/octoprint.log

#### Link to contents of Javascript console in the browser

Link to javascript console output: doesn't matter

FormerLurker, 2020-03-22:
It looks like I forgot to mark the rendering processor as a daemon thread, though I can't figure out how it hasn't caused a problem before.  I have pushed the changes to the devel branch, which you can install from the following link within the OctoPrint Plugin Manager:  ```https://github.com/FormerLurker/Octolapse/archive/0b693c5cecffd04bfcaa6bd3fa3ae7d80326bd15.zip```

Let me know if that takes care of it.  I searched for other threads where this was missed and didn't find any, but it's possible one or two are still lurking about somewhere.

jangxx, 2020-03-22:
The fix works perfectly, thanks a lot. Now I can finally shut down Octoprint properly without having to SIGKILL it all the time.

FormerLurker, 2020-03-22:
Not a problem, thanks for finding the bug!  I have no idea why it was shutting down at all in Rasbian, Forgetting daemon=true was not a good thing.  I'll add this commit to rc2.  Please make sure your software update plugin is set to follow Maintenance Rcs like so to ensure you'll be notified when the next version drops:

[image]

Going to go ahead and close this issue.  If you encounter the problem again let me know and I'll reopen it.  Thanks for contributing!

closed 2020-03-22 by FormerLurker