FormerLurker/Octolapse #478: Octolapse configured but no snapshots

tphandley, 2020-03-22:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: Octolapse 0.3.4

#### Version of OctoPrint

OctoPrint Version: OctoPrint 1.4.0 running on OctoPi 0.17.0

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred

1. Configured octolapse as per videos and Github instruction
2. tested camera (picamera V2) - detected
3. enabled test print and loaded gcode test print
4. Ensured G90 & M82 in Gcode
5. Started print

#### What should have happened?
Test print started with octolapse taking snap shots at end of each layer. 

#### What happened instead?
Test print continued but no frame shots taken, no pause to take shot, no warnings or errors given until end when rendering failed due to no snapshots.

#### Operating System running OctoPrint and Octolapse

OS Name: MacOS
Os Version: Catalina version 10.15.3 

#### Printer model & used firmware incl. version

Printer Model: STARTT Cartesian printer
Printer Firmware Version: Marlin 1.0.9

#### Browser and version of browser, operating system running browser

Browser: Safari
Browser OS: Version 13.0.5 (15608.5.11)

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: <script src="https://pastebin.com/embed_js/2DRxpEAP"></script>

#### Link to settings.json

Link to settings.json with all passwords removed: <script src="https://pastebin.com/embed_js/R0rRTq0S"></script>

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  <script src="https://pastebin.com/embed_js/7sQwAZZH"></script>

#### Link to octoprint.log

Link to octoprint.log: <script src="https://pastebin.com/embed_js/GnC5UDCC"></script>

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2020-03-22:
Octolapse 0.3.4 isn't fully compatible with Octoprint 1.4.0.  I just released the first non-development release candidate (0.4.0rc1), which may solve your issue.  You can upgrade by going into the software update plugin, edit the settings, and change the OctoPrint release channel to Maintenance RCs.  You should then be prompted to upgrade after clicking check for updates.  Let me know if you are willy to try it out!

tphandley, 2020-03-22:
Perfect. Updated to latest dev release and it seems to be working in test mode. Once done I’ll try a print.

FormerLurker, 2020-03-22:
Fantastic!  If your test print works, let me know.  I'll be about most of the day today.

If you have time I'd be appreciative if you could leave some feedback in the [v0.4.0rc1 feedback issue here](https://github.com/FormerLurker/Octolapse/issues/475).  All comments are welcome there, including things you like, hate, or just to let others know that it installed properly for you.

Also, you might want to edit your printer profile and see if your printer is listed by model in the make/model drop down.  This latest version of Octolapse has much better default settings too, but since you upgraded most of your old settings will be maintained.  If you want you could restore the defaults by clicking the gear icon in the Octolapse tab, then clicking *Restore Defaults*.  You'll have to reconfigure Octolapse, but if your printer is listed in the make/model drop down it should be a breeze.

tphandley, 2020-03-22:
Worked perfectly with a full print. Print quality unaffected. Much easier interface, no issues with install. Comments left on above feedback link too. My printer is a heavily modified STARTT from iMakr so no common and therefore not in dropdown, but since it,s been heavily modified (including print volume) if defaults were there it would be of no use anyway.
Thanks for an excellent plugin, hooked!

bsjelin, 2020-05-28:
How do I switch just the octolapse plugin to the latest dev release? All I can see in the software update is to switch octoprint itself to a different channel.

FormerLurker, 2020-05-28:
@bsjelin, check out [this new guide](https://github.com/FormerLurker/Octolapse/wiki/V0.4---Getting-Started).  See the link at the top for installation instructions, then proceed to the rest of the guide.

You might want to use this link for now to install, since I've not yet gotten around to publishing to the master branch (still working on documentation):  ```https://github.com/FormerLurker/Octolapse/archive/devel.zip```

In general I don't recommend installing from the devel branch, but I've added a few significant bug fixes that I think you will want.  Lots of documentation has been updated and links have been changed to point to the new V0.4 wiki.