FormerLurker/Octolapse #479: Tinycam pro

lordstryfe, 2020-03-25:
I am using tinycam pro on an android device to stream my wyzecam. I can see the stream just fine in the normal octoprint timelapse setting. Yet no matter what I so I cannot get octolapse to pull an image no matter what settings I put in. 

Any hwlpl would be great.

FormerLurker, 2020-03-25:
What snapshot url are you using in octoprint vs octolapse?  Please fill in the issue template next time you post.  Otherwise my first few responses will probably be to ask for information that was in the template.

FormerLurker, 2020-05-08:
I'm going to assume you figured this out.  If you want me to reopen the issue, fill in the template and I'd be happy to reopen it.  Thanks for posting!

closed 2020-05-08 by FormerLurker