FormerLurker/Octolapse #480: Layer offset in during printing after exiting the web interface

Druard, 2020-03-25:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: OCTOLAPSE_VERSION_GOES_HERE

#### Version of OctoPrint

OctoPrint Version: OCTOPRINT_VERSION_GOES_HERE

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES_OR_NO

#### What were you doing when the problem occurred

1. STEP_ONE_GOES_HERE
2. STEP_TWO_GOES_HERE
3. STEP_...

#### What should have happened?
PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: OS_NAME_GOES_HERE
Os Version: OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: PRINTER_MODEL_GOES_HERE
Printer Firmware Version: PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2020-03-25:
Can you please fill in the issue template?  It is very difficult to help diagnose your problem without it.  I especially need the version of Octolapse you are using (0.3.4 can cause layer shifts if certain option are enabled).

Druard, 2020-03-25:
Version of Octolapse
Octolapse Version: Octolapse (0.4.0rc1.dev4)

Version of OctoPrint
OctoPrint Version: OctoPrint 1.4.0 running on Debian

Operating System running OctoPrint and Octolapse
OS Name: Windows
Os Version: Windows 10

Printer model & used firmware incl. version
Printer Model: Sapphire pro
Printer Firmware Version: Marlin 2.0

Browser and version of browser, operating system running browser
Browser: Chrome

Please consider becoming a patron
If you like this project, please support my work by becoming a patron, and consider adding a 'star' to the repository. It takes a lot of time and effort to maintain the project and respond to issues. The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various videos and tutorials by subscribing to my Youtube channel. You can also follow me on Twitter.

FormerLurker, 2020-03-25:
Thank you!  Can you take a photo or video of what you are seeing to help me understand?  Also, please try yo export your octolapse settings (gear icon on octolapse tab, export settings), and post that to gist.github.com.  thanks!

Druard, 2020-03-25:
I hope I did the right thing

<script src="https://gist.github.com/Druard/da9b16e3cdc95b45c50e150e555fbcad.js"></script>

[image]

FormerLurker, 2020-03-25:
Yes, this is what I needed.  I adjusted the gist link, and all is good there now too!

Questions:  

* Have you only seen this once, or does it appear with every timelapse?
* Do you ever have this problem without Octolapse?  
* Is your firmware stock marlin 2, or something specific to your printer?

There are firmwares out there that do not properly handle M400, and hopefully this is not your issue.  I've heard from many people using Marlin 2.0, and haven't seen this before, so if your firmware is stock that shouldn't be the issue.  You could try lowering your travel speed (according to the MFG your printer supports travel up to 1000mm/sec, though I find that difficult to believe) to see if that has an impact.

It's also possible that you need to increase your z-lift a bit.  Right now it is equal to your layer height, so there is no margin for error.  Perhaps try 0.6mm as a test.

Also, since you are using Prusa Slicer, you could enable automatic slicer settings.  See [this guide](https://github.com/FormerLurker/Octolapse/wiki/V0.4-Slic3r,-Slic3r-PE-and-PrusaSlicer) for enabling verbose gcode, which should improve quality when using the smart trigger.  However, I don't think this will affect the layer shift.

FormerLurker, 2020-03-26:
Also, you might want to update to the latest release candidate.  To do this:

1. Open the OctoPrint settings (wrench/spanner icon)
2. Select the *Software Update* from the left menu
3. Click the spanner/wrench icon to edit the software update plugin settings
4. Change the OctoPrint Release Channel to 'Maintenance RCs' and save your changes.
5. Click *Check for Updates Now*
6. OctoPrint will report an update is available.  Update, reboot, and the new version should be installed.

Let me know if you have any problems!

Druard, 2020-03-27:
Did as you wrote, let's see what happens. Thanks.

FormerLurker, 2020-04-04:
Any update?  Hopefully the new version solved your issues.  It is now my #1 debugging recommendation :)

buffalowolff, 2020-05-05:
I was having the same exact problem where it would seem like the 'center' would move throughout a print, often several times.  I don't think I completed a single print once setting up octolapse.  The RC advised in this thread seems to have fixed the problem for me.  (Prusa I3 MK2.5)

FormerLurker, 2020-05-08:
@Druard, it's been a while since your last response, so I'm going to go ahead and close this issue out.  Let me know if you want it reopened.  Thanks!

Druard, 2020-05-09:
Thank you, i hope my problem dissolve.

FormerLurker, 2020-05-09:
@druard, so your issue is not resolved?  I definitely want to fix it.  Try installing for m the dev branch (let me know if you need help doing that) and see if it is fixed.  I'm currently working on issues to prepare for the RC2 release so today is the day.

buffalowolff, 2020-05-30:
After updating to rc2 yesterday this problem has returned. I'm going to try and downgrade back to rc1 if possible to see if it resolves.

FormerLurker, 2020-05-30:
I don't think any changes from rc1 to rc2 could have caused this.  Maybe try stabilizing in the center as a test?  I had some layer shift issues when stabilizing in the back left, and it ended up being a crooked belt that was getting stuck randomly.

buffalowolff, 2020-05-30:
I did switch to 'smart' animate from gcode animate.  I'll switch back and
test when I have another opportunity.