FormerLurker/Octolapse #481: Octolapse only takes 1 picture at the start of print.

omega-art, 2020-03-26:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: OCTOLAPSE_VERSION_GOES_HERE
Octolapse v0.3.4

#### Version of OctoPrint

OctoPrint Version: OCTOPRINT_VERSION_GOES_HERE
OctoPrint 1.3.12 running on OctoPi 0.14.0

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES_OR_NO

NO
#### What were you doing when the problem occurred

Im attempting to print my first timelapse using octolapse on my new prusa I3 Mk3s
1. STEP_ONE_GOES_HERE I took my raspberry pi from my old printer set up a printer profile for my new  prusa I3 mk3s
2. STEP_TWO_GOES_HERE I installed octolapse on it this morning and configured the printer profile starting with the Prusa I3 Mk3 profile already loaded.
3. STEP_... I printed a small thing using one of the pre-sliced gcode profiles that came on the sdcard that came with the printer.
The print and the octolapse worked as planned.
4. I then sliced something on my own and tried to print it after confirming the start code contained g90/g91 and M90/91.

#### What should have happened?
PUT_YOUR_DESCRIPTION_HERE
The octolapse should have worked the same way it worked with the pre sliced g code.

#### What happened instead?
PUT_YOUR_DESCRIPTION_HERE
Octolapse took 1 picture before starting the first layer and then nothing.

I have tried copying the start code from the pre sliced model and pasting it into my Gcode and it did not have any effect.
#### Operating System running OctoPrint and Octolapse

OS Name: OS_NAME_GOES_HERE  
OctoPi
Os Version: OS_VERSION_GOES_HERE 
OctoPrint 1.3.12 running on OctoPi 0.14.0

#### Printer model & used firmware incl. version

Printer Model: PRINTER_MODEL_GOES_HERE
Prusa I3 Mk3
Printer Firmware Version: PRINTER_FIRMWARE_VERSION_GOES_HERE
3.8.1 2869

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

omega-art, 2020-03-26:
Update I decided to slice the same model as the pre slice gcode and it started taking pictures but not on every layer change. out of over 125 layer it only took a picture of 23 of them.

Under trigger state it keeps going between saying "waiting for feature to trigger snapshot" and "waiting for extruder to trigger snapshot."

FormerLurker, 2020-03-26:
You are using high quality mode snapshot profile.  Select the compatibility mode.  V0.4 uses new feature detection that doesn't mess up as often as the old version.

FormerLurker, 2020-03-26:
Hey, since you're using an MK3, you might want to try out the new release candidate (v0.4.0rc1), which has the comment based feature detection (though you will need to change 1 setting in PrusaSlicer).  It also will extract your slicer settings directly from the gcode file, which makes life easier.  If you want to try it out let me know.

omega-art, 2020-03-26:
Compatibility mode helped but the z seam was terrible.  What settings can fix that? How do i get the new release?

FormerLurker, 2020-03-26:
1. Open the OctoPrint settings (wrench/spanner icon)
2. Select the *Software Update* from the left menu
3. Click the spanner/wrench icon to edit the software update plugin settings
4. Change the OctoPrint Release Channel to 'Maintenance RCs' and save your changes.
5. Click *Check for Updates Now*
6. OctoPrint will report an update is available.  Update, reboot, and the new version should be installed.

Read [this guide](https://github.com/FormerLurker/Octolapse/wiki/V0.4-Getting-Started-with-Octolapse) first!   Let me know if you have any problems!

FormerLurker, 2020-03-26:
Let me know if you still get a z seam.  If you are printing in vase mode, use the new snap to print trigger profile (that is a very special case... Hard to eliminate z seam entirely in that case, but increasing the camera frame rate and setting your camera delay to 0 will help.  Make sure retraction is enabled too!)

FormerLurker, 2020-04-04:
Any update on this?  I'm hoping the new version solved your issue mostly.

omega-art, 2020-04-04:
I dont have an update. I'm waiting on a new raspberry pi and sd card which isnt due to be delivered for another 2 weeks due to Covid 19. I had borrowed the r pi from another printer which is running an old version of octolapse but in order to do updates I have to reimage the sdcard and since Ive already got it set up to run my other printer I decided to wait on the new one.

FormerLurker, 2020-04-04:
No problem!  Take whatever time you need, and thanks for responding.

omega-art, 2020-04-04:
I appreciate your help. By the way I saw a really cool time lapse the other day where they had an image taken at a slightly different position on the perimeter for each layer. The result was it looked like the print head was moving at slow speed but the print was growing at high speed the way you would see 3d printers work in cartoons. It would be awesome if octoprint could do that.

FormerLurker, 2020-04-04:
That time lapse was probably done with octolapse using one of the default animated stabilizations :) check your drop down and look for orbit or animated printing.  You can add your own customized animation too!

omega-art, 2020-04-07:
The new pi arrived early. Ive only done one small print but it looks like 0.4.0 worked.

FormerLurker, 2020-05-08:
Forgot to close this, sorry.  Thanks for posting!