FormerLurker/Octolapse #482: DAVE11674 octolapse no custom page for raspicam v2.1

dave11674, 2020-03-26:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: OCTOLAPSE_VERSION_GOES_HERE

#### Version of OctoPrint

OctoPrint Version: OCTOPRINT_VERSION_GOES_HERE

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES_OR_NO

#### What were you doing when the problem occurred

1. STEP_ONE_GOES_HERE
2. STEP_TWO_GOES_HERE
3. STEP_...

#### What should have happened?
PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: OS_NAME_GOES_HERE
Os Version: OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: PRINTER_MODEL_GOES_HERE
Printer Firmware Version: PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

[... 576 lines cut ...]
"currentResolution": "255"
},
{
"id": "12",
"name": "Y/CrCb 4:2:0",
"compressed": "false",
"emulated": "false",
"current": "true",
"resolutions": {"0": "32x3280"}
,
"currentResolution": "255"
},
{
"id": "13",
"name": "32-bit BGRA/X 8-8-8-8",
"compressed": "false",
"emulated": "false",
"current": "true",
"resolutions": {"0": "32x3280"}
,
"currentResolution": "255"
}

]
}

FormerLurker, 2020-03-26:
Thank you very much!  I will look into this and hopefully get the custom page working very soon.

FormerLurker, 2020-03-26:
Well, since I am waiting for some work stuff to finish, I went ahead and added the necessary controls.  Try upgrading Octolapse by installing via the plugin manager (no need to reinstall) using this url in the 'from url...' box:  https://github.com/FormerLurker/Octolapse/archive/devel.zip

Let me know if it works!

dave11674, 2020-03-26:
Just sat eating mate will try after i fin
I cant hook up go the printer yet
As its 77% through a 26hr print lol

dave11674, 2020-03-26:
ok about to try now mate, just installed from the link and now rebooting

dave11674, 2020-03-26:
looks good now mate :)

[image: image.png]

dave11674, 2020-03-26:
[image: image.png]
you put one too many "with with" in the preview field mate

dave

FormerLurker, 2020-03-26:
Thanks!  Fyi, your pic didn't show up.  Can you repost?

dave11674, 2020-03-26:
[image: image.png]
this pic ?

FormerLurker, 2020-03-26:
Still didn't work, lol!  Well, i can probably figure it out.  Thx for trying!!!

dave11674, 2020-03-26:
or this pic with the settings

FormerLurker, 2020-04-04:
I just fixed the redundant 'with' and pushed to the repo on the devel branch.  Here is the link if you want to verify this:  https://github.com/FormerLurker/Octolapse/archive/devel.zip

I also added G2/G3 support (necessary for compatibility with another plugin I'm writing) and fixed a shut down issue (non-daemon thread for some linux builds..  strange it was working on the pi, but it's fixed now).

If it looks good should I consider this issue closed?

FormerLurker, 2020-04-26:
It's been a while since this ticket has been updated, and I think both the initial issue and the typo have been corrected, so I'm going to go ahead and close it.  Let me know if you want it reopened.

Thanks for posting!

closed 2020-04-26 by FormerLurker