FormerLurker/Octolapse #483: Seam when using Snap to Print in vase mode

LP0101, 2020-03-27:
#### If this is a feature request describe it here

#### Version of Octolapse
 
Octolapse Version: v0.4.0rc1

#### Version of OctoPrint

OctoPrint Version: 1.4.0

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: No

#### What were you doing when the problem occurred

Printing in vase mode

#### What should have happened?
Print quality should not be affected

#### What happened instead?
There is a clear seam along the stabilization axis

#### Operating System running OctoPrint and Octolapse

OS Name: Raspian
Os Version: 10

#### Printer model & used firmware incl. version

Printer Model: Prusa MK3S
Printer Firmware Version: Latest stable

#### Browser and version of browser, operating system running browser

#### Link to the gcode file you were printing when the problem occurred

#### Link to settings.json

#### Link to plugin_octolapse.log

#### Link to octoprint.log

#### Link to contents of Javascript console in the browser

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: https://imgur.com/a/0OyKfHq

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2020-03-27:
You won't be able to eliminate it entirely, but you can do a few things to reduce it:

1.  Set your camera delay to 0 (camera profile) and dont use a dlsr (too slow).
2.  Increase your camera's framerate to at least 30fps (not much more else you may have other issues). 
3.  Don't over retract.  Just use enough to take the pressure off of the nozzle.

In summary, you want to minimize the snapshot time.  Ideally it will be difficult to tell when Octolapse takes a pic.  It should be that fast.

After all of this, provided your printer is also perfectly calibrated, you will still have a slight seam.  I have worked hard to minimize this, and hopefully someday will get it perfect.  Vase mode is a real challenge, and if any seam at all is unacceptable, you probably will want to turn off Octolapse for those prints, or make a thicker edge and stability towards the center (try the smooth snap to print for that.)

FormerLurker, 2020-04-04:
Any update about your seam issue?  I'm hoping my suggestions reduced the seam you were seeing.  I'm continuing to do some additional tests with vase mode, but so far haven't been able to eliminate the seam 100% (maybe 90%).  The good news is there aren't any gaps or missing infill in my tests.  Not perfect, but it's a drastic improvement from V0.3.4.  I'm still not officially recommending Octolapse for use with Vase Mode (I thought I would be able to, but after testing it looks like a no), but would like to make it work as well as possible.

Thanks again for you post!

FormerLurker, 2020-05-08:
It's been a while, so I'm going to assume your issues are resolved.  If you want me to reopen the issue, I'd be happy to do so.

closed 2020-05-08 by FormerLurker