FormerLurker/Octolapse #484: Camera Error Issue I can't figure out

mordenfreeman, 2020-03-27:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: Octolapse 0.3.4

#### Version of OctoPrint

OctoPrint Version: OctoPrint 1.4.0 running on OctoPi 0.17.0

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred

1. SEND PRINT WITH PROPER GCODE
2. CAMERA SHUTTER GOES OFF ON MY SONY A7sii, Picture gets added to the temp snapshot dir
3. OCTOLAPSE CAMERA ERROR shows up and when it builds the timelapse it fails because there is only 1 frame.   

#### What should have happened?
Write photos to the snapshot directory / create a sequence. 

#### What happened instead?
OCTOLAPSE CAMERA ERROR, no other message.   It worked perfect for one print and now it fails evertime.

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: OctoPi 0.17.0

#### Printer model & used firmware incl. version

Printer Model: Ender 3 Pro
Printer Firmware Version: Updated Marlin

#### Browser and version of browser, operating system running browser

Browser: Chrome
Browser OS: Windows

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: https://github.com/mordenfreeman/octolapse_logs/blob/master/settings.json

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://github.com/mordenfreeman/octolapse_logs/blob/master/plugin_octolapse%20(3).log

#### Link to octoprint.log

Link to octoprint.log:  https://github.com/mordenfreeman/octolapse_logs/blob/master/octoprint%20(2).log

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

mordenfreeman, 2020-03-27:
MORE INFO:

Camera : Sony a7sii via usb to the pi4, 

 "ghoto2 --auto-detect"  works and returns:

Model                          Port
----------------------------------------------------------
Sony Alpha-A7S II (Control)    usb:001,006

and

"sudo -S gphoto2 --capture-image-and-download --filename "${SNAPSHOT_FULL_PATH}" works as well and writes an image.

It's very strange because I can't seem to figure out why it will write a single frame then from that point on it says "CAMERA ERROR" and no longer writes frames.   

Strangely enough if I SSH into the temp snapshot directory and delete that single {timelapsename.00000.jpg) frame, then the next octolapse snapshot will write sucessfully and it will once again be stuck with that latest photo and repeat the error.

+ thanks if you get around to looking at this, the one timelapse that worked with your DSLR instructions was awesome, thank you for what you do.

FormerLurker, 2020-03-27:
Sounds like some kind of permission error.  Fyi, the sony cams are notoriously difficult to get work with Octolapse.  It has been done, but takes some effort and patience.  I will dig through the issues to see what I can find.

FormerLurker, 2020-03-29:
So, I looked through the logs and didn't find any major problems.  However, the version of Octolapse you are using does not capture all of the script output.  Try upgrading to the latest release candidate and creating a more detailed script log by following these steps:

1. Open the octoprint settings.
2. Select the software update plugin.
3. Edit the settings (wrench/spanner icon).
4. Change the Octoprint Release Channel to Maintenance Rcs:
[image]
5. Save the changes.
6. Expand 'Advanced Settings' at the bottom and click 'force check for updates':
[image]
7. Update octolapse and reboot (v0.4.0rc1).
8. Read [this guide](https://github.com/FormerLurker/Octolapse/wiki/V0.4-Getting-Started-with-Octolapse) to get started using the new version.
9. Select the *Debug* logging profile in the octolapse tab and enable test mode:
[image]

10. Edit your debug settings and clear the current log:
[image]

11. Print a couple layers (making sure some of the images are not stabilized), then cancel the print.
12. Edit your logging settings, download and post the log:
[image]

I know that looks like a lot, but I think it's our best bet.

mordenfreeman, 2020-03-30:
Wasn't a problem & appreciate the snapshots.   

Posting the updated log below and funny enough it's looking like that test timelapse worked just fine.   I also had no issues after the first dslr octolapse so I'm gonna keep running more prints and clearing the log between prints and see if I hit another error.

sidenote ( this ui is so much better than the standard release, I'm actually able to play timelapses with the play button on the octolapse preview and It updates the snapshot after each photo.   ( i wasn't getting that even with my pi cam, let alone the DSLR ) 

https://github.com/mordenfreeman/octolapse_logs/blob/master/plugin_octolapse%20(%20v0.4.0rc1).txt

FormerLurker, 2020-04-04:
I'm glad you like the improvements!  It's been over 18 months in the making :)

So how are things running?  Did the upgrade solve your issues?  Scripting cameras + rendering was always a bit flaky in v0.3.4, so I'm hoping 0.4 fixed your problems.

FormerLurker, 2020-05-08:
I'm going to assume that upgrading fixed your issues and am going to close this.  If you want me to reopen it I'd be glad to do so.

closed 2020-05-08 by FormerLurker