FormerLurker/Octolapse #485: Unable to enable Custom Image Preferences

sciaio, 2020-03-28:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.3.4

#### Version of OctoPrint

OctoPrint Version: 1.3.12

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: NO

#### What were you doing when the problem occurred

1. Trying to enable cotrol.htm to enable Custom Image Preferences
2. I followed the guide to manipulate the octopi.txt file in order to get access to control.htm
3.  The same error "Unable To Enable Custom Preferences" is shown
4. I double checked that there aren't sintax error in the octopi.txt
5. I reboot several times via comand reboot, so not restarted

#### What should have happened?
It should permit to enable the "Enable And Apply Preferences Before Print" setting

#### What happened instead?
It shows the error "Unable To Enable Custom Preferences" with this description: "mjpegstreamer-control-error: The server denied access to the MJPG-Streamer control.html for the 'Default Webcam' camera profile. Please see this link to correct this error., or disable 'Custom Image Preferences'."

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: 0.15.1

#### Printer model & used firmware incl. version

Printer Model: PRINTER_MODEL_GOES_HERE
Printer Firmware Version: PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://gist.github.com/sciaio/b360dd6f5e5ca6993741fe88348db7fd

#### Link to octoprint.log

Link to octoprint.log:  https://gist.github.com/sciaio/a0000e6168028fcdac2f6fbc45f9d035

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2020-03-29:
What kind of camera?

FormerLurker, 2020-04-04:
Hey, any update on this?  I'm more than willing to help, but I need a bit more info, specifically which model of camera are you using.  Sorry I missed the bit about you having *rebooted several times* (I was on mobile when I read this).  I've been quite busy this past week.

sciaio, 2020-04-20:
Sorry about the late response, I received no notifications.

> What kind of camera?

I have a GUCEE HD92 FullHD 1080p

FormerLurker, 2020-05-08:
Sorry it has taken me a while, but I've been sort of buried :)

So, can you navigate to the following url:

http://{IP_OF_PI}:8080/control.htm

You should get a screen that shows some camera controls.  If you don't, can you paste in the last few lines of octopi.txt (or just octopi, I think there was a file name change recently)?

Also, I don't see this in your logs, but if you have The Spaghetti Detective or Octoprint Anywhere installed you might want to disable those for testing since they use their own camera server.

FormerLurker, 2020-06-12:
Any update on this?

spanzetta, 2020-07-06:
hi FormerLurker,
I have the same problem.. "unable to enable Custom Preferences" popup error..
I am using Octolapse 0.40 on Octopi 0.16 (Raspberry PI 3 A+) and Ender 3 Pro with latest FW (1.1.6 official)
But pls note that Octolapse works fine (I already grabbed few Timelapse.. but all at 640x480 res even if I modified octopi.txt with the line "camera_raspi_options="-x 1920 -y 1080 -fps 10""
Any suggestions?
Thanks

FormerLurker, 2020-07-06:
Resolution is not controlled by octolapse unfortunately, so you may want to look elsewhere for that.  Regarding custom image preferences, [see this link](https://github.com/FormerLurker/Octolapse/wiki/V0.4---Enabling-Camera-Controls)

spanzetta, 2020-07-06:
Thanks for your quick reply..
I saw that link.. and I found strange that have to configure the raspberry PI Camera as .. USB Camera -> _camera_usb_options_="-r 1920x1080 -f 30")... instead of "_camera_raspi_options=_....."
Pls note also that all images at the provided link are not shown.. could you pls check?
Thanks
[image]

FormerLurker, 2020-07-06:
Hmm, here is what I see:

[image]

Not sure what's going on there!? 

Also, the reason why you need to change the usb options and not the raspi option is because while following the instructions on that wiki page we switched to using the raspberry pi camera driver, which acts like a USB camera, which is required to control the camera via mjpgstreamer.

spanzetta, 2020-07-06:
sorry.. my proxy was blocking (only) images..
Thanks for your clarifications.. I will try to follow that guide
Thanks again

FormerLurker, 2020-07-06:
Any time, enjoy!  I will reopen this if you continue to have issues.

closed 2020-07-06 by FormerLurker

spanzetta, 2020-07-07:
Hello FormerLurker, 

I did follow the mentioned guide (at this this [link](https://github.com/FormerLurker/Octolapse/wiki/V0.4---Configuring-a-Raspberry-Pi-Camera)) and now the situation is the following:

1) video image is Zoomed (cropped) and with horrible colors.. do you know how to fix the problem? Which settings to manipulate? I am using a Raspberry PI Camera (Labists B01)
See below the screenshot with original Settings (which captures Timelapse at 640x480) compared to the new settings (1920x1080@30fps) ...  

2) (minor) the "http://ipaddress/webcam" link is not working anymore .. 
maybe is it normal? What is the new link?

Thanks in advance for your help (sorry but I approached the 3D Printer AND Raspberry world only one month ago..)

Compliments for all of your great job (you, Gina, etc)

Regards

[image]
[image]

spanzetta, 2020-07-07:
Pls check my latest reply at
https://github.com/FormerLurker/Octolapse/issues/485#issuecomment-654713887
thanks

FormerLurker, 2020-07-07:
Are you sure you followed the last section [here](https://github.com/FormerLurker/Octolapse/wiki/V0.4---Configuring-a-Raspberry-Pi-Camera#step-10---dialing-in-your-camera)?  It looks like you did not, because I can see your video bitrate setting is not maxed as is suggested there.  You need to increase the video bitrate if you increase the resolution because more data exists in a higher resolution image.  Crank that all the way up.  The other options under *Codec Controls* don't seem to do much.  I also don't see much (if any) effect from the jpg quality or compression quality sliders, but perhaps try maxing those out too.

Second, due to the way the raspberry pi handles different resolutions, depending on which one you choose you may get the zoomed in look, or what we call partial field of view.  See [this link](https://picamera.readthedocs.io/en/latest/fov.html#sensor-modes) for details.  Be sure to choose a resolution that has a full field of view.  I actually like mine 'zoomed in' due to where I have my camera positioned, so this is simply a matter of preference.

After changing the resolution to one you want, you may need to reduce the FPS somewhat if you increase the resolution a lot, depending on the quality of your network (wired is definitely best).  Once you have made your changes be sure to do a **full** reboot (not just restarting octoprint).

Regarding the colors, they will mostly improve when you increase the bitrate.  Let me know how it goes.  There are a few other settings that might need to be changed in order to get good color accuracy.

FYI, the raspberry pi camera is pretty notorious for being difficult to configure and finicky.  If you are ever in the market for a webcam I would strongly recommend you get a USB webcam (like a logitech 920 perhaps), which are much more user friendly, easier to setup, and more versatile.  That being said, I do like my picam, and use it in conjunction with my USB camera to get multiple angles.  I use my picam for closeup shots of my nozzle for print monitoring and my usb webcam to get a pleasing perspective.

spanzetta, 2020-07-07:
thanks a lot.. 
I will follow your suggestions.. and let you know
Thanks also for the patience (yes.. I missed the bitrate part.. sorry)
Regards

FormerLurker, 2020-07-07:
Not a problem, and welcome to the hobby!

spanzetta, 2020-07-08:
all solved now!! Thanks a lot for your great support!
Thanks