FormerLurker/Octolapse #486: camera error notif with no desc // rendering failled. ~~help

ken7ray, 2020-03-30:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: Octolapse 0.3.4

#### Version of OctoPrint

OctoPrint Version: 
OctoPrint 1.4.0 running on OctoPi 0.17.0

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred

1. very successful 8 hour time lapse after initial setup
2. changing vid. encoder to h264 (which later found out not compatible with external camera)
3. turn off "Clean Snapshots After Rendering" 
(hoping be able to render personally --later on found out there are specific tutorial to achieve that)

4. at this point "camera error" notif start to pop out and rendering failed as shown in the image.
 not sure if i accidentally changed other setting. (try several setting as suggested in forum and tutorial ~~me not smart ~~no fix ~~still error ~~still failled ~~sigh..)

[image]

5. camera tend to skip layer (first timelapse manage to capture every layer) ~~hv no idea why this happened, but fix anyw by changing "prime_smtng" frm 0.75 to 0.4 

6. reset all octolapse setting and set it up frm fress like in youtube tutorial ~~forget which one, but the one make my initial timelapse possible.

7. try to disable any other plugin.

8. try this : 
[image]

9. no result.

nb: hv no usb attached webcam/raspicam. will try using android with "ipcam" apps later (which just successfully integrated to my octoprint setup) as shown in this tutorial: https://github.com/foosel/OctoPrint/wiki/Using-an-Android-phone-as-a-webcam

#### What should have happened?
cool satisfying timelapse
then,
follow this tutorial:
https://community.octoprint.org/t/octolapse-how-to-turn-off-saving-and-rendering-to-the-pi/4341/5

#### What happened instead?
no timelapse - "camera error" pop-out - failed rendering only one image captured

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi 0.17.0
Os Version: OctoPi 0.17.0

#### Printer model & used firmware incl. version

Printer Model: creality ender 3 pro
Printer Firmware Version: 

#### Browser and version of browser, operating system running browser

Browser: 
Browser OS: 

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: 
https://github.com/ken7ray/ken7ray.git
https://github.com/ken7ray/ken7ray
(hv no idea how to upload/attach file)anyw,

first success timelapse with patric_buddha.gcode file
then try and failled with the other two files
nb: not trying the patric_budha.gcode file anymore because it takes too long.

#### Link to settings.json

Link to settings.json with all passwords removed:  
https://github.com/ken7ray/ken7ray
(Settings_octolapse_30-3-2020.json)

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  
https://github.com/ken7ray/ken7ray
(plugin_octolapse_log_30-03-2020_0630.log)

#### Link to octoprint.log

Link to octoprint.log:  
https://github.com/ken7ray/ken7ray
(octoprint_log_30-03-2020_0630.log)

#### Link to contents of Javascript console in the browser

Link to javascript console output: hv no idea

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: 
[image]

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2020-03-30:
So, it looks like you were mislead by the rendering error.  I will try to respond tomorrow after I have a chance to review all of your settings to see how things are configured.  I will also provide instructions for upgrading to v0.4.0rc1, which allows you to download a .zip file of your snapshots.  That should make it much easier to render manually.

Also, Octolapse 0.4.0rc1 supports H.264 and H.265, but my pi (Pi3 b+) doesn't have enough memory to render anything substantial with H.264, and I've not been able to get 265 working on the RPIl.  My PC works fine for both though.  I've added a warning for both codecs in v0.4.0rc1.

ken7ray, 2020-04-03:
hi, just want to post some update.
just tried octolapse with my old android phone with 'ipcam' app (setup as webcam). 
It's all success with no problem whatsoever even with h264 output.
quality kinda bad tho (old phone cam problem), and it's quit unstable (prob be the app itself).

[image]
…]

so far still no luck with my dslr setup. which hopefully work out soon, for a better quality output and stable connection.

should i update to 0.4.0rc1? or is it not the main problem?

anyw thnx for the great app, really look forward to work well with my setup. cheers!

FormerLurker, 2020-04-04:
Hmm.. I'm really not sure if it's due to the old version, but since it could be, why not update?  Try installing using the following instructions:

1. Open the OctoPrint settings (wrench/spanner icon)
2. Select the *Software Update* from the left menu
3. Click the spanner/wrench icon to edit the software update plugin settings
4. Change the OctoPrint Release Channel to 'Maintenance RCs' and save your changes.
5. Click *Check for Updates Now*
6. OctoPrint will report an update is available.  Update, reboot, and the new version should be installed.

Read [this guide](https://github.com/FormerLurker/Octolapse/wiki/V0.4-Getting-Started-with-Octolapse) first!   Let me know if you have any problems OR if it seems to work great.  I'll be here to help should you need assistance!

FormerLurker, 2020-05-08:
Have you had a chance to try my suggestion above for upgrading to the lastest release candidate and then trying your DSLR again?