FormerLurker/Octolapse #487: octolapse no Z change after picture is taken

dave11674, 2020-03-30:
settings and gcode on the forums FormerLurker,
[octolapse and gcode.zip](https://github.com/FormerLurker/Octolapse/files/4405879/octolapse.and.gcode.zip)

trying to add here also

thanks

dave11674

FormerLurker, 2020-03-30:
Thanks @dave11674!  This is a brand new issue for me, which is pretty rare, so I appreciate you taking the time to post here!

dave11674, 2020-03-30:
anytime mate, i like to help, its you who should be thank a million times
over for producing such cracking software/plugins  ;)
and the support too
amazing

dave

FormerLurker, 2020-04-04:
> anytime mate, i like to help, its you who should be thank a million times over for producing such cracking software/plugins  ;)

Well, I don't do cracking software, but thank you!

So, I imported your settings and ran the gcode through the debugger.  I'm seeing layer changes being sent to the printer.  Perhaps I misunderstood what you are reporting.  Are you saying that the layer isn't changing immediately after the snapshot, but does change layers at some point?  If so, when using the smart layer triggers, Octolapse will usually take snapshots somewhere in the middle of a layer since it's trying really hard to find the fastest/lowest impact place to take a snapshot.

Let me know if I'm understanding your issue properly.  I really want to resolve this before I drop RC2, which I would otherwise release today.

Also, sorry about the delayed response.  I've been occupied this week printing and packaging face shields.  I've also been working on another plugin, but not too much.  Keep an eye on my repositories for the new project :)

Thanks!

dave11674, 2020-04-04:
Ok

Hope i can explain correctly

Once layer 1 is complete the head and bed go and assume the position andddd
SNAPSHOT....
So at this point i believe z should step up to layer 2 and print layer 2
layers
But its not stepping up
Its staying at level 1 and printing layer 2 print at this level
And never steps up after each snapshot
Stays st layer 1 :(

FormerLurker, 2020-04-04:
OK, that's what I thought you said originally, but I wasn't seeing it when running the debugger..  hmmm...

Try this:  turn on test mode (slider on octolapse tab) so you won't waste filament or mess up your bed.  Then set the logging profile to 'debug'.  Edit the logging profile and click 'Clear Log'.  Print 3 or 4 layers, cancel the print, then edit your logging profile again and click 'Download Log'.  Paste the log into gist.github.com, and paste a link to the gist here.  That should clear a few things up..

dave11674, 2020-04-04:
ok just updated it to the latest
will do the test print now

give me 10 mins pls

dave

FormerLurker, 2020-04-04:
Take your time!  Am in no rush and have lots of time 😊

dave11674, 2020-04-04:
https://gist.github.com/dave11674/4771f453cb1275e4e5b0cf046475cefd

ok uploaded the log
it looked like it was working then
the Z was moving up

ill try a normal print now
see what that does
fingers crossed lol

dave

FormerLurker, 2020-04-04:
V0.4 solves so many issues.  I can't wait to deploy.  Hopefully it solves this as well.  Fingers crossed.

dave11674, 2020-04-04:
ok

see how this print goes

if still the same hopefully your new RC2 will fix it :)

whats new ?

dave

FormerLurker, 2020-04-04:
Not much from rc1.  G2/g3 support, fixed some shut down issues and a few other minor things.

What i meant was that v0.4 fixes so many problems vs 0.3.4.  i was talking about other problems, not yours specifically.  Sorry for the confusion

dave11674, 2020-04-04:
ok i think ive sussed it
its not the Z axis thats the problem
its the E axis
it starts of extruding as normal then stops :S
driving me insane lol
tried about 3 different prints now, all in octolapse they do this
ive had the head stripped down too just to make sure there is nothing
jamming up inside lol
ive not set it going from the SD card, no linux involved lol
let you know

dave

FormerLurker, 2020-04-04:
Excellent, i think i know the issue.  Edit your printer profile and set G90/G91 Influences extruder to true.  Let me know if that works.

dave11674, 2020-04-04:
Oh lol
Ive switched the bastard off lol
Driving me insane
Cant get it to load a filament at all now
Think the FINDA is fluffed as it keeps reporting N/A
ive read on prusa forums thats BAD
So will order a new one and let you know how i get on

FormerLurker, 2020-04-04:
Is it a mini??  If so, check the printer profile and look for your make and model.  I think I set that correctly.

Let me know when you get back to it, and good luck with your printer!!

dave11674, 2020-04-04:
MK3S  & MMU2S

FormerLurker, 2020-04-04:
Are you running release candidate firmware by chance?  They changed g90/91 behavior recently, but changed it back.  Try updating to the latest release, then revert that setting to false (g90/g91 influence extruder).  I posted to their github about this, and they were very understanding.

FormerLurker, 2020-04-04:
The latest firmware release for the mk3 i mean.

dave11674, 2020-04-05:
Yes i always update to the latest firmware
So you think i should roll back one ?

Dave

FormerLurker, 2020-04-05:
Send me the version number please.  You can get it off the printer display.  Making dinner so I may be slow to respond, fyi, and tomorrow is my wife's b day :) will definitely respond asap though, but just in case I thought I would warn you.

dave11674, 2020-04-05:
Thanks mate
And a happy birthday to your missus 👍

dave11674, 2020-04-05:
Know where i can get a finda from ??
Just in case this one is junk
As it keeps reporting NA
BUT its hit n miss
Grrrr

FormerLurker, 2020-04-05:
You will have to go through their customer service.  Just go to their store and start a chat.  They are very helpful.

dave11674, 2020-04-05:
Taking forever go respond
Im usually looking away when they do
Then close chat right away lol

dave11674, 2020-04-05:
Ok flashing backwards now

dave11674, 2020-04-05:
Funny you say the 390 had issues as i just flashed new fw before setting
all this octolapse up

Dave

FormerLurker, 2020-04-05:
It should be 3.9.0rc1 only, not the master.  [Check out this issue on the prusa firmware repo]( https://github.com/prusa3d/Prusa-Firmware/issues/2516#event-3169430249)

Looks like 3.9.0 hasn't dropped yet, so I imagine you have the release candidate..  Whew.. that explains a lot.  I'm SO SO glad they reverted this change, else it would have broken many stable configurations!  If you read the linked issue thread, you'll see they were quite reasonable about everything!

dave11674, 2020-04-05:
I went back to 3.8.1
Its sll working fine now :)

dave11674, 2020-04-05:
ok that worked 80% lol
i need zoom out for the cam
is that option available  in the settings ?
i cant seem to find it anywhere

thanks

dave

FormerLurker, 2020-04-05:
Raspi cam??  There is no zoom option for that camera.  Make sure image stabilization is disabled.  Due to a bug in the driver, you need to reboot if it was already enabled :(

Other than that, changing the resolution results in a kind of digital zoom.  Use the highest resolution possible .

Fyi, this is a known issue Im looking into.  It isn't related to octolapse, but rather the pi cam driver.

dave11674, 2020-04-05:
Whats highest res for this cam
And sill it not cause problems with my print ?
And how do i know the resolution is set ?
Is there any commands in shh i can ping the camera
To report back the res ?

Thanks

Dave

FormerLurker, 2020-05-08:
Your issue got buried, sorry.  The camera resolution is not something Octolapse can affect, so you may want to look for some picam assistance.  Have all of your other problems been resolved?

dave11674, 2020-05-08:
All good now mate
Sweet as a nut
I put res back to 720 all is sweet again ;)

Dave

FormerLurker, 2020-05-08:
Thanks Dave!  Closed :)

closed 2020-05-08 by FormerLurker