FormerLurker/Octolapse #488: Python 3 compatibility

Idus, 2020-04-01:
I was just asked by my OctoPrint to do a Python 3 test and report an issue to all plugins that were found incompatible with Python 3 (if not already such an issue was already stated).
Did the test and got back that the your plugin was Incompatible with Python 3 and could not find any other open issue about it - so here it is. :)

Cheers

FormerLurker, 2020-04-04:
Thanks for the report.  The issue is I have not deployed a release version of Octolapse V0.4.0, which is python 3 compatible.  Until I update the plugin repository, it will still show up as incompatible.  I'm getting close, and plan to release V0.4.0rc2 today.  I think this will be the last RC, since I've only had 1 major issue and 1 minor enhancement (G2/G3 support) since rc1.

It would be nice if the python 3 test could look at pre-release versions, but I believe it's the plugin repository that needs to be updated still, and I can't post pre-releases there unfortunately.

Sorry about the delay in responding, btw.  I've been very engaged with some other projects this week (new plugin and COVID face shield prints and delivery).  I'm getting back to work today though.

Idus, 2020-04-05:
[..] _Sorry about the delay in responding, btw. I've been very engaged with some other projects this week (new plugin and COVID face shield prints and delivery). I'm getting back to work today though._ [..]

Sorry on my behalf - I was just forwarding the result and taking it to your attention IF you by any chance did not know, no need to stress. I know that you and many others have quite few other things to prioritize - like life in general and such. ;) Me too is printing COVID face-shields with the little resources I have available - but many trickles creates a river. ;)

Thank you for your time and effort. I hope that the RC release works as you have predicted.

Cheers

closed 2020-04-05 by Idus

FormerLurker, 2020-04-05:
Oh, I'm VERY glad you posted!  I'm always glad to hear from users, especially when they are pointing out critical details like this (python 3 support).  Absolutely no need to apologize!  Please continue to post similar tickets in the future, they are welcome!