FormerLurker/Octolapse #490: [Request] Python 3.6 support

Nickjallday, 2020-04-05:
#### If this is a feature request describe it here

Ocotoprint will no longer support previous versions of python in the near future, it would be awesome if this plugin could get updated to be compatible with python 3.6.

FormerLurker, 2020-04-05:
See the ticket for [v0.4.0rc1](https://github.com/FormerLurker/Octolapse/issues/475).  I need testers still if you are interested.  See [this guide](https://github.com/FormerLurker/Octolapse/wiki/V0.4-Getting-Started-with-Octolapse), and install via the plug-in manager with this url: https://github.com/FormerLurker/Octolapse/archive/v0.4.0rc1.zip

FormerLurker, 2020-04-10:
Were you able to try this out?

nilsbebelaar, 2020-04-11:
Just chipping in here, I managed to build Python3.6 on my Raspberry Pi from source and I'm now running Octolapse v0.4.0rc1 without any issues so far. I needed Python3.6 for a script that is connecting to my GoPro over Wi-Fi and taking a picture, for now it keeps the pictures on the SD card of the GoPro. I might look into transfering the picture to the Pi over Wi-Fi as well and then have Octolapse do the timelapse rendering. (Currently doing that manually which isn't so bad tbh). 
Now that I had installed Python3.6, I thought I might as well try the new Octolapse, I've not had any issues with it so far!

One other question, not sure if it's okay to put it here, else just let me know where to move it to. Would it somehow be possible to parse the total amount of photos that will be taken to the start or picture trigger scripts? I'm looking into making an orbiting setup for my GoPro, it would be amazing to know how far the camera should orbit per photo.

FormerLurker, 2020-04-11:
@nilsbebelaar, thanks for chiming in!  It's awesome to hear from people who have gotten it working (I usually only hear about issues).

Regarding your other question, can you clarify a bit?  The snapshot preview when using the smart triggers will show you how many pictures will be taken.  Is there some other place you need this info?

nilsbebelaar, 2020-04-11:
> can you clarify a bit?

So I'm using an external script to trigger the GoPro to shoot a picture, now I want to make a system that orbits the printer during one print. So ideally I'd have the script calculate how far the camera would have to orbit per picture, to make sure that it completes it's orbit over the full length of the print. So basically what I'm asking is, if it's possible to parse the total amount of pictures that is shown in the snapshot preview to the script that is called for taking a picture (or the script that runs at the start of printing, both should be workable solutions)

FormerLurker, 2020-04-11:
Yeah, that should be possible.  I could pass in -1 if the number of images is unknown (not using the smart triggers).  I will look into it.  Can you open a feature request and be as specific as possible about it?  Once I release 0.4.0 I'll look into adding it.  Shouldn't be too hard.

FormerLurker, 2020-04-26:
I think this issue is resolved, so I'm going to close it.  Let me know if you want it reopened.

Thanks!

closed 2020-04-26 by FormerLurker