FormerLurker/Octolapse #491: Webcam Camera Settings Overexposed

artinflight, 2020-04-06:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.3.4

#### Version of OctoPrint

OctoPrint Version: 1.4.0

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred

External Webcam Settings cannot be set to Auto while running External DSLR Octolapse.
Have manual external webcam settings set in Octolapse but when print starts it overexposes the image to barely visible. Not using external webcam for octolapse. Have tried disabling/enabling external webcam. Have tried Manual/Auto camera settings for external webcam. Have tried deleting/adding external webcam profile. All produce the same result of an overexposed webcam feed. Please see screenshot. Have tried adjusting all auto exposure related settings in profile. I just need the external webcam settings set to Auto, it works perfectly when I disable Octolapse.

#### What should have happened?
PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: Octopi
Os Version: 0.17.0

#### Printer model & used firmware incl. version

Printer Model: Anycubic Mega i3
Printer Firmware Version: PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: Chrome
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: https://misctemp.s3.ca-central-1.amazonaws.com/Settings.json

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  Paste crashed browser in both Github and Pastebin. Log available here instead: https://misctemp.s3.ca-central-1.amazonaws.com/plugin_octolapse.log

#### Link to octoprint.log

Link to octoprint.log:  https://gist.github.com/artinflight/cf222cd1117dc7ce591559c41687327a

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: https://misctemp.s3.ca-central-1.amazonaws.com/octolapse.jpg

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2020-04-10:
I sort of remember this issue (I think you are the first to report it, but I noticed and fixed in the latest beta).  Change your camera type to 'webcam', scroll down a bit in the camera profile and uncheck 'apply settings before print'.  Then switch the camera type back to external script and save the settings.  Let me know if that helps.  Otherwise I suggest you update to the latest pre-release (I reworked all of the camera controls and have improved the script camera functions/tests/logging).

Thanks!

FormerLurker, 2020-05-08:
Have you been able to try  my suggestion?

artinflight, 2020-05-09:
Thanks for following up. I haven't been able to yet, have been printing PPE
non-stop and haven't had time to run any time-lapses. Will follow up as
soon as I have a chance to try this out.