FormerLurker/Octolapse #492: connection reset after 1 snapshot

Clarepz, 2020-04-06:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.3.4

#### Version of OctoPrint

OctoPrint Version: 1.4.0

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred

1. I installed gphoto2 (for a Canon powershot G10)  with this guide: https://pimylifeup.com/raspberry-pi-dslr-camera-control/  because the "official" method : https://github.com/FormerLurker/Octolapse/wiki/Configuring-an-External-Camera   didn't work for me
2. I followed the "official" guide from step 3
3. Sliced a benchy gcode with cura and upload it to the octopi
4. I started the print

#### What should have happened?
run the print and take a photo at each layers.

#### What happened instead?
the print starts and the camera takes 1 snapshot, then stop, I cannot stop the print or shut down the system  I refresh the page but it's inaccesible. i have to unplug the pi and plug it back to access it.

sometimes the print continues, the first layer is made and another snapshot is taken but it makes maximum 3 layers (and 3 photos)and then the connection is stoped.

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version:  0.17.0

#### Printer model & used firmware incl. version

Printer Model: Creality Ender 3
Printer Firmware Version: Marlin (i didn't change anything)

#### Browser and version of browser, operating system running browser

Browser: Chrome
Browser OS: windows 10

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: https://gist.github.com/Clarepz/d1950c1c5d02233c2960bd01cefa8560

#### Link to settings.json

Link to settings.json with all passwords removed: https://gist.github.com/Clarepz/d1950c1c5d02233c2960bd01cefa8560

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://gist.github.com/Clarepz/d1950c1c5d02233c2960bd01cefa8560

#### Link to octoprint.log

Link to octoprint.log:  https://gist.github.com/Clarepz/d1950c1c5d02233c2960bd01cefa8560

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2020-04-10:
It looks like the script isn't returning, though I'm not sure why Octoprint would loose connectivity...  Would you be willing to upgrade to v0.4.0rc1?  It's got much better script handling and log output.  If you'd like to try, check out [this guide](https://github.com/FormerLurker/Octolapse/wiki/V0.4-Getting-Started-with-Octolapse).  Read the whole thing before starting, then in the [installation section](https://github.com/FormerLurker/Octolapse/wiki/V0.4-Getting-Started-with-Octolapse#installation), follow the 2nd method (adjusting the software update plugin settings to get the new version).  If not we can continue debugging in the current version.

FYI, DSLRs can be very trick to get working since there is no real standard.  Sometimes a newer version of GPhoto2 is necessary, and some cameras just don't work.  However, we can try our best!

Thanks!

Clarepz, 2020-04-10:
Thank you for your advices I will try :)

FormerLurker, 2020-05-08:
Have you been able to try out the new version to see if it solves your issue?

Clarepz, 2020-05-08:
No, not yet, my printer main board fried when I was trying to upgrade the
Firmware for other reason. I finally eceived a new one last week. I will
try the new version as soon as a can.

FormerLurker, 2020-05-09:
No problem!  Just going through all of the issues today.  Man are there a lot :)

Clarepz, 2020-05-09:
Good luck!