FormerLurker/Octolapse #493: Installation failed due to missing zlib

enyo, 2020-04-09:
During install on OctoPrint I got this error:

```
The headers or library files could not be found for zlib,
a required dependency when compiling Pillow from source.

Please see the install instructions at:
https://pillow.readthedocs.io/en/latest/installation.html

----------------------------------------
ERROR: Failed building wheel for pillow
Successfully built Octolapse
Failed to build pillow
Installing collected packages: pillow, Octolapse
Running setup.py install for pillow: started
Running setup.py install for pillow: finished with status 'error'
```

#### Version of OctoPrint

OctoPrint Version: `1.4.0`

#### Operating System running OctoPrint and Octolapse

Running Armbian on an Orange Pi.

enyo, 2020-04-09:
The solution was to install zlib-dev:

```
sudo apt-get install zlib1g-dev
```

closed 2020-04-09 by enyo

FormerLurker, 2020-04-10:
Thank you for posting your solution!  I will add it to the troubleshooting instructions.

enyo, 2020-04-16:
No problem. Thanks for this tool. I suggest adding it to the installation instructions, since it's likely it's not installed.